mkeeter/antimony #124: Problem building on ubuntu

rvba, 2015-10-08:
Hi!

While trying to build antinomy on Ubuntu 14.04 I get this first error

```
g++ -c -m64 -pipe -Werror=switch -O3 -fPIC -Wall -W -D_REENTRANT -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I../../../lib/fab -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I../../../lib/fab/inc -I../../../lib/fab/vendor -I. -I. -I. -I../../../lib/fab -I.  -o fab.o ../../../lib/fab/src/fab.cpp
In file included from /usr/include/boost/python/detail/prefix.hpp:13:0,
                 from /usr/include/boost/python/args.hpp:8,
                 from /usr/include/boost/python.hpp:11,
                 from ../../../lib/fab/src/fab.cpp:1:
/usr/include/boost/python/detail/wrap_python.hpp:50:23: fatal error: pyconfig.h: Aucun fichier ou dossier de ce type
 # include <pyconfig.h>
                       ^
compilation terminated.

```

I did add 

-I/usr/include/python3.4

on the build/lib/fab/makefile

but then I get this new error

```
g++ -c -m64 -pipe -Werror=switch -O3 -fPIC -Wall -W -D_REENTRANT -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I../../../lib/fab -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I../../../lib/fab/inc -I../../../lib/fab/vendor -I. -I. -I. -I../../../lib/fab -I. -I/usr/include/python3.4 -o fab.o ../../../lib/fab/src/fab.cpp
In file included from ../../../lib/fab/src/fab.cpp:5:0:
../../../lib/fab/inc/fab/types/shape.h:14:14: error: ‘tuple’ in namespace ‘std’ does not name a template type
 typedef std::tuple<int,int,int> int3;
              ^
../../../lib/fab/inc/fab/types/shape.h:21:11: error: ‘int3’ has not been declared
           int3 color=int3(-1,-1,-1));
           ^
../../../lib/fab/inc/fab/types/shape.h:46:10: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
     std::shared_ptr<MathTree> tree;
          ^
../../../lib/fab/inc/fab/types/shape.h:21:35: error: ‘int3’ was not declared in this scope
           int3 color=int3(-1,-1,-1));
                                   ^
make[1]: *** [fab.o] Erreur 1

```

Any idea ?
Thanks!

mkeeter, 2015-10-08:
What compiler version are you using?

You'll need a C++11-compliant compiler (which includes `std::tuple` and `std::shared_ptr`).
There are instructions [here](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md#linux) on how to update Ubuntu's default compiler.

rvba, 2015-10-08:
My bad ! I forgot to use qmake from Qt 5. All when smoothly !
Giving it a try now, thanks && congrats !

closed 2015-10-08 by rvba