mkeeter/antimony #126: python 3.5

oberon2007, 2015-10-28:
I am trying to run antimony in Manjaro (Arch-based)
The PKGBUILD from the AUR had to be modified to compile with python 3.5 but now the application does not seem to run. It gives ``Could not find fab Python module.
Antimony will now exit.``
Is there a way to make that work? Or is Antimony currently not ready to run with python 3.5?

willingc, 2015-10-28:
@oberon2007 I was able to successfully build and run Antimony yesterday using Python 3.5 on Mac OS X 10.10.5. 

If you pastebin your process, I may be able to help troubleshoot.

mkeeter, 2015-10-30:
Yes, there's not reason to believe it can't work with Python 3.5
@oberon2007 , you should see a more detailed output log in the terminal.

I walked someone else through debugging a similar issue [here](https://github.com/mkeeter/antimony/issues/100).  Can you follow those debugging steps and see if anything emerges?

oberon2007, 2015-10-30:
The reason why I assume that python version is to blame is that building from the AUR PKGBUILD gives error
> /usr/bin/ld: cannot find -lpython3.4m
When patching https://github.com/mkeeter/antimony/blob/develop/qt/python.pri#L22
to ``LIBS += -lpython3.5m`` it compiles successfully, but then the program exits with
> Import Error:
> Could not find fab Python module.
> Antimony will now exit.

willingc, 2015-10-31:
@oberon2007 You mention that you are using AUR PKGBUILD. Did you update the PKGBUILD file (https://aur.archlinux.org/cgit/aur.git/plain/PKGBUILD?h=antimony) for the correct python3.5 location in the sed command?

Also you may need to update the symlink to Qt (https://wiki.archlinux.org/index.php/Qt#Qt5_Platform_Plugins_and_Missing_Dependencies).

Another issue may be where you are building is not relative to the location of fab module.

You could also try make clean and then make again.

ApostolosB, 2015-11-01:
@oberon2007 better contact the PKGBUILD maintainer for this issue.

oberon2007, 2015-11-01:
yes, i have already, thanks :)

closed 2015-11-01 by oberon2007

ApostolosB, 2015-11-18:
@mkeeter 

In order for antimony to build on Arch it needs to patch what is described here https://github.com/mkeeter/antimony/issues/126#issuecomment-152673873 Does this change affect anything else in another distro or OS? And is there a way to make it work for all things?

mkeeter, 2015-11-18:
We already search for a few different boost::Python names [here](https://github.com/mkeeter/antimony/blob/develop/qt/python.pri#L43-L48).  If you submit a patch to do the same for `python3.4m` and `python3.5m`, I'll merge it in.

ApostolosB, 2015-11-18:
I am no coder (no Qt no python) by any means and this might be a stupid solution but here goes:

Replace: 
LIBS += -lpython3.4m
With:
LIBS += $$system(/usr/bin/python3-config --ldflags | grep -o  -- '-lpython[0-9].[0-9]m\>')

I spend the whole afternoon looking at Qt docs trying to find a way to do it but in the end came up with the above.
The idea is that since it is already in the QMAKE_LFLAGS -as you mention in the comment- i extract it the same way and strip whatever is not needed with grep.

I made the change locally and it compiled with my PKGBUILD on arch so it seems to work. If it is acceptable tell me to make a proper patch.

mkeeter, 2015-11-18:
I cleaned up the regex a tiny bit (assuming first digit is 3 and removed trailing `>`).

This seems to work for me: 
```
LIBS += $$system(/usr/bin/python3-config --ldflags | grep -o -- '-lpython3.[0-9]m')
```
(on a system with Python 3.4 instead of 3.5).

If you want to open up a PR making this change, I'll merge it in
(or will make the change myself when I get around to it).

ApostolosB, 2015-11-18:
Done :)