mkeeter/antimony #127: Define "black boxes" from within the program using existing components

2mac, 2015-10-31:
Is there not a way to take a particular box and, given its inputs, define a custom object type that can be reproduced at will?

For instance, in the demonstration video, a screwdriver handle is created. Is there not a way to save that handle concept as its own black box for use later?

mkeeter, 2015-10-31:
Right now, nodes are this kind of black box (but have a script inside them, rather than a graph).

Nested graphs are in development...

2mac, 2015-10-31:
Right. As a programmer, I'm okay defining nodes in Python, but people who just want to do CAD will not appreciate having to drop to that level to use their own custom designs.

From what I can tell, your design for Antimony was for a CAD design to resemble a Lisp program in structure. In Lisp, these things I define can be stored as procedures. Similarly, the user should be able to define a custom node from within the program given its current definition, and all they should need to type would be definitions of the inputs and outputs and how they correspond to the structures in that tree.

This is an opportunity to create an intuitive visual programming language with all the power of Lisp in its structure combined with the power of Python for creating new primitives.