mkeeter/antimony #130: Problem building on Linux Mint 17.1 Rebecca

Pupo1, 2015-11-04:
Hello anyone

I have problems building antimony on Linux Mint 17.1, it is the same error if I do "make install" or "make -j8". I did everything as indicated in the building manual of Github, except for "~/Qt5.4.0/5.4/gcc_64/bin/qmake ../sb.pro" which didn't work so I typed "qmake ../sb.pro", which showed no error.
```
pulmo@pulmo-ThinkPad-T410s:~/antimony/build$ make install
cd lib/fab/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/pulmo/antimony/lib/fab/fab.pro -o Makefile ) && make -f Makefile install
make[1]: Entering directory `/home/pulmo/antimony/build/lib/fab'
make[1]: Nothing to be done for `install'.
make[1]: Leaving directory `/home/pulmo/antimony/build/lib/fab'
cd lib/graph/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/pulmo/antimony/lib/graph/graph.pro -o Makefile ) && make -f Makefile install
make[1]: Entering directory `/home/pulmo/antimony/build/lib/graph'
make[1]: Nothing to be done for `install'.
make[1]: Leaving directory `/home/pulmo/antimony/build/lib/graph'
cd app/ && ( test -e Makefile || /usr/lib/x86_64-linux-gnu/qt5/bin/qmake /home/pulmo/antimony/app/app.pro -o Makefile ) && make -f Makefile install
make[1]: Entering directory `/home/pulmo/antimony/build/app'
g++ -c -m64 -pipe -Werror=switch -I/usr/include/python3.4m -I/usr/include/python3.4m -D'GITREV="5c144cc"' -D'GITTAG=""' -D'GITBRANCH="develop"' -O3 -std=c++0x -Wall -W -D_REENTRANT -fPIE -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -DQT_NETWORK_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -I../../app -I../../app/src -I../../lib/fab/inc -I../../lib/graph/inc -I/usr/include/qt5 -I/usr/include/qt5/QtOpenGL -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I. -I. -I. -o src/graph/node/serializer.o ../../app/src/graph/node/serializer.cpp
In file included from /usr/include/qt5/QtCore/qjsonobject.h:45:0,
                 from /usr/include/qt5/QtCore/QJsonObject:1,
                 from ../../app/src/graph/node/serializer.h:9,
                 from ../../app/src/graph/node/serializer.cpp:5:
/usr/include/qt5/QtCore/qjsonvalue.h: In function ‘QJsonObject SceneSerializer::run(Graph*, QMap<Node*, QPointF>)’:
/usr/include/qt5/QtCore/qjsonvalue.h:119:12: error: ‘QJsonValue::QJsonValue(const void*)’ is private
     inline QJsonValue(const void *) {}
            ^
../../app/src/graph/node/serializer.cpp:28:17: error: within this context
     out["type"] = "sb";
                 ^
../../app/src/graph/node/serializer.cpp: In function ‘QJsonObject SceneSerializer::serializeNode(Node*, QMap<Node*, QPointF>)’:
../../app/src/graph/node/serializer.cpp:49:34: error: no matching function for call to ‘QJsonArray::QJsonArray(<brace-enclosed initializer list>)’
             inspectors[node].y()});
                                  ^
../../app/src/graph/node/serializer.cpp:49:34: note: candidates are:
In file included from /usr/include/qt5/QtCore/QJsonArray:1:0,
                 from ../../app/src/graph/node/serializer.h:10,
                 from ../../app/src/graph/node/serializer.cpp:5:
/usr/include/qt5/QtCore/qjsonarray.h:206:5: note: QJsonArray::QJsonArray(QJsonPrivate::Data*, QJsonPrivate::Array*)
     QJsonArray(QJsonPrivate::Data *data, QJsonPrivate::Array *array);
     ^
/usr/include/qt5/QtCore/qjsonarray.h:206:5: note:   candidate expects 2 arguments, 1 provided
/usr/include/qt5/QtCore/qjsonarray.h:61:5: note: QJsonArray::QJsonArray(const QJsonArray&)
     QJsonArray(const QJsonArray &other);
     ^
/usr/include/qt5/QtCore/qjsonarray.h:61:5: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘const QJsonArray&’
/usr/include/qt5/QtCore/qjsonarray.h:58:5: note: QJsonArray::QJsonArray()
     QJsonArray();
     ^
/usr/include/qt5/QtCore/qjsonarray.h:58:5: note:   candidate expects 0 arguments, 1 provided
make[1]: *** [src/graph/node/serializer.o] Error 1
make[1]: Leaving directory `/home/pulmo/antimony/build/app'
make: *** [sub-app-install_subtargets-ordered] Error 2
```

Could someone tell me what is wrong? Thank you in advance for any idea

mkeeter, 2015-11-04:
Can you check versions of Qt and gcc?  (`qmake --version`; `gcc --version`).

Pupo1, 2015-11-06:
Sorry for not replying immediately, I didn't see your comment. 
``
QMake version 3.0
Using Qt version 5.2.1 in /usr/lib/x86_64-linux-gnu
pulmo@pulmo-ThinkPad-T410s:~/Arbeitsfläche$ 
``

and 

``
gcc (Ubuntu 4.9.3-5ubuntu1~14.04) 4.9.3
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
``

Pupo1, 2015-11-06:
Just updated to gcc 5.2.1, which brought no effect.

Pupo1, 2015-11-06:
``$gcc --version
gcc (Ubuntu 5.2.1-23ubuntu1~14.04.2) 5.2.1 20151031
``
and 
``qmake --version
QMake version 3.0
Using Qt version 5.2.1 in /usr/lib/x86_64-linux-gnu``

tried the make -j8 again and received the same error.

Pupo1, 2015-11-06:
Now I deleted the antimony folder and, and repeated the process:
``git clone https://github.com/mkeeter/antimony
cd antimony
mkdir build
cd build
qmake ../sb.pro
make -j8``

still the same error.

mkeeter, 2015-11-06:
Please install a more recent version of Qt, as described in the [BUILDING](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md) guide.

I'm still on 5.4, but other folks have been successful with 5.5.

Pupo1, 2015-11-07:
I corrected the link to Qt 5.5: 
``$qmake -v
QMake version 3.0
Using Qt version 5.5.1 in /home/pulmo/Qt5.5.1/5.5/gcc_64/lib``

now it works! I can run antimony now. Thank you very much!

closed 2015-11-07 by Pupo1