mkeeter/antimony #131: Memory leak or crash when viewing certain file

2mac, 2015-11-05:
I created a model yesterday, and on multiple machines when I open it again, it will either crash with a `std::bad_alloc` or it will consume lots of memory and saturate the disk I/O.

You can [download this copy of the saved file](http://mcmackins.org/dl/af1-napkin-holder.sb) for testing.

2mac, 2015-11-05:
I was able to get around it by opening the file and quickly closing the 3D view window. It seems the initial render is causing the issue, if that's any help.

mkeeter, 2015-11-05:
Yeah, I think that it's trying to allocate a huge pixel buffer for the render.  The fix will be to only render the section of the model that's actually going to be drawn on the screen.