mkeeter/antimony #133: appdata file

sagitter, 2015-11-07:
Please, add an appdata file in the source archive; something like:

```
<?xml version="1.0" encoding="UTF-8"?>
<!-- Copyright 2015 Matthew Keeter <jmatt.j.keeter@gmail.com> -->
<application>
  <id type="desktop">antimony.desktop</id>
  <metadata_license>CC0-1.0</metadata_license>
  <project_license>MIT</project_license>
  <summary>Computer-aided design CAD tool</summary>
  <releases>
   <release version="%{version}-%{release}" timestamp="1441242000" />
  </releases>
  <categories>
   <category>Science</category>
  </categories>
  <description>
    <p>
      Antimony is a computer-aided design (CAD) tool from a
     parallel universe in which CAD software evolved
     from Lisp machines rather than drafting tables.
     
     Antimony provides the Python3 Fab that is
     a set of software tools for personal fabrication,
     intended for use with machines common to fab labs.
     For info about Fab visit kokompe.cba.mit.edu/.
    </p>  
  </description>
  <url type="homepage">http://www.mattkeeter.com/projects/antimony/3/</url>
  <screenshots>
    <screenshot type="default">"Link to a screenshot (16:9 aspect ratio) (png or jpeg format </screenshot>
    <screenshot type="default">"Link to another screenshot</screenshot>
  </screenshots>
  <updatecontact>jmatt.j.keeter_at_gmail.com</updatecontact>
</application>
```

Thanks.

mkeeter, 2015-11-07:
Pull requests are welcome; put it somewhere in `deploy/Linux`.

sagitter, 2015-11-07:
Have you already some screenshots ready?

mkeeter, 2015-11-07:
How's this?
[image]

sagitter, 2015-11-07:
Yes, it's fine.
Even this is interesting: http://www.mattkeeter.com/projects/antimony/3/images/app.png