mkeeter/antimony #139: error: '__builtin_ia32_readeflags_u64' was not declared in this scope

hhanff, 2015-11-19:
Hi!

When trying to compile antimony I get lots of errors:

```
cd lib/fab/ && ( test -e Makefile || /opt/Qt/5.4/gcc_64/bin/qmake /home/hhanff/Projekte/antimony/lib/fab/fab.pro -o Makefile ) && make -f Makefile 
make[1]: Verzeichnis »/home/hhanff/Projekte/antimony/build/lib/fab« wird betreten
make[1]: Für das Ziel »first« ist nichts zu tun.
make[1]: Verzeichnis »/home/hhanff/Projekte/antimony/build/lib/fab« wird verlassen
cd lib/graph/ && ( test -e Makefile || /opt/Qt/5.4/gcc_64/bin/qmake /home/hhanff/Projekte/antimony/lib/graph/graph.pro -o Makefile ) && make -f Makefile 
make[1]: Verzeichnis »/home/hhanff/Projekte/antimony/build/lib/graph« wird betreten
g++ -c -pipe -Werror=switch -I/usr/include/python3.4m -I/usr/include/python3.4m -O3 -fPIC -std=c++0x -Wall -W -D_REENTRANT -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -I../../../lib/graph -I. -I../../../lib/graph/inc -I/opt/Qt/5.4/gcc_64/incl
ude -I/opt/Qt/5.4/gcc_64/include/QtGui -I/opt/Qt/5.4/gcc_64/include/QtCore -I. -I/opt/Qt/5.4/gcc_64/mkspecs/linux-g++ -o src/hooks/output.o ../../../lib/graph/src/hooks/output.cpp
In file included from /usr/lib/gcc/x86_64-linux-gnu/4.9/include/x86intrin.h:27:0,
                 from /usr/include/x86_64-linux-gnu/c++/4.9/bits/opt_random.h:33,
                 from /usr/include/c++/4.9/random:50,
                 from /usr/include/c++/4.9/bits/stl_algo.h:66,
                 from /usr/include/c++/4.9/algorithm:62,
                 from /usr/include/boost/function/detail/prologue.hpp:13,
                 from /usr/include/boost/function/function_template.hpp:13,
                 from /usr/include/boost/function/detail/maybe_include.hpp:13,
                 from /usr/include/boost/function/function0.hpp:11,
                 from /usr/include/boost/python/errors.hpp:13,
                 from /usr/include/boost/python/handle.hpp:11,
                 from /usr/include/boost/python/args_fwd.hpp:10,
                 from /usr/include/boost/python/args.hpp:10,
                 from /usr/include/boost/python.hpp:11,
                 from ../../../lib/graph/inc/graph/hooks/output.h:2,
                 from ../../../lib/graph/src/hooks/output.cpp:3:
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/ia32intrin.h: In function 'long long unsigned int __readeflags()':
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/ia32intrin.h:246:41: error: '__builtin_ia32_readeflags_u64' was not declared in this scope
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/ia32intrin.h: In function 'void __writeeflags(long long unsigned int)':
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/ia32intrin.h:254:36: error: '__builtin_ia32_writeeflags_u64' was not declared in this scope
In file included from /usr/lib/gcc/x86_64-linux-gnu/4.9/include/immintrin.h:45:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/4.9/include/x86intrin.h:46,
                 from /usr/include/x86_64-linux-gnu/c++/4.9/bits/opt_random.h:33,
                 from /usr/include/c++/4.9/random:50,
                 from /usr/include/c++/4.9/bits/stl_algo.h:66,
                 from /usr/include/c++/4.9/algorithm:62,
                 from /usr/include/boost/function/detail/prologue.hpp:13,
                 from /usr/include/boost/function/function_template.hpp:13,
                 from /usr/include/boost/function/detail/maybe_include.hpp:13,
                 from /usr/include/boost/function/function0.hpp:11,
                 from /usr/include/boost/python/errors.hpp:13,
                 from /usr/include/boost/python/handle.hpp:11,
                 from /usr/include/boost/python/args_fwd.hpp:10,
                 from /usr/include/boost/python/args.hpp:10,
                 from /usr/include/boost/python.hpp:11,
                 from ../../../lib/graph/inc/graph/hooks/output.h:2,
                 from ../../../lib/graph/src/hooks/output.cpp:3:
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/avx512fintrin.h: At global scope:
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/avx512fintrin.h:33:29: error: attribute(target("avx512f")) is unknown
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/avx512fintrin.h:58:33: error: attribute(target("avx512f")) is unknown
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/avx512fintrin.h:70:39: error: attribute(target("avx512f")) is unknown
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/avx512fintrin.h:80:55: error: attribute(target("avx512f")) is unknown
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/avx512fintrin.h:91:51: error: attribute(target("avx512f")) is unknown
In file included from /usr/lib/gcc/x86_64-linux-gnu/4.9/include/immintrin.h:45:0,
                 from /usr/lib/gcc/x86_64-linux-gnu/4.9/include/x86intrin.h:46,
                 from /usr/include/x86_64-linux-gnu/c++/4.9/bits/opt_random.h:33,
                 from /usr/include/c++/4.9/random:50,
                 from /usr/include/c++/4.9/bits/stl_algo.h:66,
                 from /usr/include/c++/4.9/algorithm:62,
                 from /usr/include/boost/function/detail/prologue.hpp:13,
                 from /usr/include/boost/function/function_template.hpp:13,
                 from /usr/include/boost/function/detail/maybe_include.hpp:13,
                 from /usr/include/boost/function/function0.hpp:11,
                 from /usr/include/boost/python/errors.hpp:13,
                 from /usr/include/boost/python/handle.hpp:11,
                 from /usr/include/boost/python/args_fwd.hpp:10,
                 from /usr/include/boost/python/args.hpp:10,
[... 53 lines cut ...]
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/avx512fintrin.h:290:47: error: attribute(target("avx512f")) is unknown
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/avx512fintrin.h: In function '__m512 _mm512_maskz_mov_ps(__mmask16, __m512)':
/usr/lib/gcc/x86_64-linux-gnu/4.9/include/avx512fintrin.h:295:23: error: '__builtin_ia32_movaps512_mask' was not declared in this scope
/
```

This is my setup:

```
> gcc --version
gcc (Ubuntu 4.9.3-5ubuntu1~14.04) 4.9.3
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

> /opt/Qt/5.4/gcc_64/bin/qmake ../sb.pro
```

I installed all packages named in the Installation guidelines.

Could anybody help me on solving this?

Greetings,

Hendrik

mkeeter, 2015-11-19:
This looks like a compiler issue, not an Antimony bug.  Can you check that `g++` is also 4.9?

hhanff, 2015-11-23:
Hi!

Thanks for your reply. 

> g++ --version
g++ (Ubuntu 4.9.3-5ubuntu1~14.04) 4.9.3
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Please let me know if I can check anything else...

Greetings from Bremen,

Hendrik