mkeeter/antimony #144: v2parse errors when trying to build in Ubuntu

rahji, 2015-12-31:
It gets pretty far in the process then fails with this error:

```
g++ -L/usr/lib/python3.4/config-3.4m-x86_64-linux-gnu -L/usr/lib -lpython3.4m -lpthread -ldl -lutil -lm -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-O1 -Wl,-rpath,/usr/local/Qt5.5.1/5.5/gcc_64 -Wl,-rpath,/usr/local/Qt5.5.1/5.5/gcc_64/lib -o Antimony src/graph/node/serializer.o src/graph/node/deserializer.o src/graph/hooks/ui.o src/graph/hooks/hooks.o src/graph/hooks/export.o src/graph/hooks/title.o src/app/main.o src/app/app.o src/ui/main_window.o src/ui/canvas/canvas.o src/ui/canvas/graph_scene.o src/ui/canvas/inspector/inspector.o src/ui/canvas/inspector/inspector_title.o src/ui/canvas/inspector/inspector_text.o src/ui/canvas/inspector/inspector_row.o src/ui/canvas/inspector/inspector_buttons.o src/ui/canvas/inspector/inspector_export.o src/ui/canvas/port.o src/ui/viewport/viewport.o src/ui/viewport/viewport_scene.o src/ui/viewport/view_selector.o src/ui/viewport/depth_image.o src/ui/script/syntax.o src/ui/script/editor.o src/ui/script/script_pane.o src/ui/util/button.o src/ui/util/colors.o src/ui/dialogs/resolution_dialog.o src/ui/dialogs/exporting_dialog.o src/render/render_task.o src/render/render_worker.o src/render/render_proxy.o src/render/render_image.o src/export/export_mesh.o src/export/export_heightmap.o src/control/control.o src/control/control_root.o src/control/proxy.o src/control/point.o src/control/wireframe.o src/app/undo/stack.o src/app/undo/undo_command.o src/app/undo/undo_move.o src/app/undo/undo_change_expr.o src/app/undo/undo_change_script.o src/app/undo/undo_delete_node.o src/app/undo/undo_delete_link.o src/app/undo/undo_delete_multi.o src/app/undo/undo_add_multi.o src/app/undo/undo_add_node.o src/app/undo/undo_add_link.o src/ui/canvas/connection.o usr/local/src/antimony-develop/build/app/qrc_gl.o usr/local/src/antimony-develop/build/app/moc_main_window.o usr/local/src/antimony-develop/build/app/moc_app.o usr/local/src/antimony-develop/build/app/moc_canvas.o usr/local/src/antimony-develop/build/app/moc_graph_scene.o usr/local/src/antimony-develop/build/app/moc_inspector.o usr/local/src/antimony-develop/build/app/moc_inspector_title.o usr/local/src/antimony-develop/build/app/moc_inspector_text.o usr/local/src/antimony-develop/build/app/moc_inspector_row.o usr/local/src/antimony-develop/build/app/moc_inspector_buttons.o usr/local/src/antimony-develop/build/app/moc_inspector_export.o usr/local/src/antimony-develop/build/app/moc_port.o usr/local/src/antimony-develop/build/app/moc_viewport.o usr/local/src/antimony-develop/build/app/moc_viewport_scene.o usr/local/src/antimony-develop/build/app/moc_view_selector.o usr/local/src/antimony-develop/build/app/moc_depth_image.o usr/local/src/antimony-develop/build/app/moc_editor.o usr/local/src/antimony-develop/build/app/moc_script_pane.o usr/local/src/antimony-develop/build/app/moc_button.o usr/local/src/antimony-develop/build/app/moc_resolution_dialog.o usr/local/src/antimony-develop/build/app/moc_render_task.o usr/local/src/antimony-develop/build/app/moc_render_worker.o usr/local/src/antimony-develop/build/app/moc_render_proxy.o usr/local/src/antimony-develop/build/app/moc_render_image.o usr/local/src/antimony-develop/build/app/moc_export_mesh.o usr/local/src/antimony-develop/build/app/moc_export_worker.o usr/local/src/antimony-develop/build/app/moc_export_heightmap.o usr/local/src/antimony-develop/build/app/moc_control.o usr/local/src/antimony-develop/build/app/moc_control_root.o usr/local/src/antimony-develop/build/app/moc_proxy.o usr/local/src/antimony-develop/build/app/moc_connection.o   -L../lib/fab -lSbFab -L../lib/graph -lSbGraph -lpython3.4m -lboost_python-py34 -lpng -L/usr/local/Qt5.5.1/5.5/gcc_64/lib -lQt5OpenGL -L/usr/lib64 -lQt5Widgets -lQt5Gui -lQt5Network -lQt5Core -lGL -lpthread 
../lib/fab/libSbFab.a(v2parser.o): In function `v2parse':
v2parser.cpp:(.text+0x217): undefined reference to `yylex_init(void**)'
v2parser.cpp:(.text+0x224): undefined reference to `yy_scan_string(char const*, void*)'
v2parser.cpp:(.text+0x24c): undefined reference to `yylex(void*)'
v2parser.cpp:(.text+0x258): undefined reference to `yyget_text(void*)'
v2parser.cpp:(.text+0x28e): undefined reference to `yy_delete_buffer(yy_buffer_state*, void*)'
v2parser.cpp:(.text+0x298): undefined reference to `yylex_destroy(void*)'
collect2: error: ld returned 1 exit status
make[1]: *** [Antimony] Error 1
make[1]: Leaving directory `/usr/local/src/antimony-develop/build/app'
make: *** [sub-app-make_first-ordered] Error 2
```
I believe I have all the dependencies. The OS is 3.13.0-74-generic #118-Ubuntu SMP

Any ideas?
Rob

mkeeter, 2015-12-31:
Have you tried just running `make` again??  There may be a weird dependency ordering issue with the generated files...

rahji, 2016-01-01:
I did .. Tried qmake, make, make clean, in various orders :) I ended up downloading the deb binary but I can start again if it will help debug.

Sent from TypeApp

mkeeter, 2016-06-26:
If you have the time, try again with the latest [build instructions](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md#linux) – maybe `cmake` will make your system happier.