mkeeter/antimony #145: antimony not working

kiranskunjumon80, 2016-01-04:
when i type 
<b>./app/antimony</b>
while installing antimony the following things shown up and saying cannot find <b>fab</b> python module
 

<b>Traceback (most recent call last):
  File "/home/ksk95/antimony/build/app/sb/fab/__init__.py", line 1, in <module>
    import fab.shapes as shapes
  File "/home/ksk95/antimony/build/app/sb/fab/shapes.py", line 917, in <module>
    shape = triangle(0, 0, 0.35, 1, 0.1, 0)
  File "/home/ksk95/antimony/build/app/sb/fab/shapes.py", line 136, in triangle
    max(x0, x1, x2), max(y0, y1, y2))
RuntimeError: Could not construct Shape object.</b>

please help me

mkeeter, 2016-01-08:
What version of Python and boost::python are you linking against?

kiranskunjumon80, 2016-01-09:
python 3

mkeeter, 2016-06-26:
Try again with the latest [build instructions](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md#linux) – maybe `cmake` will make your system happier.