mkeeter/antimony #148: Adding Operator

dcellucci, 2016-02-04:
I use Antimony to generate periodic cellular solids, but one of the problems with it for this approach is that it ignores the periodicity of the model, resulting in extremely large equation definitions and, at least on my computer, total crashes for anything larger than 2x2 unit cells.

Would it be possible to add a periodic operator? Maybe just a 'P' binary operator such that PXf such that the remainder of X/f is returned? This would reduce the representation of the lattice to just the unit cell, and then the function bounds would define the overall size of the lattice.

I've tried it with the latest release version (0.9.0c), and i seemed to making good headway, but I could never get it to compile- I kept running into problems with the v2syntax.lemon.cpp file

I guess 1) is this a good idea that no one has tried before and 2) if so, if I were to add a new operator, what changes would I have to make to the lemon file? Is there some sort of generator for this file? it seems pretty generic. 

Thanks, 
DC

mkeeter, 2016-03-14:
Definitely possible – check out how I implemented `OP_MOD` in [Ao](https://github.com/mkeeter/ao/search?utf8=%E2%9C%93&q=OP_MOD).

I don't have much intuition about the syntax file; pinging @fros1y about it.  I'd recommend using `%` and only adding it to the new (infix) syntax mode for simplicity.

fros1y, 2016-03-14:
Yep,

This should be doable.  The syntax (or format) that Antimony uses for its max expressions has to be transformed into a data structure by using a parser.  Parsers can be handwritten (and Antimony's was, initially), but a typical thing to do is to use parser-generator tools to take a description of the syntax and use it to automatically generate code that is used to parse the input and create the right kind of data structure.

If you wanted to add the % operator, you would want to first add it as a "token" into the v2syntax.l file.  This is a description of a "lexer."  As Matt suggests, I'd only do if for the V2 section. Then you'd want to add it into the v2syntax.y file.  That's the file  that is used to automatically build the v2syntax.lemon.cpp file at compile time. Note that the v2syntax.y file is building "nodes," the data structure that Antimony uses.  You'll have to add your own node for %, and update the implementations in node_c.c, opcodes.c, printers*.c, and eval.c.  The actual implementation for how to eval the node will be in the various math_*.c files.  In general, I think you can probably copy and paste heavily from the MUL operator, with the exception of the interval and derivative calculations.