mkeeter/antimony #149: Sine functions

chalker7, 2016-02-10:
I am trying to generate a 3 dimensional straight extrusion with a sine-wave shaped cross section (similar to the individual components seen in this picture: https://c2.staticflickr.com/4/3148/3102467395_8827b28d4b_z.jpg?zz=1) 
Is generating sine waves (preferably with a specified frequency, amplitude and number of periods) something that is possible in Antimony?

mkeeter, 2016-02-11:
Yes, `sin` is a valid function that can be used in math strings.

Here's a simple example:

```py
import fab

title('sine wave demo')

input('f', float, 1)

# Create a sine wave and clip it within a bounding box
sine = (fab.types.Shape('=Y - sin(X * {f});'.format(**locals()))
        & fab.shapes.rectangle(-10, 10, -2, 1))

# Subtract out the same wave, shifted down a little bit
ribbon = sine & ~fab.shapes.move(sine, 0, -1)

output('s', ribbon)
```

...and the output:
[image]

chalker7, 2016-02-11:
Beautiful, thanks so much!

mkeeter, 2016-02-11:
You're welcome!

closed 2016-02-11 by mkeeter