mkeeter/antimony #152: Fedora build - no sub-menu items for the Add menu items

planmac, 2016-03-11:
Only able to add a script - no other items available in the 2D, 3D etc. menu items

mcarlo95, 2016-04-14:
Known bug. Same problem here

imperfectlink, 2016-05-20:
Same problem here. F23 Gnome desktop.

mkeeter, 2016-05-22:
Sounds pretty reproducible – I'll spin up a Fedora VM and see if I can solve this for the upcoming 0.9.1 release.

mkeeter, 2016-05-22:
I'm having trouble reproducing this issue.

I built on Fedora 23, then ran the app from the build folder (i.e. `./app/antimony`), and all of the nodes showed up.  @imperfectlink, how are you invoking the application?

sagitter, 2016-05-23:
Antimony is runned by `antimony %f` command; `Add` menu is totally invisible to me.

mkeeter, 2016-05-24:
Did you build it from source or install it from the Fedora package?

sagitter, 2016-05-24:
> Did you build it from source or install it from the Fedora package?

I'm the Fedora packager of Antimony; latest Antimony (0.9.1c) code has been compiled and packed as RPM.

mkeeter, 2016-05-24:
Thanks for updating the package – I'll install from there and see if I can debug.

sagitter, 2016-05-24:
It's not available online yet. Let me rebuild it for you.

sagitter, 2016-05-24:
> Thanks for updating the package – I'll install from there and see if I can debug.

Please, use Fedora 23 i686 if you can; exists a bug of `boost-python3` on Fedora 23 x86_64.
http://copr-fe.cloud.fedoraproject.org/coprs/sagitter/Ipopt-EPEL/build/316346/

imperfectlink, 2016-05-24:
I'm just calling it up with `antimony` in the terminal. No errors, just odd behaviour.

I installed using `dnf install antimony --enablerepo=updates-testing`

I removed and installed the i686 package from here: [](http://koji.fedoraproject.org/koji/buildinfo?buildID=699437)

Here is the output:
```
QGtkStyle could not resolve GTK. Make sure you have installed the proper libraries.
QGtkStyle could not resolve GTK. Make sure you have installed the proper libraries.
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
Unrecognized OpenGL version
Unrecognized OpenGL version
```

awgrover, 2016-06-25:
Same problem.

mkeeter, 2016-06-25:
I don't have the resources to support prebuilt packages for all the flavors of Linux, but will support folks using the app (on Mac) and building from source (on Linux).

@awgrover, I'm in the process of switching from `qmake` to `cmake`, which will hopefully make the build / bundle / deploy process more robust – I'll ping you when it's ready for testing.

mkeeter, 2016-06-26:
@awgrover / @sagitter 
I've updated everything to use `cmake`, and clarified the [building instructions](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md#linux)

When you build the application, it will copy `sb/nodes` and `sb/fab` into the same folder as `antimony` (usually `Antimony/build/app/`).  This search path is used when `antimony` is called from the build folder.

On startup, the application looks for `sb/nodes`; if this folder doesn't exist, it assumes that it has been installed (to `/usr/local/bin/antimony` with `sudo ninja install`) and looks in `/usr/local/share/antimony/nodes` instead.

(this is all Linux-specific; on Mac, I just put `nodes` and `fab` into `Antimony.app/Resources`)

sagitter, 2016-06-26:
Hi @mkeeter,

>I've updated everything to use cmake, and clarified the building instructions

Are your changes for next release ?

mkeeter, 2016-06-27:
@sagitter 
I don't know how Fedora usually handles packaging (whether they prefer a tagged release or building from `develop` is fine).  If they prefer a specific release number, I'll make a few more updates and push out 0.9.2.

sagitter, 2016-06-28:
> I don't know how Fedora usually handles packaging (whether they prefer a tagged release or building from develop is fine). If they prefer a specific release number, I'll make a few more updates and push out 0.9.2.

I think, it is a modification sufficiently important to justify a new release; of course, it is not fundamental for Fedora packaging.

sagitter, 2016-07-04:
`Add` menu is still unvisible unfortunately.
`antimony 0.9.2` files installed (Fedora):

```
/usr/bin/antimony
/usr/lib/python3.5/site-packages/fab
/usr/lib/python3.5/site-packages/fab/__init__.py
/usr/lib/python3.5/site-packages/fab/__pycache__
/usr/lib/python3.5/site-packages/fab/__pycache__/__init__.cpython-35.opt-1.pyc
/usr/lib/python3.5/site-packages/fab/__pycache__/__init__.cpython-35.pyc
/usr/lib/python3.5/site-packages/fab/__pycache__/shapes.cpython-35.opt-1.pyc
/usr/lib/python3.5/site-packages/fab/__pycache__/shapes.cpython-35.pyc
/usr/lib/python3.5/site-packages/fab/__pycache__/types.cpython-35.opt-1.pyc
/usr/lib/python3.5/site-packages/fab/__pycache__/types.cpython-35.pyc
/usr/lib/python3.5/site-packages/fab/shapes.py
/usr/lib/python3.5/site-packages/fab/types.py
/usr/share/antimony
/usr/share/antimony/nodes
/usr/share/antimony/nodes/2D
/usr/share/antimony/nodes/2D → 3D
/usr/share/antimony/nodes/2D → 3D/Revolve
/usr/share/antimony/nodes/2D → 3D/Revolve/revolve_xy_x.node
/usr/share/antimony/nodes/2D → 3D/Revolve/revolve_xy_y.node
/usr/share/antimony/nodes/2D → 3D/extrude.node
/usr/share/antimony/nodes/2D → 3D/loft.node
/usr/share/antimony/nodes/2D/Circles
/usr/share/antimony/nodes/2D/Circles/circle_center.node
/usr/share/antimony/nodes/2D/Circles/circle_edge.node
/usr/share/antimony/nodes/2D/Rectangles
/usr/share/antimony/nodes/2D/Rectangles/rectangle_center.node
/usr/share/antimony/nodes/2D/Rectangles/rectangle_center_rounded.node
/usr/share/antimony/nodes/2D/Rectangles/rectangle_corner.node
/usr/share/antimony/nodes/2D/Rectangles/rectangle_corner_rounded.node
/usr/share/antimony/nodes/2D/Triangles
/usr/share/antimony/nodes/2D/Triangles/triangle.node
/usr/share/antimony/nodes/2D/Triangles/triangle_right.node
/usr/share/antimony/nodes/2D/point.node
/usr/share/antimony/nodes/2D/polygon.node
/usr/share/antimony/nodes/2D/text.node
/usr/share/antimony/nodes/3D
/usr/share/antimony/nodes/3D → 2D
/usr/share/antimony/nodes/3D → 2D/slice_xy.node
/usr/share/antimony/nodes/3D → 2D/slice_xz.node
/usr/share/antimony/nodes/3D → 2D/slice_yz.node
/usr/share/antimony/nodes/3D/Cones
/usr/share/antimony/nodes/3D/Cones/cone.node
/usr/share/antimony/nodes/3D/Cones/cone_edge.node
/usr/share/antimony/nodes/3D/Cones/cone_height.node
/usr/share/antimony/nodes/3D/Cubes
/usr/share/antimony/nodes/3D/Cubes/cube_center.node
/usr/share/antimony/nodes/3D/Cubes/cube_center_rounded.node
/usr/share/antimony/nodes/3D/Cubes/cube_corner.node
/usr/share/antimony/nodes/3D/Cubes/cube_corner_rounded.node
/usr/share/antimony/nodes/3D/Cylinders
/usr/share/antimony/nodes/3D/Cylinders/cylinder_x.node
/usr/share/antimony/nodes/3D/Cylinders/cylinder_y.node
/usr/share/antimony/nodes/3D/Cylinders/cylinder_z.node
/usr/share/antimony/nodes/3D/Pyramids
/usr/share/antimony/nodes/3D/Pyramids/pyramid_center.node
/usr/share/antimony/nodes/3D/Pyramids/pyramid_edge.node
/usr/share/antimony/nodes/3D/Spheres
/usr/share/antimony/nodes/3D/Spheres/sphere_center.node
/usr/share/antimony/nodes/3D/Spheres/sphere_edge.node
/usr/share/antimony/nodes/3D/Toruses
/usr/share/antimony/nodes/3D/Toruses/torus_x.node
/usr/share/antimony/nodes/3D/Toruses/torus_y.node
/usr/share/antimony/nodes/3D/Toruses/torus_z.node
/usr/share/antimony/nodes/3D/Volumes
/usr/share/antimony/nodes/3D/Volumes/gyroid.node
/usr/share/antimony/nodes/3D/Volumes/schwarz_d.node
[... 52 lines cut ...]
/usr/share/antimony/nodes/Scale/Taper/taper_xy_z.node
/usr/share/antimony/nodes/Scale/scale_cos_x_y.node
/usr/share/antimony/nodes/Scale/scale_cos_xy_z.node
/usr/share/antimony/nodes/Scale/scale_x.node
/usr/share/antimony/nodes/Scale/scale_xy.node
/usr/share/antimony/nodes/Scale/scale_xyz.node
/usr/share/antimony/nodes/Scale/scale_y.node
/usr/share/antimony/nodes/Scale/scale_z.node
/usr/share/antimony/nodes/Scale/scale_z_r.node
/usr/share/antimony/nodes/Translate
/usr/share/antimony/nodes/Translate/origin_xy.node
/usr/share/antimony/nodes/Translate/origin_xyz.node
/usr/share/antimony/nodes/Translate/recenter.node
/usr/share/antimony/nodes/Translate/translate.node
/usr/share/appdata/antimony.appdata.xml
/usr/share/applications/antimony.desktop
/usr/share/doc/antimony
/usr/share/doc/antimony/README.md
/usr/share/doc/antimony/SCRIPTING.md
/usr/share/doc/antimony/USAGE.md
/usr/share/icons/antimony
/usr/share/icons/antimony/antimony.svg
/usr/share/licenses/antimony
/usr/share/licenses/antimony/LICENSE
```

a2batic, 2016-07-06:
Cant open `Add` menu
[image]

a2batic, 2016-07-06:
N
[image]
o other items available in the 2D, 3D, 2D->3D, 3D->2D and CSG

mkeeter, 2016-07-06:
Okay, the `Add` menu wasn't even *trying* to populate itself – should be fixed now in the `release/0.9.2b` branch (to be merged into `develop` after the release goes out)

@sagitter 
The root cause of the issue is that Fedora likes to put things in `/usr/{bin,share}`, while other distros put things in `/usr/local/{bin,share}`.  Even Qt's application search paths don't include for `/usr/share`, so I've added a special-cased path of `../share/antimony/nodes` relative to the executable.

Can you pull the latest `release/0.9.2b` commit (`b9f01e14`) and see if that works?
If not, the list of files was very helpful for diagnosing the issue.

sagitter, 2016-07-07:
>@sagitter
The root cause of the issue is that Fedora likes to put things in /usr/{bin,share}, while other distros put things in /usr/local/{bin,share}. Even Qt's application search paths don't include for /usr/share, so I've added a special-cased path of ../share/antimony/nodes relative to the executable.

>Can you pull the latest release/0.9.2b commit (b9f01e14) and see if that works?
If not, the list of files was very helpful for diagnosing the issue.

`Add` menu is now visible except `2D -> 3D` and `3D -> 2D` sub-menus.

`antimony`'s installed files are:

```
/usr/bin/antimony
/usr/lib/python3.5/site-packages/fab
/usr/lib/python3.5/site-packages/fab/__init__.py
/usr/lib/python3.5/site-packages/fab/__pycache__
/usr/lib/python3.5/site-packages/fab/__pycache__/__init__.cpython-35.opt-1.pyc
/usr/lib/python3.5/site-packages/fab/__pycache__/__init__.cpython-35.pyc
/usr/lib/python3.5/site-packages/fab/__pycache__/shapes.cpython-35.opt-1.pyc
/usr/lib/python3.5/site-packages/fab/__pycache__/shapes.cpython-35.pyc
/usr/lib/python3.5/site-packages/fab/__pycache__/types.cpython-35.opt-1.pyc
/usr/lib/python3.5/site-packages/fab/__pycache__/types.cpython-35.pyc
/usr/lib/python3.5/site-packages/fab/shapes.py
/usr/lib/python3.5/site-packages/fab/types.py
/usr/share/antimony
/usr/share/antimony/nodes
/usr/share/antimony/nodes/2D
/usr/share/antimony/nodes/2D #U2192 3D
/usr/share/antimony/nodes/2D #U2192 3D/Revolve
/usr/share/antimony/nodes/2D #U2192 3D/Revolve/revolve_xy_x.node
/usr/share/antimony/nodes/2D #U2192 3D/Revolve/revolve_xy_y.node
/usr/share/antimony/nodes/2D #U2192 3D/extrude.node
/usr/share/antimony/nodes/2D #U2192 3D/loft.node
/usr/share/antimony/nodes/2D/Circles
/usr/share/antimony/nodes/2D/Circles/circle_center.node
/usr/share/antimony/nodes/2D/Circles/circle_edge.node
/usr/share/antimony/nodes/2D/Rectangles
/usr/share/antimony/nodes/2D/Rectangles/rectangle_center.node
/usr/share/antimony/nodes/2D/Rectangles/rectangle_center_rounded.node
/usr/share/antimony/nodes/2D/Rectangles/rectangle_corner.node
/usr/share/antimony/nodes/2D/Rectangles/rectangle_corner_rounded.node
/usr/share/antimony/nodes/2D/Triangles
/usr/share/antimony/nodes/2D/Triangles/triangle.node
/usr/share/antimony/nodes/2D/Triangles/triangle_right.node
/usr/share/antimony/nodes/2D/point.node
/usr/share/antimony/nodes/2D/polygon.node
/usr/share/antimony/nodes/2D/text.node
/usr/share/antimony/nodes/3D
/usr/share/antimony/nodes/3D #U2192 2D
/usr/share/antimony/nodes/3D #U2192 2D/slice_xy.node
/usr/share/antimony/nodes/3D #U2192 2D/slice_xz.node
/usr/share/antimony/nodes/3D #U2192 2D/slice_yz.node
/usr/share/antimony/nodes/3D/Cones
/usr/share/antimony/nodes/3D/Cones/cone.node
/usr/share/antimony/nodes/3D/Cones/cone_edge.node
/usr/share/antimony/nodes/3D/Cones/cone_height.node
/usr/share/antimony/nodes/3D/Cubes
/usr/share/antimony/nodes/3D/Cubes/cube_center.node
/usr/share/antimony/nodes/3D/Cubes/cube_center_rounded.node
/usr/share/antimony/nodes/3D/Cubes/cube_corner.node
/usr/share/antimony/nodes/3D/Cubes/cube_corner_rounded.node
/usr/share/antimony/nodes/3D/Cylinders
/usr/share/antimony/nodes/3D/Cylinders/cylinder_x.node
/usr/share/antimony/nodes/3D/Cylinders/cylinder_y.node
/usr/share/antimony/nodes/3D/Cylinders/cylinder_z.node
/usr/share/antimony/nodes/3D/Pyramids
/usr/share/antimony/nodes/3D/Pyramids/pyramid_center.node
/usr/share/antimony/nodes/3D/Pyramids/pyramid_edge.node
/usr/share/antimony/nodes/3D/Spheres
/usr/share/antimony/nodes/3D/Spheres/sphere_center.node
/usr/share/antimony/nodes/3D/Spheres/sphere_edge.node
[... 59 lines cut ...]
/usr/share/antimony/nodes/Scale/Taper/taper_xy_z.node
/usr/share/antimony/nodes/Scale/scale_cos_x_y.node
/usr/share/antimony/nodes/Scale/scale_cos_xy_z.node
/usr/share/antimony/nodes/Scale/scale_x.node
/usr/share/antimony/nodes/Scale/scale_xy.node
/usr/share/antimony/nodes/Scale/scale_xyz.node
/usr/share/antimony/nodes/Scale/scale_y.node
/usr/share/antimony/nodes/Scale/scale_z.node
/usr/share/antimony/nodes/Scale/scale_z_r.node
/usr/share/antimony/nodes/Translate
/usr/share/antimony/nodes/Translate/origin_xy.node
/usr/share/antimony/nodes/Translate/origin_xyz.node
/usr/share/antimony/nodes/Translate/recenter.node
/usr/share/antimony/nodes/Translate/translate.node
/usr/share/appdata/antimony.appdata.xml
/usr/share/applications/antimony.desktop
/usr/share/doc/antimony
/usr/share/doc/antimony/README.md
/usr/share/doc/antimony/SCRIPTING.md
/usr/share/doc/antimony/USAGE.md
/usr/share/icons/antimony
/usr/share/icons/antimony/antimony.svg
/usr/share/licenses/antimony
/usr/share/licenses/antimony/LICENSE
```

mkeeter, 2016-07-09:
Definitely looks like a unicode issue.  Why your previous install put nodes in
```
/usr/share/antimony/nodes/2D → 3D
```
while this install chose
```
/usr/share/antimony/nodes/2D #U2192 3D
```

sagitter, 2016-07-09:
Can I set `unicode` during build ?

mkeeter, 2016-07-09:
Yes?  I don't know what that question means.

I'm confused because [this post](https://github.com/mkeeter/antimony/issues/152#issuecomment-230320758) handled unicode paths correctly while [this post](https://github.com/mkeeter/antimony/issues/152#issuecomment-231022205) did not.

sagitter, 2016-07-09:
>Yes? I don't know what that question means.

I meant if there is a variable that manages `unicode`.

>I'm confused because this post handled unicode paths correctly while this post did not.

They are `0.9.2` build vs `0.9.2b` build.

awgrover, 2016-07-09:
I'm still having problems:

1. Need python3.3 which Ubuntu 14.04 doesn't have. The error is
```
Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
(Required is at least version "3.3")
```
Install from fkrull/deadsnakes
```
apt-get install -y python-software-properties
add-apt-repository -y ppa:fkrull/deadsnakes
apt-get update
apt-get install -y python3.3 python3.3-dev
```
That seems to work.

2. Compile errors complaining about std=c++11
```
/usr/include/c++/4.9/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
```
I did the quick and dirty addition to CMakeLists.txt:
`add_compile_options(-std=c++11)`
Which seems to work (though I get warnings for "c" compiles.

3. Compile errors complaining about std=c11
`../lib/fab/src/formats/stl.c:14:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode`
I did the quick and dirty addition to CMakeLists.txt:
`add_compile_options(-std=c11)`
Which seems to work (though I get warnings for "c++" compiles.

4. And a compile still fail:
```
/usr/bin/c++   -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -Wall -Wextra -g -Werror=switch -O3 -DRELEASE -fPIE -Iapp -I../app -isystem /usr/include/python3.3m -isystem /usr/include/x86_64-linux-gnu/python3.3m -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -I/usr/include/qt5/QtOpenGL -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtConcurrent -I../lib/graph/inc -I../lib/fab/inc    -D'GITREV="'b9f01e1'+ "' -D'GITTAG=""' -D'GITBRANCH="release/0.9.2b "' -std=c++11 -std=c11 -MMD -MT app/CMakeFiles/antimony.dir/app/app.cpp.o -MF "app/CMakeFiles/antimony.dir/app/app.cpp.o.d" -o app/CMakeFiles/antimony.dir/app/app.cpp.o -c ../app/app/app.cpp
../app/app/app.cpp: In member function ‘QStringList App::nodePaths() const’:
../app/app/app.cpp:67:13: error: ‘AppDataLocation’ is not a member of ‘QStandardPaths’
             QStandardPaths::AppDataLocation))
             ^
```
Here's the output of cmake:
```
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.3m.so (found suitable version "3.3.6", minimum required is "3.3") 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.2.50") 
-- Found FLEX: /usr/bin/flex (found version "2.5.35") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/awgrover/dev/thirdparty/antimony2/build
```
python3.3 -> Python 3.3.6
c++ -> c++ (Ubuntu 4.9.3-8ubuntu2~14.04) 4.9.3

sagitter, 2016-07-10:
@a2batic @mkeeter 

This problem should be fixed with [release `antimony-0.9.2-0.3`](http://koji.fedoraproject.org/koji/packageinfo?packageID=21348).

Please, leave a feedback on Bodhi: https://bodhi.fedoraproject.org/updates/FEDORA-2016-6cc0e59225

mkeeter, 2016-07-10:
I'm going to close this issue since @sagitter 's latest release fixes the titular problem.

@awgrover , I've opened up a [new issue](https://github.com/mkeeter/antimony/issues/164) with your problem.

closed 2016-07-10 by mkeeter