mkeeter/antimony #153: [Feature request] STL import

jflesch, 2016-03-15:
A very neat feature would be the ability to import STL model as a node in Antimony.

Example of use case :
I'm currently designing a casing that will contain many PCBs (of various sizes). Kicad can provides me with VRML versions of the PCB. I can convert those VRML files to STL files.
Importing those STL files in my Antimony model would allow me to make sure that everything fits correctly.

jflesch, 2016-03-15:
By the way, thank you for Antimony. It is purely awesome and I'm really looking forward to the first non-beta version ! :)

astronouth7303, 2016-03-15:
Unfortunately, Antimony represents models not as polygons or shapes, but as advanced mathematics. You build up a formula that describes the shape you want.

This makes importing STL (a polygon format) very difficult because you have to fit a mathematical model to the polygon.

TLC123, 2016-03-15:
Given a watertight and manifold mesh, the mathematical model could be as simple as a inside/outside test of the imported mesh at each given point or more tediously complemented by the MIN of all distance functions for triangles. to achive a signed distance function. As described here http://goo.gl/qxAUxl

mkeeter, 2016-03-15:
@jflesch There's a simple (but not trivial) solution to your *specific* problem, which is adding a node that displays a mesh in the 3D viewport (but doesn't let you use it in design / CSG).  This would let you make sure things fit, but is very limited.

The broader problem of doing CSG with meshes is hard.  I think the way forward is to generalize f-reps into an oracle interface, then make a mesh import that exposes that same API.  The oracle should let you query inside / outside for points *and* interval regions.  @TLC123, it's the interval part that's tricky for meshes.

TLC123, 2016-08-01:
What are interval regions? I have not come across the term before and its to common words in other context to google.
 If its the distans field i belive some version of binaryspace partion  BSP 
can solve that for arbitrary polygonal objects 
http://www.hyperfun.org/TRP/FPA_BSPfields_CAD11.pdf
apparently manifold-ness is not even an absolute http://run.usc.edu/signedDistanceField/XuBarbicSignedDistanceField2014.pdf !

doug-moen, 2017-02-11:
@TLC123, re: the papers you linked to.

*BSP-fields: An exact representation of polygonal objects by differentiable scalar
fields based on binary space partitioning*
This looks like it is too slow, both the data structure build times and the distance function evaluation times (see table 2) and can't handle polygon soup.

*Signed Distance Fields for Polygon Soup Meshes*
This can handle polygon soup (good). The reported data structure build times are even slower than the first paper. The result is an octree, which should be fast to evaluate (good). This is only an approximation to the original polyhedron, not an exact representation. They are using a 1000^3 octree which consumes almost 32 Gig of memory by itself; a smaller octree would be less accurate but I assume would build faster.

The ASDF data structure used by Antimony is supposed to be more economical of space than an octree, and does the same job.

I speculate that maybe you could use the method of "SDF for Polygon Soup" to convert an STL file to an ASDF file, as an expensive offline operation, then provide a way for Antimony to load the ASDF file.

I'm not an expert in this area; I hope somebody comes up with a more practical algorithm and data structure for solving this problem. F-Rep is awesome, we just need a way to import arbitrary STL files.

TLC123, 2017-02-11:
F-Rep is awesome and many project share this problem-space. 
Some oriented to  3d printing, some pure graphical
Solution is bound to emerge.
Im curently watching developments in:
Antimony
ImplicitCAD
IceSL
GLSL ray-marching (very powerful for future VR apps, gaze selective resolution)
OpenScad to some extent.