mkeeter/antimony #158: Open Source CAD exchange format.

Evil-Spirit, 2016-05-24:
We are working hard on SolveSpace. SolveSpace use Mesh/NURBS for geometry, but has history-based approaches.
You are working on a good tool which is really not like somthing else. You have used F-Rep for internal things. But you have also some thing like history based approach.
There is also OpenSCAD which uses pixel operations for booleans as fast visualization, and CGAL as boolean operations for generating some STL meshes.
There is also FreeCAD which is working on NURBS and has history-based approaches.

So, what do you think about making the ultimate exchange format for history-based geometry. This format must be not snapped to interal format (mesh, b-rep, f-rep, etc), just geometry with building process. It must operate on 2d primitives, 2d/3d sketches of primitives and operations (extrusion, revolve, etc) with combination function (union, subtract, etc).

My first draft:
```xml
<cad>
 <sketches>
  <sketch type="2d" id="1">
   <contour id="1" combine="add">
    <start point="0 0"/>
    <line point="0 10"/>
    <line point="10 10"/>
    <line/><!-- close contour -->
   </contour>
   <!-- circle is contour itself -->
   <circle id="2" op="sub" center="5 5" radius="1">
  </sketch>
 </sketches>
 
 <bodies>
  <body id="1">
   <feature id="1" op="extrude" combine="add">
    <sketch id="1" transform="mtx"/>
   </feature>
  </body>
 </bodies>
 
</cad>
```

We can use some general approaches from dxf and svg.

Can you think about it?

Evil-Spirit, 2016-05-25:
IRC: #make-cad-format

mkeeter, 2016-05-25:
(I'll reply on the [OpenSCAD issue](https://github.com/openscad/openscad/issues/1650), since that's the most active discussion)