mkeeter/antimony #159: [Question] Runtime interpreter or code generator?

rraallvv, 2016-05-25:
All visual programming tools out there fall into two categories:

**Runtime interpreter**
The ones that parse the graph to create nodes that are added to the interpreter at runtime and call the nodes through callbacks, propagating the data flow between nodes through a queue or messaging system according to their connections.

**Code generator**
Those that transpile the graph into source code, with all the nodes converted into procedures or functions. The nodes are called in the oder that they are connected in the graph. The data is passed around in variables through arguments to those function calls.

What system does Antimony implement? I’m curios because I haven’t found yet a FOSS project that works with a code generator.

The advantage is that graphs that are converted into source code can run faster.

mkeeter, 2016-05-25:
Antimony does run-time interpretation of the graph, detecting changes to datums and scripts then re-evaluating them and anything that uses them.

I'm not sure how code generation interacts with partial evaluation (where only nodes downstream of an edited node need to be re-evaluated) – do you have any insight into how that's usually done?

Evil-Spirit, 2016-05-25:
rraallvv,
Take a look at https://github.com/Evil-Spirit/AnandamideAPI.
Now implemented as just nodes, but there is code generator (not licensed under GPL yet).

Evil-Spirit, 2016-05-25:
@rraallvv,
Take a look at https://github.com/Evil-Spirit/AnandamideAPI.
Now implemented as just nodes, but there is code generator (not licensed under GPL yet).

rraallvv, 2016-05-26:
@mkeeter, I’m not sure how it’s done for code generators, the closest I’ve seen is the Visual Machine <https://www.youtube.com/user/MaadSkillzUK?spfreload=10>, which is powered by LLVM. I guessing that the graph is compiled and mapped to some kind of abstract syntax tree, then only those parts of the graph that have changed are compiled again, although that is pure speculation from my part. Regarding Antimony, I asked because I'd love to have something like that but that transpiles to source code.

@Evil-Spirit, thanks for the info.

closed 2016-05-26 by rraallvv

mkeeter, 2016-05-26:
Whoa, thanks for the pointer to Visual Machine – that's awesome.

rraallvv, 2016-05-26:
@mkeeter, my bad, Visual Machine isn't the name of the project, it’s actually Vision Machine ;-)