mkeeter/antimony #160: Why not using gmio to load/save STL meshes?

HuguesDelorme, 2016-06-27:
Hello,
  I've just released the first version of gmio.
  **gmio is a free and open-source C library** providing I/O support for the STL file format(more CAD formats in next releases).
- it's **fast**: gmio is x18 times faster to write binary STL than Assimp, see [benchmarks](https://github.com/fougue/gmio/wiki/4.-Benchmarks)
- it's **feature-rich**: ascii/binary, little/big endian, format probing, infos retrieval, ... see [gmio_stl](http://www.fougue.pro/docs/gmio/0.1/group__gmio__stl.html#details)
- it's **reusable**: based on streams, seamless integration with other 3rd party libs, see [gmio_support](http://www.fougue.pro/docs/gmio/0.1/group__gmio__support.html#details)
- it's **portable**: C90 conformant but it can use recent C features if available

**Antimony could benefit better STL format support with gmio**, just try it!
`>>` https://github.com/fougue/gmio
`>>` [v0.1.0 ZIP archive](https://github.com/fougue/gmio/archive/v0.1.0.zip) or `git clone https://github.com/fougue/gmio.git`
`>>` [Build instructions](https://github.com/fougue/gmio/wiki/2.-Build-instructions)
`>>` [Examples of use](https://github.com/fougue/gmio/wiki/3.-Getting-started)

PS: Antimony seems to be a great project, I hope to try it soon!

astronouth7303, 2016-06-27:
Great pitch. Workshop that?

mkeeter, 2016-07-03:
Thanks for the link to `gmio` – it looks like a neat project.  Antimony does very little with meshes, so it doesn't make sense to add a dependency to replace a [33 line-function](https://github.com/mkeeter/antimony/blob/develop/lib/fab/src/formats/stl.c), but I'll keep an eye on `gmio` in case my mesh-handling needs change in the future.

closed 2016-07-03 by mkeeter

HuguesDelorme, 2016-07-04:
Hello Matt,

Here are my resulting benchmarks that compares Antimony's `save_stl()` against gmio :

File : [Final_Eiffel_Tower.stl](https://grabcad.com/library/eiffel-tower/files/Final%20Eiffel%20Tower.STL)
Format: Binary STL (little-endian)
Count of triangles: 174762
Size: 8.33MB 

                 | gmio | Antimony | ratio
-----------------|------|----------|-------
read             | 4ms  | N/A      | N/A  
write(binary/le) | 4ms  | 206ms    | 51.50
write(binary/be) | 5ms  | N/A      | N/A  

Source code of this benchmark is [here](http://pastebin.com/ZVV5LQr4).

Notice that gmio can export to STL ascii and binary little/big endian(see [gmio_stl_write()](http://www.fougue.pro/docs/gmio/0.2/group__gmio__stl.html#ga38b85c3a6b77a4fbbd617143e799490e) function).

There is no problem to embed the source code of `gmio `within the git repository of Antimony(as this is already the case for Eigen in `lib/fab/vendor` I guess)