mkeeter/antimony #161: Readme file: Installation on Fedora uncorrect

sagitter, 2016-06-28:
Hi,

[here](https://github.com/mkeeter/antimony/blob/develop/README.md#try-it) I read:

>There is also an unofficial, unsupported package for Fedora 22/23:
dnf install antimony --enablerepo=updates-testing

`antimony` is **officially** packed for Fedora >= 22.
`--enablerepo=updates-testing` option is to install package in testing; generally antimony can be installed by:

`dnf install antimony`

mkeeter, 2016-06-28:
To clarify, is the line `dnf install antimony --enablerepo=updates-testing` not necessary?

sagitter, 2016-06-28:
To clarify:

for stable packaging release:
`dnf install antimony`

for packaging release in testing:
`dnf install antimony --enablerepo=updates-testing`

All releases in testing will be pushed on stable when software package works fine.