mkeeter/antimony #162: [antimony-0.9.2]  'yyTokenName' was not declared in this scope

sagitter, 2016-07-04:
Hi.

Please, take a look to this error (Fedora 24):

```
/home/sagitter/rpmbuild/BUILD/antimony-0.9.2/lib/fab/src/tree/v2syntax.c: In function 'void yy_destructor(yyParser*, unsigned char, YYMINORTYPE*)':
/home/sagitter/rpmbuild/BUILD/antimony-0.9.2/lib/fab/src/tree/v2syntax.c:117:31: warning: unused variable 'environment' [-Wunused-variable]
/home/sagitter/rpmbuild/BUILD/antimony-0.9.2/lib/fab/src/tree/v2syntax.c:903:3: note: in expansion of macro 'v2ParseARG_FETCH'
/home/sagitter/rpmbuild/BUILD/antimony-0.9.2/lib/fab/src/tree/v2syntax.y: In function 'void yy_syntax_error(yyParser*, int, const char*)':
/home/sagitter/rpmbuild/BUILD/antimony-0.9.2/lib/fab/src/tree/v2syntax.y:40:17: error: 'yyTokenName' was not declared in this scope
  int n = sizeof(yyTokenName) / sizeof(yyTokenName[0]);
                 ^~~~~~~~~~~
lib/fab/CMakeFiles/SbFab.dir/build.make:681: recipe for target 'lib/fab/CMakeFiles/SbFab.dir/v2syntax.lemon.cpp.o' failed

```

Build log: https://paste.fedoraproject.org/387766/

sagitter, 2016-07-04:
> Please, take a look to this error (Fedora 24):

```
/home/sagitter/rpmbuild/BUILD/antimony-0.9.2/lib/fab/src/tree/v2syntax.c: In function 'void yy_destructor(yyParser*, unsigned char, YYMINORTYPE*)':
/home/sagitter/rpmbuild/BUILD/antimony-0.9.2/lib/fab/src/tree/v2syntax.c:117:31: warning: unused variable 'environment' [-Wunused-variable]
/home/sagitter/rpmbuild/BUILD/antimony-0.9.2/lib/fab/src/tree/v2syntax.c:903:3: note: in expansion of macro 'v2ParseARG_FETCH'
/home/sagitter/rpmbuild/BUILD/antimony-0.9.2/lib/fab/src/tree/v2syntax.y: In function 'void yy_syntax_error(yyParser*, int, const char*)':
/home/sagitter/rpmbuild/BUILD/antimony-0.9.2/lib/fab/src/tree/v2syntax.y:40:17: error: 'yyTokenName' was not declared in this scope
  int n = sizeof(yyTokenName) / sizeof(yyTokenName[0]);
                 ^~~~~~~~~~~
lib/fab/CMakeFiles/SbFab.dir/build.make:681: recipe for target 'lib/fab/CMakeFiles/SbFab.dir/v2syntax.lemon.cpp.o' failed
```

Only default flags permit to compile `antimony` correctly without above error. However, I need to set additional compiler/linker flags according to the Fedora rules. Can you say me what causes that error ?

mkeeter, 2016-07-04:
Ah, I see the problem – there's a bug in the grammer that fails with `NDEBUG` is defined.

(`yyTokenName` is only present if `NDEBUG` isn't defined, but there's a reference to it outside of a guarded block).

Until I have time to fix the grammar, can you adjust the Fedora build settings so that `NDEBUG` isn't defined?

mkeeter, 2016-07-04:
@sagitter Okay, the grammar should be fixed now.  Try building from [`release/0.9.2b`](https://github.com/mkeeter/antimony/tree/release/0.9.2b) and see if that works.

I'm not sure what you mean about `CMAKE_PREFIX_PATH` – according to [the docs](https://cmake.org/cmake/help/v3.0/variable/CMAKE_PREFIX_PATH.html), it's

> Path used for searching by FIND_XXX(), with appropriate suffixes added.

(and doesn't have anything to do with install prefixes)

If there's a different variable that I should be using for install prefixes, let me know and I'll make the appropriate fixes.

sagitter, 2016-07-04:
>Until I have time to fix the grammar, can you adjust the Fedora build settings so that NDEBUG isn't defined?

Yes. It's working now.

>Antimony files are always installed in /usr/local, CMAKE_PREFIX_PATH options looks ignored.

This little change permits a right installation (for me):

`sed -e 's|/usr/local|/usr|g' -i app/CMakeLists.txt`

sagitter, 2016-07-04:
>If there's a different variable that I should be using for install prefixes, let me know and I'll make the appropriate fixes.

For `antimony` could be useful `CMAKE_INSTALL_PREFIX` and `SHARE_INSTALL_PREFIX` variables.

sagitter, 2016-07-04:
>@sagitter Okay, the grammar should be fixed now. Try building from release/0.9.2b and see if that works.

Fixed now.

mkeeter, 2016-07-06:
@sagitter I've also fixed the `CMAKE_INSTALL_PREFIX` issue.
I'm going to close this issue and post a reply in #152 about the missing menu items.

closed 2016-07-06 by mkeeter