mkeeter/antimony #164: Problems building on Ubuntu 14.04

mkeeter, 2016-07-10:
Copied from @awgrover 's comment [here](https://github.com/mkeeter/antimony/issues/152#issuecomment-231560916)

--------------------------------------------------------------------------------------------------------------

I'm still having problems:
checked out on release/0.9.2b @ b9f01e14d2f9000e2c74361ffd9bf19c0b042616

- Need python3.3 which Ubuntu 14.04 doesn't have. The error is

```
Could NOT find PythonLibs (missing: PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS)
(Required is at least version "3.3")
```
Install from fkrull/deadsnakes
```
apt-get install -y python-software-properties
add-apt-repository -y ppa:fkrull/deadsnakes
apt-get update
apt-get install -y python3.3 python3.3-dev
```
That seems to work. Should you specify "apt-get install python3.3 python3.3-dev" ?

- Compile errors complaining about std=c++11
```
/usr/include/c++/4.9/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
```
I did the quick and dirty addition to CMakeLists.txt:
`add_compile_options(-std=c++11)`
Which seems to work (though I get warnings for "c" compiles.

- Compile errors complaining about std=c11
`../lib/fab/src/formats/stl.c:14:5: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode`
I did the quick and dirty addition to CMakeLists.txt:
`add_compile_options(-std=c11)`
Which seems to work (though I get warnings for "c++" compiles.

- Compile error because I have < Qt 5.4 (ubuntu 14.04 has 5.2.1)
```
/usr/bin/c++   -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -Wall -Wextra -g -Werror=switch -O3 -DRELEASE -fPIE -Iapp -I../app -isystem /usr/include/python3.3m -isystem /usr/include/x86_64-linux-gnu/python3.3m -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -I/usr/include/qt5/QtOpenGL -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtConcurrent -I../lib/graph/inc -I../lib/fab/inc    -D'GITREV="'b9f01e1'+ "' -D'GITTAG=""' -D'GITBRANCH="release/0.9.2b "' -std=c++11 -std=c11 -MMD -MT app/CMakeFiles/antimony.dir/app/app.cpp.o -MF "app/CMakeFiles/antimony.dir/app/app.cpp.o.d" -o app/CMakeFiles/antimony.dir/app/app.cpp.o -c ../app/app/app.cpp
../app/app/app.cpp: In member function ‘QStringList App::nodePaths() const’:
../app/app/app.cpp:67:13: error: ‘AppDataLocation’ is not a member of ‘QStandardPaths’
             QStandardPaths::AppDataLocation))
             ^
```
I see some messages about "QStandardPaths::AppDataLocation ... was introduced in Qt 5.4"

@mkeeter:
**Do you know how I tell CMake that I have Qt5.4 in a specific directory (where I built it, but did not install it in the system directories)?**

Here's the output of cmake:
```
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.3m.so (found suitable version "3.3.6", minimum required is "3.3") 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.2.50") 
-- Found FLEX: /usr/bin/flex (found version "2.5.35") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/awgrover/dev/thirdparty/antimony2/build
```
Some versions:
```
python3.3 -> Python 3.3.6
(ubuntu 14.04 has Python 3.4.3)
c++ -> c++ (Ubuntu 4.9.3-8ubuntu2~14.04) 4.9.3
cmake version 2.8.12.2
```

mkeeter, 2016-07-10:
In general, now that LTS 16.04 is out, I'll be putting less effort into supporting 14.04.

You can try to keep bringing in packages from the future, but you may need a non-default Qt, `gcc`, and perhaps even `cmake` (all of which work fine by default in 16.04; see 9f99631ccbd4d5d2023fb33a83c6679c9db4d48e which removed 14.04 special-cases).

--------------------------------------------------------------------------------
>Do you know how I tell CMake that I have Qt5.4 in a specific directory (where I built it, but did not install it in the system directories)?

In the Mac build instructions, we point `cmake` to a particular Qt folder:
```
cmake -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt5/5.6.1 -GNinja ..
```
Can you see if that works for you?

--------------------------------------------------------------------------------

I think that the C++11 and C99 warnings may be because you have an older version of `cmake` that doesn't support the `CXX_STANDARD` property.

I've pushed a commit to `develop` (9a3b127a7a91230db4c62142586d93a850219e4a) that asks `cmake` to use `C99` as the C default (in the same way that we ask for C++11), but if the issue is `cmake`, then this won't fix the problem.

awgrover, 2016-07-12:
Using `-DCMAKE_PREFIX_PATH=` improves things. It seems to use my custom Qt.

Now I get an error, "ui_base_window.h: No such file or directory":
```
FAILED: /usr/bin/c++   -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -Wall -Wextra -g -Werror=switch -O3 -DRELEASE -fPIE -Iapp -I../app -isystem /usr/include/python3.3m -isystem /usr/include/x86_64-linux-gnu/python3.3m -I/home/awgrover/dev/thirdparty/qt5.4/install/5.4/gcc_64/include -I/home/awgrover/dev/thirdparty/qt5.4/install/5.4/gcc_64/include/QtWidgets -I/home/awgrover/dev/thirdparty/qt5.4/install/5.4/gcc_64/include/QtGui -I/home/awgrover/dev/thirdparty/qt5.4/install/5.4/gcc_64/include/QtCore -I/home/awgrover/dev/thirdparty/qt5.4/install/5.4/gcc_64/mkspecs/linux-g++ -I/home/awgrover/dev/thirdparty/qt5.4/install/5.4/gcc_64/include/QtOpenGL -I/home/awgrover/dev/thirdparty/qt5.4/install/5.4/gcc_64/include/QtNetwork -I/home/awgrover/dev/thirdparty/qt5.4/install/5.4/gcc_64/include/QtConcurrent -I../lib/graph/inc -I../lib/fab/inc    -D'GITREV="'a6a2c52'+ "' -D'GITTAG="0.9.2b "' -D'GITBRANCH="master "' -std=c++11 -std=c11 -MMD -MT app/CMakeFiles/antimony.dir/window/quad.cpp.o -MF "app/CMakeFiles/antimony.dir/window/quad.cpp.o.d" -o app/CMakeFiles/antimony.dir/window/quad.cpp.o -c ../app/window/quad.cpp
cc1plus: warning: command line option ‘-std=c11’ is valid for C/ObjC but not for C++
In file included from ../app/window/base_viewport_window.h:4:0,
                 from ../app/window/quad.h:3,
                 from ../app/window/quad.cpp:3:
../app/window/base.h:5:28: fatal error: ui_base_window.h: No such file or directory
 #include "ui_base_window.h"

```

mkeeter, 2016-07-12:
Yes, and that should automatically be built with the [`AUTOUIC`](https://cmake.org/cmake/help/v3.0/manual/cmake-qt.7.html#autouic) directive in `app/CMakeLists.txt`.

Have you upgraded `cmake` yet?  Their documents don't give good indications of when features are added, but I suspect that many of your issues are because `cmake 2.8` doesn't have appropriate features.

To enforce this behavior, I've pushed an update to `develop` to require at least `cmake 3.1` (which was released in December 2014, so it's hardly bleeding-edge).

awgrover, 2016-07-21:
Ok, updated to cmake 3.2.2. Now I get:
.```
./app/viewport/render/task.cpp: In member function ‘void RenderTask::render(Shape*, Bounds, float)’:
../app/viewport/render/task.cpp:220:5: warning: missing initializer for member ‘Region_::voxels’ [-Wmissing-field-initializers]
     };

```

awgrover, 2016-07-22:
Building for Ubuntu 14.04

You are going to need cmake 3.x, python 3.3:

```
apt-get install -y python-software-properties
add-apt-repository -y ppa:fkrull/deadsnakes
add-apt-repository ppa:george-edison55/cmake-3.x
apt-get update
apt-get install -y python3.3 python3.3-dev cmake
```

Download and build qt5.6
If you built qt5, but didn't install it in the system, then the build statements are:

```
# the /some/dir/for/qt5 should have lib/libQt5*.so inside it
cmake -DCMAKE_PREFIX_PATH=/some/dir/for/qt5 -GNinja .. 
ninja

```
I added
```
add_compile_options(-std=c++11)
add_compile_options(-std=c11)
```
to CMakeLists.txt

Need to test your cflag thing noted above.

awgrover, 2016-12-25:
Just built 0.9.3
Did not need to edit CMakeLists.txt

mkeeter, 2016-12-25:
Okay, sounds good – let's close this issue.

closed 2016-12-25 by mkeeter