mkeeter/antimony #170: possible to send list?

zelogik, 2016-08-27:
Hi,

I don't know if it's a missing feature or a I have a brainless issue.

Trying to add script like:

list = [0,1,20,99]
output('List', list)

To making a kind of iteration array without symmetric step, for exporting to an 2D  form, but without luck.
Any way to doing it?
Thanks,
Regards.

mkeeter, 2016-08-27:
I tested this out without any issues:
[image]

Code for `range`:
```python
import fab

title('range')
input('n', int)
output('ns', list(range(n)))
```

Code for `double`:
```python
import fab

title('double')
input('ns', list)
output('ds', list([n*2 for n in ns]))
```

Two things to be care about:
- Naming a variable `list` shadows the `list` type, which may be unwise
- Python 3 replaces lists with generators in many places, e.g. `range` – wrap in a `list` call to force conversions

zelogik, 2016-08-28:
Hi,

Yes my question was unclear.
Of course naming a variable with "list" is stupid, IDE normally color it, and my python is a little rusted ;-)

I don't exactly want to sending a list, but int(float,double) each element one by one, more like:

```
import fab
 
title('range')
coordonate = [ 0, 10, 20, 100]
for i in coordonate
   output('ns', i)
```

But of course I have an: Datum was already defined in this script.

mkeeter, 2016-08-28:
That's because every `Datum` (e.g. something defined with `input` or `output`) must have a unique name.  You could create one output per item in the list with the following:

```python
import fab

title('range')
coordinate = [0, 10, 20, 100]
for i,c in enumerate(coordinate):
   output('ns%i' % i, c)
```

which will create outputs `ns0 = 0`, `ns1 = 10`, `ns2 = 20`, etc.

zelogik, 2016-08-29:
So there is no way to make a simple 'array' style things with "random" iteration.

For example to make multiple 2D Circle/rectangle and each with coordinate_x = [0, 10, 20, 100]
 and coordinate_y = [0, 0, 0, 0]