mkeeter/antimony #171: libboost_python3.so.1.59.0: cannot open shared object file

Seoka, 2016-09-04:
Hey there,
first of all thanks! thanks for a great CAD program.
I am using Arch Linux and my antimony (build from the AUR) will not start. It always gives me this output on the console: 
"
[seokaroot@Laptop ~]$ antimony
antimony: error while loading shared libraries: libboost_python3.so.1.59.0: cannot open shared object file: No such file or directory
"
Recompiling did not give me any benefits. Any suggestions?
Here some information that might help:
```
ldd /usr/bin/antimony
	linux-vdso.so.1 (0x00007ffe763ab000)
	libpython3.5m.so.1.0 => /usr/lib/libpython3.5m.so.1.0 (0x00007f3e81c16000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f3e819f9000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f3e817f5000)
	libutil.so.1 => /usr/lib/libutil.so.1 (0x00007f3e815f2000)
	libboost_python3.so.1.60.0 => /usr/lib/libboost_python3.so.1.60.0 (0x00007f3e813a8000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x00007f3e81172000)
	libQt5Widgets.so.5 => /usr/lib/libQt5Widgets.so.5 (0x00007f3e80919000)
	libQt5Gui.so.5 => /usr/lib/libQt5Gui.so.5 (0x00007f3e801dd000)
	libQt5Network.so.5 => /usr/lib/libQt5Network.so.5 (0x00007f3e7fe74000)
	libQt5Core.so.5 => /usr/lib/libQt5Core.so.5 (0x00007f3e7f79d000)
	libGL.so.1 => /usr/lib/libGL.so.1 (0x00007f3e7f529000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f3e7f1a1000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f3e7ee9d000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f3e7ec86000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f3e7e8e8000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f3e820e4000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007f3e7e6e0000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007f3e7e4ca000)
	libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0 (0x00007f3e7e24e000)
	libproxy.so.1 => /usr/lib/libproxy.so.1 (0x00007f3e7e029000)
	libssl.so.1.0.0 => /usr/lib/libssl.so.1.0.0 (0x00007f3e7ddb8000)
	libcrypto.so.1.0.0 => /usr/lib/libcrypto.so.1.0.0 (0x00007f3e7d941000)
	libicui18n.so.57 => /usr/lib/libicui18n.so.57 (0x00007f3e7d4c7000)
	libicuuc.so.57 => /usr/lib/libicuuc.so.57 (0x00007f3e7d11f000)
	libpcre16.so.0 => /usr/lib/libpcre16.so.0 (0x00007f3e7ceb6000)
	libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007f3e7cba7000)
	libsystemd.so.0 => /usr/lib/libsystemd.so.0 (0x00007f3e821fb000)
	libexpat.so.1 => /usr/lib/libexpat.so.1 (0x00007f3e7c97d000)
	libxcb-dri3.so.0 => /usr/lib/libxcb-dri3.so.0 (0x00007f3e7c77a000)
	libxcb-present.so.0 => /usr/lib/libxcb-present.so.0 (0x00007f3e7c577000)
	libxcb-sync.so.1 => /usr/lib/libxcb-sync.so.1 (0x00007f3e7c370000)
	libxshmfence.so.1 => /usr/lib/libxshmfence.so.1 (0x00007f3e7c16d000)
	libglapi.so.0 => /usr/lib/libglapi.so.0 (0x00007f3e7bf3e000)
	libXext.so.6 => /usr/lib/libXext.so.6 (0x00007f3e7bd2c000)
	libXdamage.so.1 => /usr/lib/libXdamage.so.1 (0x00007f3e7bb29000)
	libXfixes.so.3 => /usr/lib/libXfixes.so.3 (0x00007f3e7b923000)
	libX11-xcb.so.1 => /usr/lib/libX11-xcb.so.1 (0x00007f3e7b721000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007f3e7b3df000)
	libxcb.so.1 => /usr/lib/libxcb.so.1 (0x00007f3e7b1b6000)
	libxcb-glx.so.0 => /usr/lib/libxcb-glx.so.0 (0x00007f3e7af9a000)
	libxcb-dri2.so.0 => /usr/lib/libxcb-dri2.so.0 (0x00007f3e7ad95000)
	libXxf86vm.so.1 => /usr/lib/libXxf86vm.so.1 (0x00007f3e7ab8f000)
	libdrm.so.2 => /usr/lib/libdrm.so.2 (0x00007f3e7a980000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007f3e7a6c4000)
	libgraphite2.so.3 => /usr/lib/libgraphite2.so.3 (0x00007f3e7a498000)
	libicudata.so.57 => /usr/lib/libicudata.so.57 (0x00007f3e78a1c000)
	libpcre.so.1 => /usr/lib/libpcre.so.1 (0x00007f3e787a9000)
	libresolv.so.2 => /usr/lib/libresolv.so.2 (0x00007f3e78592000)
	libcap.so.2 => /usr/lib/libcap.so.2 (0x00007f3e7838e000)
	liblzma.so.5 => /usr/lib/liblzma.so.5 (0x00007f3e78168000)
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007f3e77f56000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007f3e77c47000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007f3e77a33000)
	libXau.so.6 => /usr/lib/libXau.so.6 (0x00007f3e7782f000)
	libXdmcp.so.6 => /usr/lib/libXdmcp.so.6 (0x00007f3e77629000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007f3e77419000)
locate libboost_python3.so
/usr/lib/libboost_python3.so
/usr/lib/libboost_python3.so.1.60.0
```

sagitter, 2016-09-04:
There is a `libboost_python3.so` conflict: your system has `libboost_python3.so.1.60.0`, `antimony` needs `libboost_python3.so.1.59.0`.

Seoka, 2016-09-04:
sagitter,
thank you for responding so fast.
I am a little confused about your answer, because a other Archer told me antimony was working with the newer libboost_python3.so - file, as you can [read here] (https://aur.archlinux.org/packages/antimony/). Nevertheless I downgraded the packages boost and its dep. boost-libs (which contains the problematic file). Now antimony, after recompiling, at least gives an graphical error:
"
Import Error:
Could not find *fab* Python module. Antimony will now exit.
"
And it also throws errors on the console:
```
Traceback (most recent call last):
  File "/usr/local/bin/sb/fab/__init__.py", line 1, in <module>
    import fab.shapes as shapes
  File "/usr/local/bin/sb/fab/shapes.py", line 914, in <module>
    shape = triangle(0, 0, 0.35, 1, 0.1, 0)
  File "/usr/local/bin/sb/fab/shapes.py", line 136, in triangle
    max(x0, x1, x2), max(y0, y1, y2))
RuntimeError: Could not construct Shape object.
```
Some information:
[pacman -Qil antimony](https://gist.github.com/9f87057e8bc7c838e57ad2090365bf52
)
[pacman -Q](https://gist.github.com/43ea0ef925a548717428d12c4479817c
)
I hope those information will help resolving and first of all identifying the problem
Thank you for your help

sagitter, 2016-09-06:
> Import Error:
> Could not find fab Python module. Antimony will now exit.

Set the directory where `fab` is installed with [`PYTHONPATH`variable](https://docs.python.org/3/using/cmdline.html#envvar-PYTHONPATH).

Seoka, 2016-09-20:
@sagitter thanks,
after locating a file named "fab" antimony gave two Errors out.

I hope not answering for such a long time did not upset you - I had been busy and to tired for caring about linux issues. Meanwhile I started my business career. A real "Sorry" for that.

Down below I posted what I did and the systems responds. The mentioned files are attached.
Seoka

```
[seokaroot@Laptop ~]$ export PYTHONPATH=/usr/local/bin/sb/fab
[seokaroot@Laptop ~]$ echo $PYTHONPATH
/usr/local/bin/sb/fab
[seokaroot@Laptop ~]$ antimony
Error processing line 1 of /usr/lib/python3.5/site-packages/protobuf-3.0.0b4-py3.5-nspkg.pth:

Failed to import the site module
Traceback (most recent call last):
  File "/usr/lib/python3.5/site.py", line 167, in addpackage
    exec(line)
  File "<string>", line 1, in <module>
AttributeError: module 'types' has no attribute 'ModuleType'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/site.py", line 563, in <module>
    main()
  File "/usr/lib/python3.5/site.py", line 550, in main
    known_paths = addsitepackages(known_paths)
  File "/usr/lib/python3.5/site.py", line 327, in addsitepackages
    addsitedir(sitedir, known_paths)
  File "/usr/lib/python3.5/site.py", line 206, in addsitedir
    addpackage(sitedir, name, known_paths)
  File "/usr/lib/python3.5/site.py", line 177, in addpackage
    import traceback
  File "/usr/lib/python3.5/traceback.py", line 5, in <module>
    import linecache
  File "/usr/lib/python3.5/linecache.py", line 8, in <module>
    import functools
  File "/usr/lib/python3.5/functools.py", line 22, in <module>
    from types import MappingProxyType
ImportError: cannot import name 'MappingProxyType'
[seokaroot@Laptop ~]$
 ```
[site.py.txt](https://github.com/mkeeter/antimony/files/481334/site.py.txt)
[functools.py.txt](https://github.com/mkeeter/antimony/files/483602/functools.py.txt)
[linecache.py.txt](https://github.com/mkeeter/antimony/files/483608/linecache.py.txt)
[traceback.py.txt](https://github.com/mkeeter/antimony/files/483607/traceback.py.txt)

greizgh, 2018-05-13:
I had the same error as you and fixed it by recompiling antimony.
From what I understand, antimony was dynamically linked to an older release of libboost and this link was broken when we updated the libboost package.
Recompiling antimony fixed the problem (antimony-git on AUR).

closed 2018-12-13 by Seoka