mkeeter/antimony #172: Maybe add -std=gnu++11 to the CMAKE_CXX_CFLAGS

erniejunior, 2016-09-10:
Hi,
I am trying to build antimony on debian and ran into the following issues:

The required cmake version is too high (3.1 is required, I have only 3.0 here). I just lowered it in the CMakeLists file (in the master branch that is not necessary).

Also C++11 does not seem to be enabled by default. So I had to add `-std=gnu++11` to the CMAKE_CXX_FLAGS in CMakeLists and also the line `set(CMAKE_C_FLAGS "-std=c11")` to start building.

Now I ran into errors which seem to be related to Qt which I can not fix any more. Maybe somebody else has an idea:

```
[4/76] Building CXX object app/CMakeFiles/antimony.dir/app/main.cpp.o
FAILED: /usr/bin/c++   -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -Wall -Wextra -g -Werror=switch -std=c++11 -O3 -DRELEASE -fPIE -Iapp -I../app -isystem /usr/include/python3.4m -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -isystem /usr/include/x86_64-linux-gnu/qt5/QtOpenGL -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtConcurrent -I../lib/graph/inc -I../lib/fab/inc    -D'GITREV="'c711f93'+ "' -D'GITTAG=""' -D'GITBRANCH="develop "' -MMD -MT app/CMakeFiles/antimony.dir/app/main.cpp.o -MF app/CMakeFiles/antimony.dir/app/main.cpp.o.d -o app/CMakeFiles/antimony.dir/app/main.cpp.o -c ../app/app/main.cpp
../app/app/main.cpp: In function ‘int main(int, char**)’:
../app/app/main.cpp:28:9: error: ‘setDefaultFormat’ is not a member of ‘QSurfaceFormat’
         QSurfaceFormat::setDefaultFormat(format);
         ^
../app/app/main.cpp:56:17: error: ‘AppDataLocation’ is not a member of ‘QStandardPaths’
                 QStandardPaths::AppDataLocation))
                 ^
[4/76] Building CXX object app/CMakeFiles/antimony.dir/app/app.cpp.o
FAILED: /usr/bin/c++   -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -Wall -Wextra -g -Werror=switch -std=c++11 -O3 -DRELEASE -fPIE -Iapp -I../app -isystem /usr/include/python3.4m -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -isystem /usr/include/x86_64-linux-gnu/qt5/QtOpenGL -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtConcurrent -I../lib/graph/inc -I../lib/fab/inc    -D'GITREV="'c711f93'+ "' -D'GITTAG=""' -D'GITBRANCH="develop "' -MMD -MT app/CMakeFiles/antimony.dir/app/app.cpp.o -MF app/CMakeFiles/antimony.dir/app/app.cpp.o.d -o app/CMakeFiles/antimony.dir/app/app.cpp.o -c ../app/app/app.cpp
../app/app/app.cpp: In member function ‘QStringList App::nodePaths() const’:
../app/app/app.cpp:67:13: error: ‘AppDataLocation’ is not a member of ‘QStandardPaths’
             QStandardPaths::AppDataLocation))
             ^
[4/76] Building CXX object app/CMakeFiles/antimony.dir/graph/hooks/ui.cpp.o
ninja: build stopped: subcommand failed.
```
That would be awesome!

mkeeter, 2016-09-10:
C++11 is specified by
```
set_property(TARGET ${ANTIMONY_APP} PROPERTY CXX_STANDARD 11)
```
in various build files.

That's why the minimum `cmake` version is 3.1: earlier versions don't respect that property.

What version of Qt are you running?  `AppDataLoation` was added in Qt 5.4.

erniejunior, 2016-09-11:
Hmm I am running Qt 5.3 so it is all a matter of outdated software ...

mkeeter, 2016-09-11:
Okay, glad that's all figured out.

closed 2016-09-11 by mkeeter