mkeeter/antimony #173: Core Dump (Backtrace included)

pwnage-pineapple, 2016-09-28:
Occasionally, Antimony will crash, with a message informing me of a cure dump. Here is the backtrace:

```
#0  0x00007f47a8544ac7 in QMetaObject::cast(QObject const*) const () from /usr/lib/libQt5Core.so.5
#1  0x00007f4796a1e53f in ?? () from /usr/lib/libKF5XmlGui.so.5
#2  0x00007f47a853b8b6 in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#3  0x00007f47a9584e78 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#4  0x00007f47a958c5b1 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#5  0x00007f47a853bc80 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#6  0x00007f47a853e3fd in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/lib/libQt5Core.so.5
#7  0x00007f47a8590173 in ?? () from /usr/lib/libQt5Core.so.5
#8  0x00007f47a5724e67 in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#9  0x00007f47a57250d0 in ?? () from /usr/lib/libglib-2.0.so.0
#10 0x00007f47a572517c in g_main_context_iteration () from /usr/lib/libglib-2.0.so.0
#11 0x00007f47a859057f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#12 0x00007f47a853a0da in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#13 0x00007f47a85425cc in QCoreApplication::exec() () from /usr/lib/libQt5Core.so.5
#14 0x000000000046cb5e in main ()
```

My system info is as follows:
* Arch Linux with kernel 4.7.4-1
* Qt 5.7.0-2
* Antimony 0.9.0-1, compiled using the AUR package.

mkeeter, 2016-09-29:
Anything i particular that's causing this crash?  The traceback is appreciated but not too helpful – all I can get from it is that I'm probably corrupting memory somewhere, somehow...

pwnage-pineapple, 2016-09-29:
I'm working with a fairly complicated model, and it only happens when panning or rotating within the viewer. I hope that's somewhat helpful.

mkeeter, 2016-10-01:
Okay, from 0.9.0 there's not much I can do – that whole section got rewritten since then.

You should try building a more recent version from source (which should be fairly painless) or ping the package manager to bump the release to 0.9.2.