mkeeter/antimony #178: Undefined references to 'boost'

sagitter, 2017-02-07:
Hi all.

`antimony-0.9.3b` is not compiling on Fedora _rawhide_ for `undefined reference` errors:

```
[100%] Linking CXX executable antimony
cd /builddir/build/BUILD/antimony-0.9.3b/build/app && /usr/bin/cmake -E cmake_link_script CMakeFiles/antimony.dir/link.txt --verbose=1
/usr/bin/c++   -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -fPIC -pie -DNDEBUG -DRELEASE  -Wl,-z,relro -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -fPIC -pie CMakeFiles/antimony.dir/app/main.cpp.o CMakeFiles/antimony.dir/app/app.cpp.o CMakeFiles/antimony.dir/app/update.cpp.o CMakeFiles/antimony.dir/app/colors.cpp.o CMakeFiles/antimony.dir/canvas/scene.cpp.o CMakeFiles/antimony.dir/canvas/canvas_view.cpp.o CMakeFiles/antimony.dir/canvas/info.cpp.o CMakeFiles/antimony.dir/canvas/inspector/frame.cpp.o CMakeFiles/antimony.dir/canvas/inspector/export.cpp.o CMakeFiles/antimony.dir/canvas/inspector/title.cpp.o CMakeFiles/antimony.dir/canvas/inspector/buttons.cpp.o CMakeFiles/antimony.dir/canvas/inspector/util.cpp.o CMakeFiles/antimony.dir/canvas/datum_row.cpp.o CMakeFiles/antimony.dir/canvas/datum_editor.cpp.o CMakeFiles/antimony.dir/canvas/datum_port.cpp.o CMakeFiles/antimony.dir/canvas/subdatum/subdatum_frame.cpp.o CMakeFiles/antimony.dir/canvas/subdatum/subdatum_editor.cpp.o CMakeFiles/antimony.dir/canvas/subdatum/subdatum_row.cpp.o CMakeFiles/antimony.dir/canvas/connection/base.cpp.o CMakeFiles/antimony.dir/canvas/connection/connection.cpp.o CMakeFiles/antimony.dir/canvas/connection/dummy.cpp.o CMakeFiles/antimony.dir/window/base.cpp.o CMakeFiles/antimony.dir/window/canvas.cpp.o CMakeFiles/antimony.dir/window/viewport.cpp.o CMakeFiles/antimony.dir/window/quad.cpp.o CMakeFiles/antimony.dir/window/base_viewport_window.cpp.o CMakeFiles/antimony.dir/window/script_window.cpp.o CMakeFiles/antimony.dir/graph/constructor/populate.cpp.o CMakeFiles/antimony.dir/graph/proxy/graph.cpp.o CMakeFiles/antimony.dir/graph/proxy/node.cpp.o CMakeFiles/antimony.dir/graph/proxy/script.cpp.o CMakeFiles/antimony.dir/graph/proxy/datum.cpp.o CMakeFiles/antimony.dir/graph/proxy/subdatum.cpp.o CMakeFiles/antimony.dir/graph/proxy/base_datum.cpp.o CMakeFiles/antimony.dir/graph/hooks/hooks.cpp.o CMakeFiles/antimony.dir/graph/hooks/export.cpp.o CMakeFiles/antimony.dir/graph/hooks/ui.cpp.o CMakeFiles/antimony.dir/graph/hooks/title.cpp.o CMakeFiles/antimony.dir/graph/serialize/serializer.cpp.o CMakeFiles/antimony.dir/graph/serialize/deserializer.cpp.o CMakeFiles/antimony.dir/script/syntax.cpp.o CMakeFiles/antimony.dir/script/editor.cpp.o CMakeFiles/antimony.dir/script/frame.cpp.o CMakeFiles/antimony.dir/undo/undo_command.cpp.o CMakeFiles/antimony.dir/undo/undo_add_node.cpp.o CMakeFiles/antimony.dir/undo/undo_add_datum.cpp.o CMakeFiles/antimony.dir/undo/undo_delete_datum.cpp.o CMakeFiles/antimony.dir/undo/undo_delete_multi.cpp.o CMakeFiles/antimony.dir/undo/undo_delete_node.cpp.o CMakeFiles/antimony.dir/undo/undo_delete_link.cpp.o CMakeFiles/antimony.dir/undo/undo_move_node.cpp.o CMakeFiles/antimony.dir/undo/undo_move_datum.cpp.o CMakeFiles/antimony.dir/undo/undo_change_script.cpp.o CMakeFiles/antimony.dir/undo/undo_change_expr.cpp.o CMakeFiles/antimony.dir/undo/undo_stack.cpp.o CMakeFiles/antimony.dir/dialog/exporting.cpp.o CMakeFiles/antimony.dir/dialog/resolution.cpp.o CMakeFiles/antimony.dir/export/export_mesh.cpp.o CMakeFiles/antimony.dir/export/export_heightmap.cpp.o CMakeFiles/antimony.dir/export/export_worker.cpp.o CMakeFiles/antimony.dir/viewport/scene.cpp.o CMakeFiles/antimony.dir/viewport/gl.cpp.o CMakeFiles/antimony.dir/viewport/view.cpp.o CMakeFiles/antimony.dir/viewport/image.cpp.o CMakeFiles/antimony.dir/viewport/control/control.cpp.o CMakeFiles/antimony.dir/viewport/control/control_instance.cpp.o CMakeFiles/antimony.dir/viewport/control/point.cpp.o CMakeFiles/antimony.dir/viewport/control/wireframe.cpp.o CMakeFiles/antimony.dir/viewport/render/instance.cpp.o CMakeFiles/antimony.dir/viewport/render/task.cpp.o CMakeFiles/antimony.dir/antimony_automoc.cpp.o CMakeFiles/antimony.dir/antimony_automoc.dir/qrc_gl_2HPAS2ZRQXYTL6.cpp.o  -o antimony -rdynamic /usr/lib64/libQt5OpenGL.so.5.7.1 /usr/lib64/libQt5Network.so.5.7.1 /usr/lib64/libQt5Concurrent.so.5.7.1 ../lib/graph/libSbGraph.a ../lib/fab/libSbFab.a /usr/lib64/libQt5Widgets.so.5.7.1 /usr/lib64/libQt5Gui.so.5.7.1 /usr/lib64/libQt5Core.so.5.7.1 -lpython3.6m -lboost_python3 -lpng -lz 
../lib/graph/libSbGraph.a(proxy.cpp.o): In function `PyInit__proxy':
/builddir/build/BUILD/antimony-0.9.3b/lib/graph/src/proxy.cpp:103: undefined reference to `boost::python::detail::init_module(PyModuleDef&, void (*)())'
../lib/graph/libSbGraph.a(hooks.cpp.o): In function `PyInit__scriptIO':
/builddir/build/BUILD/antimony-0.9.3b/lib/graph/src/hooks/hooks.cpp:23: undefined reference to `boost::python::detail::init_module(PyModuleDef&, void (*)())'
CMakeFiles/antimony.dir/graph/hooks/hooks.cpp.o: In function `PyInit__AppHooks':
/builddir/build/BUILD/antimony-0.9.3b/app/graph/hooks/hooks.cpp:20: undefined reference to `boost::python::detail::init_module(PyModuleDef&, void (*)())'
../lib/fab/libSbFab.a(fab.cpp.o): In function `PyInit__fabtypes':
/builddir/build/BUILD/antimony-0.9.3b/lib/fab/src/fab.cpp:22: undefined reference to `boost::python::detail::init_module(PyModuleDef&, void (*)())'
collect2: error: ld returned 1 exit status
```
Full log: https://kojipkgs.fedoraproject.org//work/tasks/7313/17647313/build.log

Boost-1.63
GCC-7.0.1

mkeeter, 2017-02-07:
This kind of error is caused because `boost::python` method signatures change between Python 2 and Python 3 bindings (Antimony is using the latter).  [Here's](http://stackoverflow.com/questions/5810101/undefined-boost-python-symbol-boostpythondetailinit-module) someone seeing a similar issue.

This is a bit confusing, though, because the linker flags include `-lboost_python3`.

The SO link above suggests installing `python-dev`; I'd also try dumping the symbols from `boost_python3` and making sure that there exists one with the right signature.

sagitter, 2017-02-08:
Could be related to this `CMake` warning?

```
-- Detecting CXX compile features - done
-- Found PythonLibs: /usr/lib64/libpython3.6m.so (found suitable version "3.6.0", minimum required is "3.3") 
CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:1518 (message):
  No header defined for python3; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:17 (find_package)
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.11") 
```

sagitter, 2017-02-09:
Looks like a bug of Boost on Fedora: https://bugzilla.redhat.com/show_bug.cgi?id=1420086#c8

mkeeter, 2017-02-09:
Hah, there's a similar issue in the MINGW packaging– I filed a bug for it [here](https://github.com/Alexpux/MINGW-packages/issues/2028).  I guess no one is actually using `boost::python3`, so these things are left unnoticed...

jwakely, 2017-02-16:
Caused by https://github.com/boostorg/build/issues/163

The Fedora boost package is rebuilding now with a workaround.

jhasse, 2017-07-20:
This should now be fixed, right?

jwakely, 2017-07-20:
Yes. If anybody is still getting this error on Fedora they need to update their system to the latest boost packages.

mkeeter, 2017-07-20:
Thanks for following up – I'll close since this is now fixed upstream.

closed 2017-07-20 by mkeeter

ComFreek, 2018-07-21:
I am facing the same issue with Msys2 and Mingw.

My Boost build looks as follows:

- Instead of `./bootstrap.sh` I am using directly a custom `project-config.jam`, see below.
- I call `./b2 include=/mingw64/include/python3.7m -q cxxflags="-Wno-deprecated-declarations -Wno-strict-aliasing -Wno-maybe-uninitialized -D_hypot=hypot"`.<br>
  The first parts of its output are also below, note especially the `The system cannot find the file specified` error message. Unfortunately it does not say which file path.
  That message disappears if I comment out the `using python` line in `project-config.jam`. I've tripled-checked that every path there exists within an Mingw64 shell.

With Process Monitor, I also traced failed syscalls with result `NO SUCH FILE` and some other properties on the queried path to exclude all syscalls that have definitely nothing to do with Python. See the list at the end of this comment, maybe someone can spot a file which should actually be there.

```
# Boost.Build Configuration
# Automatically generated by bootstrap.sh

import option ;
import feature ;

# Compiler configuration. This definition will be used unless
# you already have defined some toolsets in your user-config.jam
# file.
if ! gcc in [ feature.values <toolset> ]
{
    using gcc ; 
}

project : default-build <toolset>gcc ;

# List of --with-<library> and --without-<library>
# options. If left empty, all libraries will be built.
# Options specified on the command line completely
# override this variable.
libraries =  --with-python ;

# These settings are equivivalent to corresponding command-line
# options.
option.set prefix : /usr/local ;
option.set exec-prefix : /usr/local ;
option.set libdir : /usr/local/lib ;
option.set includedir : /usr/local/include ;

# Stop on first error
option.set keep-going : false ;

using python : 3.7 : /mingw64/bin/python3.7m.exe : /mingw64/include/python3.7m : /mingw64/lib/python3.7/config-3.7m ;
```

```
The system cannot find the file specified.
Performing configuration checks

    - default address-model    : 64-bit (cached)
    - default architecture     : x86 (cached)

Building the Boost C++ Libraries.

    - symlinks supported       : yes (cached)

Component configuration:

    - atomic                   : not building
[ CUT ]
```

```
"Path","Result","Detail"
"D:\Downloads\boost_1_67_0\python.jam","NO SUCH FILE","Type: QueryDirectory, Filter: python.jam"
"D:\Downloads\boost_1_67_0\stage\lib\libboost_python37-mgw73-mt-d-x32-1_67.a","NO SUCH FILE","Type: QueryDirectory, Filter: libboost_python37-mgw73-mt-d-x32-1_67.a"
"D:\Downloads\boost_1_67_0\stage\lib\libboost_python37-mgw73-mt-x32-1_67.a","NO SUCH FILE","Type: QueryDirectory, Filter: libboost_python37-mgw73-mt-x32-1_67.a"
"D:\Downloads\boost_1_67_0\stage\lib\libboost_python37-mgw73-mt-d-x64-1_67.a","NO SUCH FILE","Type: QueryDirectory, Filter: libboost_python37-mgw73-mt-d-x64-1_67.a"
[... 126 lines cut ...]
"C:\tools\msys64\mingw64\include\python3.7m\optional","NO SUCH FILE","Type: QueryDirectory, Filter: optional"
"C:\tools\msys64\mingw64\include\python3.7m\variant","NO SUCH FILE","Type: QueryDirectory, Filter: variant"
"C:\tools\msys64\mingw64\include\python3.7m\deque","NO SUCH FILE","Type: QueryDirectory, Filter: deque"
"C:\tools\msys64\mingw64\include\python3.7m\list","NO SUCH FILE","Type: QueryDirectory, Filter: list"
"C:\tools\msys64\mingw64\include\python3.7m\vector","NO SUCH FILE","Type: QueryDirectory, Filter: vector"
"C:\tools\msys64\mingw64\include\python3.7m\map","NO SUCH FILE","Type: QueryDirectory, Filter: map"
"C:\tools\msys64\mingw64\include\python3.7m\set","NO SUCH FILE","Type: QueryDirectory, Filter: set"
"C:\tools\msys64\mingw64\include\python3.7m\bitset","NO SUCH FILE","Type: QueryDirectory, Filter: bitset"
"C:\tools\msys64\mingw64\include\python3.7m\complex","NO SUCH FILE","Type: QueryDirectory, Filter: complex"
"C:\tools\msys64\mingw64\include\python3.7m\array","NO SUCH FILE","Type: QueryDirectory, Filter: array"
"C:\tools\msys64\mingw64\include\python3.7m\tuple","NO SUCH FILE","Type: QueryDirectory, Filter: tuple"
"C:\tools\msys64\mingw64\include\python3.7m\ostream.h","NO SUCH FILE","Type: QueryDirectory, Filter: ostream.h"
"C:\tools\msys64\mingw64\include\python3.7m\cstdio","NO SUCH FILE","Type: QueryDirectory, Filter: cstdio"
"C:\tools\msys64\mingw64\include\python3.7m\mpi.h","NO SUCH FILE","Type: QueryDirectory, Filter: mpi.h"
"C:\tools\msys64\mingw64\include\python3.7m\cstdarg","NO SUCH FILE","Type: QueryDirectory, Filter: cstdarg"
"C:\tools\msys64\mingw64\include\python3.7m\excpt.h","NO SUCH FILE","Type: QueryDirectory, Filter: excpt.h"
"C:\tools\msys64\mingw64\include\python3.7m\initializer_list","NO SUCH FILE","Type: QueryDirectory, Filter: initializer_list"
"C:\tools\msys64\mingw64\include\python3.7m\locale","NO SUCH FILE","Type: QueryDirectory, Filter: locale"
"C:\tools\msys64\mingw64\include\python3.7m\iostream.h","NO SUCH FILE","Type: QueryDirectory, Filter: iostream.h"
"C:\tools\msys64\mingw64\include\python3.7m\istream","NO SUCH FILE","Type: QueryDirectory, Filter: istream"
"C:\tools\msys64\mingw64\include\python3.7m\unordered_set","NO SUCH FILE","Type: QueryDirectory, Filter: unordered_set"
"C:\tools\msys64\mingw64\include\python3.7m\unordered_map","NO SUCH FILE","Type: QueryDirectory, Filter: unordered_map"
"C:\tools\msys64\mingw64\include\python3.7m\queue","NO SUCH FILE","Type: QueryDirectory, Filter: queue"
"D:\Downloads\boost_1_67_0\python.jam","NO SUCH FILE","Type: QueryDirectory, Filter: python.jam"
```