mkeeter/antimony #179: cmake problem

ARKopp, 2017-03-01:
CMake Error: Internal CMake error, TryCompile generation of cmake failed
-- Check for working C compiler using: Ninja -- broken
CMake Error at /usr/share/cmake-3.5/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler "/usr/bin/cc" is not able to compile a simple test program.

  It fails with the following output:

   

  i get this problem when trying to cmake

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

-- Configuring incomplete, errors occurred!
See also "/home/chrx/antimony/CMakeFiles/CMakeOutput.log".
See also "/home/chrx/antimony/CMakeFiles/CMakeError.log".

mkeeter, 2017-03-01:
What OS are you on?  What have you tried to debug the problem?

ARKopp, 2017-03-01:
I am using gallium os with is ubuntu.I tried to install all the dependences.

mkeeter, 2017-03-02:
You can try running without `ninja` by leaving out `-GNinja` from the `cmake` invocation.

I'm suspicious about `cmake` not knowing about Gallium OS's paths, since not having a valid C compiler seems like a pretty fundamental issue...

ARKopp, 2017-03-02:
ok it it cmake and what after it?

ARKopp, 2017-03-02:
ok read a tutorial. it worked with mkdir build-dir, cd build-dir, cmake ..

ARKopp, 2017-03-02:
but there i do not know what to do afterwards. it worked no as well with 
mkdir build-dir, cd buidl-dir, cmake -GNinja .., ninja

ARKopp, 2017-03-02:
i would also like to try on the raspberry pi 3. Do you know if anybody has tried?