mkeeter/antimony #180: fails to build with ninja

funkycode, 2017-04-20:
**ogg@batata ~/tmp/antimony/build $** cmake -GNinja ..
```
CMake Warning at /usr/share/cmake/Modules/FindBoost.cmake:1531 (message):
  No header defined for python-3.4; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:21 (find_package)

-- Found ZLIB: /usr/lib/libz.so (found version "1.2.11") 
-- Found PNG: /usr/lib/libpng.so (found version "1.6.28+apng") 
-- Found FLEX: /usr/bin/flex (found version "2.6.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zogg/tmp/antimony/build
```
**zogg@batata ~/tmp/antimony/build $** ninja
```
[7/130] Generating v2syntax.lemon.hpp, v2syntax.lemon.cpp
Parser statistics: 60 terminals, 8 nonterminals, 58 rules
                   159 states, 0 parser table entries, 0 conflicts
[35/130] Building CXX object lib/fab/CMakeFiles/SbFab.dir/v2syntax.lemon.cpp.o
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.y:7:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
  #pragma clang diagnostic ignored "-Wunused-parameter"
 ^
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.y:8:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
  #pragma clang diagnostic ignored "-Wunused-variable"
 ^
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.c: In function ‘void yy_destructor(yyParser*, unsigned char, YYMINORTYPE*)’:
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.c:88:31: warning: unused variable ‘environment’ [-Wunused-variable]
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.c:845:3: note: in expansion of macro ‘v2ParseARG_FETCH’
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.c: At global scope:
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.c:843:16: warning: unused parameter ‘yypminor’ [-Wunused-parameter]
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.c:1025:63: warning: unused parameter ‘yypMinor’ [-Wunused-parameter]
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.c:1561:7: warning: unused parameter ‘yymajor’ [-Wunused-parameter]
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.c:1562:15: warning: unused parameter ‘yyminor’ [-Wunused-parameter]
[36/130] Building CXX object lib/fab/CMakeFiles/SbFab.dir/v2syntax.yy.cpp.o
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.l:7:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
  #pragma clang diagnostic ignored "-Wdeprecated-register"
 ^
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.l:8:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
  #pragma clang diagnostic ignored "-Wunused-parameter"
 ^
/home/zogg/tmp/antimony/lib/fab/src/tree/v2syntax.l:9:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
  #pragma clang diagnostic ignored "-Wunused-function"
 ^
/home/zogg/tmp/antimony/build/lib/fab/v2syntax.yy.cpp: In function ‘int yy_get_next_buffer(yyscan_t)’:
/home/zogg/tmp/antimony/build/lib/fab/v2syntax.yy.cpp:676:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for ( n = 0; n < max_size && \
                  ^
/home/zogg/tmp/antimony/build/lib/fab/v2syntax.yy.cpp:1397:3: note: in expansion of macro ‘YY_INPUT’
   YY_INPUT( (&YY_CURRENT_BUFFER_LVALUE->yy_ch_buf[number_to_move]),
   ^
/home/zogg/tmp/antimony/build/lib/fab/v2syntax.yy.cpp: In function ‘yy_buffer_state* yy_scan_bytes(const char*, int, yyscan_t)’:
/home/zogg/tmp/antimony/build/lib/fab/v2syntax.yy.cpp:1957:17: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  for ( i = 0; i < _yybytes_len; ++i )
                 ^
/home/zogg/tmp/antimony/build/lib/fab/v2syntax.yy.cpp: At global scope:
/home/zogg/tmp/antimony/build/lib/fab/v2syntax.yy.cpp:1501:17: warning: ‘void yyunput(int, char*, yyscan_t)’ defined but not used [-Wunused-function]
     static void yyunput (int c, char * yy_bp , yyscan_t yyscanner)
                 ^
[38/130] Automatic MOC, UIC and RCC for target antimony
Generating MOC source antimony_autogen/VJIZ3MDCXP/moc_app.cpp
Generating MOC source antimony_autogen/VJIZ3MDCXP/moc_update.cpp
Generating MOC source antimony_autogen/KGLXD3X7AI/moc_canvas_view.cpp
Generating MOC source antimony_autogen/7DE644ELHY/moc_connection.cpp
Generating MOC source antimony_autogen/7DE644ELHY/moc_dummy.cpp
Generating MOC source antimony_autogen/KGLXD3X7AI/moc_datum_editor.cpp
Generating MOC source antimony_autogen/KGLXD3X7AI/moc_datum_port.cpp
Generating MOC source antimony_autogen/KGLXD3X7AI/moc_datum_row.cpp
Generating MOC source antimony_autogen/ZNLU55MQ25/moc_buttons.cpp
Generating MOC source antimony_autogen/ZNLU55MQ25/moc_export.cpp
[... 54 lines cut ...]
../app/viewport/image.cpp:84:13: error: ‘GL_RGBA8’ was not declared in this scope
             GL_RGBA8,           /* format */
             ^
[123/130] Building CXX object app/CMakeFiles/antimony.dir/export/export_mesh.cpp.o
../app/export/export_mesh.cpp: In member function ‘virtual void ExportMeshWorker::async()’:
../app/export/export_mesh.cpp:79:5: warning: missing initializer for member ‘Region_::voxels’ [-Wmissing-field-initializers]
     };
     ^
../app/export/export_mesh.cpp:79:5: warning: missing initializer for member ‘Region_::X’ [-Wmissing-field-initializers]
../app/export/export_mesh.cpp:79:5: warning: missing initializer for member ‘Region_::Y’ [-Wmissing-field-initializers]
../app/export/export_mesh.cpp:79:5: warning: missing initializer for member ‘Region_::Z’ [-Wmissing-field-initializers]
../app/export/export_mesh.cpp:79:5: warning: missing initializer for member ‘Region_::L’ [-Wmissing-field-initializers]
[125/130] Building CXX object app/CMakeFiles/antimony.dir/export/export_heightmap.cpp.o
../app/export/export_heightmap.cpp: In member function ‘virtual void ExportHeightmapWorker::async()’:
../app/export/export_heightmap.cpp:83:5: warning: missing initializer for member ‘Region_::voxels’ [-Wmissing-field-initializers]
     };
     ^
../app/export/export_heightmap.cpp:83:5: warning: missing initializer for member ‘Region_::X’ [-Wmissing-field-initializers]
../app/export/export_heightmap.cpp:83:5: warning: missing initializer for member ‘Region_::Y’ [-Wmissing-field-initializers]
../app/export/export_heightmap.cpp:83:5: warning: missing initializer for member ‘Region_::Z’ [-Wmissing-field-initializers]
../app/export/export_heightmap.cpp:83:5: warning: missing initializer for member ‘Region_::L’ [-Wmissing-field-initializers]
[126/130] Building CXX object app/CMakeFiles/antimony.dir/export/export_worker.cpp.o
ninja: build stopped: subcommand failed.

```

funkycode, 2017-04-22:
The problem on gewntoo was that I used Qt with GLES2 and now full OpenGL, thus it doesn't supoort GLES yet

mkeeter, 2017-04-22:
Antimony requires full OpenGL, so I wouldn't expect it to work with GLES.  I'm a little unclear from your follow-up — did you get it working by using full OpenGL?

funkycode, 2017-04-22:
Yup. I had to recompile PyQt and other Qt friends without GLES2 support (on Funtoo Linux here, which is source based)

mkeeter, 2017-04-22:
Cool, glad you got things working!

closed 2017-04-22 by mkeeter