mkeeter/antimony #181: Crash on start with Qt 5.9

ApostolosB, 2017-06-10:
Arch Linux, Wayland, Qt 5.9

Compiled it and i get this on start. It shows the window for a split second and then it crashes.

```

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
Core was generated by `antimony'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0xb69fc804 in QOpenGLContext::shareGroup() const () from /usr/lib/libQt5Gui.so.5
[Current thread is 1 (Thread 0xb33d2780 (LWP 589))]
(gdb) bt
#0  0xb69fc804 in QOpenGLContext::shareGroup() const () from /usr/lib/libQt5Gui.so.5
#1  0xb6cafb06 in QOpenGLFunctions::initializeOpenGLFunctions() () from /usr/lib/libQt5Gui.so.5
#2  0x08119fa2 in ViewportView::drawBackground(QPainter*, QRectF const&) ()
#3  0xb72ff1a5 in QGraphicsView::paintEvent(QPaintEvent*) () from /usr/lib/libQt5Widgets.so.5
#4  0xb6fcb7ea in QWidget::event(QEvent*) () from /usr/lib/libQt5Widgets.so.5
#5  0xb70c079a in QFrame::event(QEvent*) () from /usr/lib/libQt5Widgets.so.5
#6  0xb7153a29 in QAbstractScrollArea::viewportEvent(QEvent*) () from /usr/lib/libQt5Widgets.so.5
#7  0xb72fd47a in QGraphicsView::viewportEvent(QEvent*) () from /usr/lib/libQt5Widgets.so.5
#8  0xb664c332 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#9  0xb6f84f99 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#10 0xb6f8cff7 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#11 0xb664c69d in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#12 0xb6fc3dc1 in QWidgetPrivate::sendPaintEvent(QRegion const&) () from /usr/lib/libQt5Widgets.so.5
#13 0xb6fc43a7 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () from /usr/lib/libQt5Widgets.so.5
#14 0xb6fc5002 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () from /usr/lib/libQt5Widgets.so.5
#15 0xb6fc3fb4 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () from /usr/lib/libQt5Widgets.so.5
#16 0xb6fc5002 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () from /usr/lib/libQt5Widgets.so.5
#17 0xb6fc3fb4 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) () from /usr/lib/libQt5Widgets.so.5
#18 0xb6f9556f in ?? () from /usr/lib/libQt5Widgets.so.5
#19 0xb6f95d8e in ?? () from /usr/lib/libQt5Widgets.so.5
#20 0xb6fb1ca7 in QWidgetPrivate::syncBackingStore(QRegion const&) () from /usr/lib/libQt5Widgets.so.5
#21 0xb6fec286 in ?? () from /usr/lib/libQt5Widgets.so.5
#22 0xb6fecc4f in ?? () from /usr/lib/libQt5Widgets.so.5
#23 0xb6f84fba in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#24 0xb6f8cff7 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#25 0xb664c69d in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#26 0xb69bb926 in QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) () from /usr/lib/libQt5Gui.so.5
#27 0xb69bc759 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () from /usr/lib/libQt5Gui.so.5
#28 0xb698fb84 in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Gui.so.5
#29 0xb31974fe in ?? () from /usr/lib/libQt5WaylandClient.so.5
#30 0xb52224c9 in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#31 0xb5222769 in ?? () from /usr/lib/libglib-2.0.so.0
#32 0xb5222834 in g_main_context_iteration () from /usr/lib/libglib-2.0.so.0
#33 0xb66adf53 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#34 0xb31976d1 in ?? () from /usr/lib/libQt5WaylandClient.so.5
#35 0xb664aa1d in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#36 0xb6653d1d in QCoreApplication::exec() () from /usr/lib/libQt5Core.so.5
#37 0xb69b0f91 in QGuiApplication::exec() () from /usr/lib/libQt5Gui.so.5
#38 0xb6f84f14 in QApplication::exec() () from /usr/lib/libQt5Widgets.so.5
#39 0x080aaea3 in main ()
(gdb)
```

mkeeter, 2017-06-14:
Unfortunately, I don't have an Arch machine *or* a Qt 5.9 install – can you try running one of the Qt OpenGL examples and seeing if that fails in the same way?

ApostolosB, 2017-06-16:
Can't get the opengl stuff to compile. Probably i am doing something wrong but i don't understand the problem. Sorry. Is there another way/app to test??

ApostolosB, 2017-07-13:
Tried it on my main machine and it doesn't build at all there. Weird :/

ApostolosB, 2017-08-16:
Tried it on my main machine and seems to work ok. Closing this.

closed 2017-08-16 by ApostolosB