mkeeter/antimony #182: Antialiasing issues on linux.

ApostolosB, 2017-08-16:
No idea if it is a bug or something i have to take care in another way but running antimony in linux elements (be it the model or the UI lines) are jagged and not smooth.

Anything that can be done about this?

mkeeter, 2017-08-18:
Are you running on a high-DPI screen?  It could be that I'm not setting all of the correct flags for correct behavior on Linux...

ApostolosB, 2017-08-18:
No. A normal 1920x1080 monitor. No idea what is going on but i see a lot of jagged edges both on the model as i move it around and on the UI lines etc. I am using mesa in case it matters.

ApostolosB, 2017-09-02:
The model issue is due to rendering the 3d view taking way too long to complete. Uses only one core of the CPU. If it was using more cores or the GPU would be faster i believe. But i know you are working on other stuff so yeah. Even after the rendering finishes it isn't perfect but anyway. The UI lines being jagged i have no idea where this comes from.