mkeeter/antimony #189: Ninja compilation issue w/ python

photocyte, 2018-04-18:
Hi there,

Thanks for making Antimony. I played around with it a bit for HTMAA 2014, and I've recently tried it again for a new project. I'm happy to see how much the software has progressed since then!  I find the combination of python scripting & the node layout very intuitive (I hope libfive + Studio will eventually support python? I wish I could help implement it...) Anyway, I'm trying to compile Antimony on my Mac (OS 10.13.4), and have run into some issues.  I modified the homebrew and cmake building instructions a little (see [here](https://github.com/mkeeter/antimony/compare/develop...photocyte:develop) for the diff), but I've since run into a Ninja / compilation issue that is unclear to me:

Build instructions executed:
```
cmake -DCMAKE_PREFIX_PATH="/usr/local/Cellar/qt/5.10.1" -DPYTHON_LIBRARY="/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/lib/libpython3.6.dylib" -DPYTHON_INCLUDE_DIR="/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/include/python3.6m/" -DBOOSTROOT="/usr/local/Cellar/boost/1.67.0/" -GNinja ..
ninja
```

Relevant error message from ninja / clang:
```
In file included from /usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/include/python3.6m/pytime.h:6:
/usr/local/Cellar/python/3.6.5/Frameworks/Python.framework/Versions/3.6/include/python3.6m/object.h:448:23: error: expected member name or ';' after declaration specifiers
    PyType_Slot *slots; /* terminated by slot==0. */
    ~~~~~~~~~~~       ^
1 error generated.
```

Any ideas what might be going wrong?

Full output text:
cmake standard out:
[cmake.stdout.txt](https://github.com/mkeeter/antimony/files/1925324/cmake.stdout.txt)

cmake standard error:
[cmake.stderr.txt](https://github.com/mkeeter/antimony/files/1925326/cmake.stderr.txt)

Ninja strandard output:
[ninja.stdout.txt](https://github.com/mkeeter/antimony/files/1925328/ninja.stdout.txt)

mkeeter, 2018-04-18:
This is caused by always needing to include `<Python.h>` before other header files, and the automoc compilation order changing.  I reproduced it locally, and the commit above fixed it on my machine – can you see if that also fixes it for you?

photocyte, 2018-04-18:
Compiled successfully after your changes (https://github.com/mkeeter/antimony/commit/f0732fe9405f39a48a08c18c0a753b3936c88cd5) plus some additional cmake changes (https://github.com/mkeeter/antimony/pull/190)

commit f0732fe9 (2018-04-18): Update build instructions (fixes #189?)