mkeeter/antimony #191: no header for libboost in ubuntu 18.04

Cdriko, 2018-05-17:
hello
I successfull compile antimony under ubuntu 16.04,
but with ubuntu 18.04, after installing all the necessaries dependencies, I get  : 
```
cmake GNinja ..
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python3; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:28 (find_package)

-- Configuring done
-- Generating done
-- Build files have been written to: /home/cdoutriaux/Documents/soft/antimony/build

```

even this check
`dpkg -l libboost-all-dev `

give : 

```
||/ Nom                             Version              Architecture         Description
+++-===============================-====================-====================-===================================================================
ii  libboost-all-dev                1.65.1.0ubuntu1      amd64                Boost C++ Libraries development files (ALL) (default version)
```
have you got an idea for fix this problem ?
the problem seems to be near like in redhat #178

Cdriko, 2018-05-25:
I finally fix this , adding the python header as python3 reference in FindBoost.cmake
in the function
function(_Boost_COMPONENT_HEADERS component _hdrs)

[as explained here](https://gitlab.kitware.com/cmake/cmake/merge_requests/201)

mkeeter, 2018-05-25:
Okay, glad you got this working!

closed 2018-05-25 by mkeeter