mkeeter/antimony #192: help on blend node

Cdriko, 2018-06-15:
hello
I try to use the blend node, to get the same result than this
[with shape.js](http://shapejs.shapeways.com/ide?shapeJsScript=/rrstatic/javascript/shapejs/2.0/scripts/gyroid.js)

but it seems to do anything totaly different.
I'm right ?

wich parameter can I set to get the same result?

mkeeter, 2018-06-15:
Can you share your Antimony code?

Cdriko, 2018-06-15:
I would like to reproduce the blending function used by shapejs

the source is here

[in /MathUtil.java line 1668](https://github.com/AbFab3D/AbFab3D/blob/master/src/java/abfab3d/core/MathUtil.java)

blendMin and Max

The result is the rounding of edges, wich its nicer and easyer to print.

not a morph kind like I see in antimony
[gyro.Variable1.zip](https://github.com/mkeeter/antimony/files/2106046/gyro.Variable1.zip)

Cdriko, 2018-06-15:
sorry it's line 1728 : public static final double blendQuadric

Cdriko, 2018-06-15:
I've tried to transpose the GLSL code, but I don't no exactly what put in the "output" function (follow doesn't work)
```
import fab

title('BlendQ')
input('a', fab.types.Shape)
input('b', fab.types.Shape)
input('q', float)

def blendQuadratic(x):
	return (1.0-x)*(1.0-x)*0.25

def blendMax(a,b,w):
	dd=max(a,b)
	if (w<=0.0):
		return dd
	d=abs(a-b)
	if(d<w):
		return dd+w*blendQuadratic(d/w)
	else:
		return dd

output('shape', 0.5*blendMax(a,b,q))
```

Cdriko, 2018-06-19:
in fact, I want to create the blend function defined in [libfive](http://libfive.com/examples/) examples : 
```
(define (blend a b m)
  (min a b (+ (sqrt (abs a))
              (sqrt (abs b))
              (- m)) ))
```
(the result is rounding joint edges)
but I fail to write it in correct F-rep syntax