mkeeter/antimony #194: compile error on mac:  Could not find boost::python3

Foadsf, 2018-07-15:
I am trying to compile on macOS High Sierra 10.13.6 but I get the error:

> CMake Error at CMakeLists.txt:36 (message):
>   Could not find boost::python3

I checked `brew install boost-python3` and got:

> Warning: boost-python3 1.67.0_1 is already installed and up-to-date
> To reinstall 1.67.0_1, run `brew reinstall boost-python3`

mkeeter, 2018-07-15:
Your install of `boost::python3` probably has a slightly different name that needs to be added to the search list – can you look for what the library is actually called, then open up a PR that adds it to [this line](https://github.com/mkeeter/antimony/blob/develop/CMakeLists.txt#L20)?

Foadsf, 2018-07-16:
@mkeeter I'm not sure what should I do. Specifically I do not understand what:

1.  > look for what the library is actually called
2.  > then open up a PR

mean. When I run `brew info boost-python3` this is what I get:
> boost-python3: stable 1.67.0 (bottled), HEAD
> C++ library for C++/Python3 interoperability
> https://www.boost.org/
> /usr/local/Cellar/boost-python3/1.67.0_1

mkeeter, 2018-07-16:
Can you see if 60a58688e552f12501980c4bdab034ab0f2ba059 fixes it?

Foadsf, 2018-07-17:
@mkeeter It solved the problem. Thanks a lot.

closed 2018-07-17 by mkeeter