mkeeter/antimony #195: Adding to the Windows and macOS package managers

Foadsf, 2018-07-17:
I just checked the [Repology page of Antimony](https://repology.org/metapackage/antimony/versions), there are some Linux distros (Arch/AUR, Fedora and nixOS) which have the app but no macOS or Windows ones. It would be a great help if you could make the HomeBrew and Chocolaty formulas at least. 

P.S. If I'm not mistaken nix package manager is extendable to macOS too.

ComFreek, 2018-07-18:
As Chocolatey (understandably) only deals with pre-built software, someone would need to host Windows binaries of Antimony.

Would it be possible to include Windows binaries on the release page?
I know this would complicate the build process which is currently done manually as it seems. Hence, I'd suggest also integrating continuous integration/builds on Travis CI (for *nix) and AppVeyor (for Windows).

Foadsf, 2018-07-18:
Chocolatey, is not the only package manager on Windows. There are also [Scoop](https://github.com/lukesampson/scoop), OneGet / NuGet, Npackd, win-get, WAPT. If I'm not mistaken Scoop tries to a be a replica of HomeBrew on Windows, hence should be able to compile from source I guess. There is also the [Zero Install](https://0install.de/) which is kind of a cross platform app store, very convenient for average user.

mkeeter, 2018-07-18:
I don't have the time to support five+ Windows package managers, especially because I'm developing on a Mac.  If folks would like to volunteer to set up CI, I'd be happy to add you as a collaborator to this repo.

ComFreek, 2018-07-19:
Yesterday I came up with the following initial AppVeyor script. Feel free to build upon it! I might have another look and complete it in circa one week.

```
version: 1.0.{build}
branches:
  only:
  - develop
clone_depth: 1
build_script:
- ps: >-
    $pacman = 'C:\msys64\usr\bin\pacman.exe'

    Start-Process $pacman -Args @('-Syuu') -Wait

    $msysSoftware = 'git make lemon flex mingw-w64-x86_64-python3 mingw-w64-x86_64-cmake mingw-w64-x86_64-qt5 mingw-w64-x86_64-toolchain'
    Start-Process $pacman -Args (@('-S', '-y') + $msysSoftware.Split(' ')) -Wait
```

There is a preinstalled Boost in `C:\Libraries\boost_1_67_0` in the default AppVeyor VM images. I am not sure whether it contains its libraries built with Python support (which Antimony requires).