mkeeter/antimony #196: Error during build on a Debian system

tiagovaz, 2018-08-17:
Hi, I can' t build antinomy in my Debian box due to the following error. This might be the relevant output line:

```
/home/user/Hacking/antimony/lib/fab/src/tree/v2syntax.y:43:32: error: invalid conversion from ‘yyParser*’ to ‘unsigned char’ [-fpermissive]
   int a = yy_find_shift_action(yypParser, (YYCODETYPE)i);
```

I've followed the instructions and installed the required deb packages.

Thanks, I've just discovered Antimony and I can't wait to start using it!

mkeeter, 2018-08-21:
I'm not sure about this, and don't have an Debian image to test this on.  Can you do some debugging yourself and open a PR if there's an obvious fix?

For background, this error looks like it's happening in the parser generator, which reads the math-like language.  This parser is implemented with [`lemon`](https://www.hwaci.com/sw/lemon/) and [`flex`](https://www.gnu.org/software/flex/).

gordon-quad, 2018-08-26:
I have a feeling that it might be a problem with versions of lemon and it is not an obvious fix. `yy_find_shift_action` in lemon [version](https://www.sqlite.org/src/artifact?ci=trunk&filename=tool/lempar.c) (grep by `yy_find_shit_action`) that I have takes `YYCODETYPE` as first argument not second.

But you can compile with CMAKE_BUILD_TYPE=Release and this error goes away since this code surrounded by ifndef.

tiagovaz, 2018-08-27:
Thanks @gordon-quad & @mkeeter. I confirm that building it against an older lemon version (3.16.2) everything goes fine. If I use a more recent one which is in Debian testing (3.24.0) then I get the error. They might had changed declaration at some point in between.

tiagovaz, 2018-11-15:
Antimony is now officially in Debian, so closing this issue.

closed 2018-11-15 by tiagovaz

l29ah, 2020-04-30:
I observe the same problem in Gentoo with the latest lemon release.

l29ah, 2020-04-30:
`cmake -DCMAKE_BUILD_TYPE=Release .` doesn't fix it:
```
/home/test/tmp/antimony/lib/fab/src/tree/v2syntax.y: In function ‘void yy_syntax_error(yyParser*, int, const char*)’:
/home/test/tmp/antimony/lib/fab/src/tree/v2syntax.y:43:32: error: invalid conversion from ‘yyParser*’ to ‘unsigned char’ [-fpermissive]
   43 |   int a = yy_find_shift_action(yypParser, (YYCODETYPE)i);
      |                                ^~~~~~~~~
      |                                |
      |                                yyParser*
```

mkeeter, 2020-05-01:
Does the patch in #215 fix it?