mkeeter/antimony #197: antimony doesn't build with cmake 3.11

rnhmjoj, 2018-09-07:
Antimony stopped building when cmake was updated from 3.10.2 to 3.11.2. This is the log:
```
builder for '/nix/store/w5rsz9wdinw8i8fphy8v6fcyx2464iih-antimony-2018-07-10.drv' failed with exit code 1; last 10 log lines:
  Call Stack (most recent call first):
    CMakeLists.txt:28 (find_package)
  
  
  CMake Error at CMakeLists.txt:36 (message):
    Could not find boost::python3
  
  
  -- Configuring incomplete, errors occurred!
  See also "/build/source/build/CMakeFiles/CMakeOutput.log".

```

[CMakeOutput.log](https://ipfs.io/ipfs/QmPysRqs6sUxqUMqb7bKJcrg4B4asUrMuQNDiu9rCQhpeW)

mkeeter, 2018-09-07:
Taken by itself, that doesn't look like a CMake issue – it looks like a problem finding libraries.

Can you look in your system to see what `boost::python3` is named, and open a PR that adds it to [this list](https://github.com/mkeeter/antimony/blob/develop/CMakeLists.txt#L20) (if that fixes the problem)?  I'm not sure why a CMake update would break things, but perhaps something else got updated or renamed at the same time.

mkeeter, 2018-09-08:
I won't be able to get into the weeds of debugging `nix` packages, perhaps you can ping the original packager?

My specific suggestion is to set up an environment with all of the dependencies installed, check in `/usr/local/lib` (or wherever NixOS puts libraries) for something like `libboost-python???`, then add that specific naming convention to the list of searched names.

rnhmjoj, 2018-09-10:
I realized python support has been disabled by default in the boost package since `libbost_python3.so` was missing. Something else must have inferfered when bisecting.
Thank you and sorry for the trouble.

closed 2018-09-10 by rnhmjoj