mkeeter/antimony #200: STL export misses subtraction of extruded text

kromanenko, 2018-10-17:
Tried to export to STL model of label table with carved (by subtraction) text.
While model in viewport and height-map image both has carved text, the STL-exported model looks like flat table without carving.

Zipped graph file, height map and zipped STL are attached.

[label.sb.txt](https://github.com/mkeeter/antimony/files/2489131/label.sb.txt)
[image]
[antimony.stl.zip](https://github.com/mkeeter/antimony/files/2489138/antimony.stl.zip)

mkeeter, 2018-10-18:
Have you tried increasing the mesh resolution?  I'd be suspicious that the thin text is just missed by the meshing sampling grid.

closed 2018-10-18 by kromanenko

kromanenko, 2018-10-18:
Btw, your viewer works fine!
[image]