mkeeter/antimony #201: dependency not met debian packaging

irazaq, 2018-10-18:
I was able to build antimony on debian stretch. When I tried to create a debian package got the following error:

dpkg-checkbuilddeps: error: Unmet build dependencies: debhelper (>= 9) desktop-file-utils

below is the full output:

dpkg-buildpackage: info: source package antimony
dpkg-buildpackage: info: source version 0.9.2
dpkg-buildpackage: info: source distribution testing
dpkg-buildpackage: info: source changed by Peter Felecan <peter@felecan.org>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source --before-build antimony
dpkg-checkbuilddeps: error: Unmet build dependencies: debhelper (>= 9) desktop-file-utils
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)

How do I resolve this?

mkeeter, 2018-10-18:
I'm not an expert on Debian packaging; that was contributed by Peter Felcan, who doesn't appear to be on Github anymore.

If you figure it out, feel free to open a PR.

irazaq, 2018-10-18:
I will definitely PR you when I find a solution.

irazaq, 2018-10-31:
Thanks pfelecan, the dependencies are already there. I found out I just have to install the missing dependencies which I assume are installed by default and it compile now. I was using a docker container to do this based on the work of rzetterberg https://github.com/rzetterberg/antimony-docker

tiagovaz, 2018-11-15:
Hi there, antimony has finally been accepted into Debian, so just apt install it from the official Sid suite: 
https://packages.debian.org/sid/antimony