mkeeter/antimony #203: I can't build antimony on ubuntu

gravit22, 2018-12-15:
I did everything along instructions. but this is what I have:
`gravit@Gravit-PC:~/antimony/build$ ninja
[16/130] Generating v2syntax.lemon.hpp, v2syntax.lemon.cpp
Parser statistics:
  terminal symbols...................    60
  non-terminal symbols...............     8
  total symbols......................    68
  rules..............................    58
  states.............................   159
  conflicts..........................     0
  action table entries...............  1936
  lookahead table entries............  1964
  total table size (bytes)...........  6594
[19/130] Building C object lib/fab/CMakeFiles/SbFab.dir/src/formats/png.c.o
../lib/fab/src/formats/png.c: In function ‘save_png16L’:
../lib/fab/src/formats/png.c:60:29: warning: ‘%g’ directive output may be truncated writing between 1 and 13 bytes into a region of size 10 [-Wformat-truncation=]
         snprintf(zmin, 10, "%g", bounds[2]);
                             ^~
../lib/fab/src/formats/png.c:60:28: note: assuming directive output of 12 bytes
         snprintf(zmin, 10, "%g", bounds[2]);
                            ^~~~
In file included from /usr/include/stdio.h:862:0,
                 from /usr/include/pngconf.h:45,
                 from /usr/include/png.h:366,
                 from ../lib/fab/src/formats/png.c:7:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 2 and 14 bytes into a destination of size 10
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../lib/fab/src/formats/png.c:66:29: warning: ‘%g’ directive output may be truncated writing between 1 and 13 bytes into a region of size 10 [-Wformat-truncation=]
         snprintf(zmax, 10, "%g", bounds[5]);
                             ^~
../lib/fab/src/formats/png.c:66:28: note: assuming directive output of 12 bytes
         snprintf(zmax, 10, "%g", bounds[5]);
                            ^~~~
In file included from /usr/include/stdio.h:862:0,
                 from /usr/include/pngconf.h:45,
                 from /usr/include/png.h:366,
                 from ../lib/fab/src/formats/png.c:7:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:64:10: note: ‘__builtin___snprintf_chk’ output between 2 and 14 bytes into a destination of size 10
   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        __bos (__s), __fmt, __va_arg_pack ());
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[39/130] Linking CXX static library lib/graph/libSbGraph.a
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
[42/130] Linking CXX executable lib/graph/SbGraphTest
FAILED: lib/graph/SbGraphTest 
: && /usr/bin/c++  -Wall -Wextra -g -Werror=switch -O3 -DRELEASE  -rdynamic lib/graph/CMakeFiles/SbGraphTest.dir/tests/datum.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/graph.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/link.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/main.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/node.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/script.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/subgraph.cpp.o  -o lib/graph/SbGraphTest  lib/graph/libSbGraph.a -lboost_python3 /usr/local/lib/libpython3.7m.a && :
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(signalmodule.o): undefined reference to symbol 'pthread_sigmask@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
[47/130] Building CXX object lib/fab/C....dir/src/tree/triangulate/mesher.cpp.o
../lib/fab/src/tree/triangulate/mesher.cpp: In member function ‘std::__cxx11::list<Eigen::Matrix<double, 3, 1> > Mesher::get_normals(const std::__cxx11::list<Eigen::Matrix<double, 3, 1> >&)’:
../lib/fab/src/tree/triangulate/mesher.cpp:137:15: warning: variable ‘v’ set but not used [-Wunused-but-set-variable]
     for (auto v : points)
               ^
../lib/fab/src/tree/triangulate/mesher.cpp: In member function ‘void Mesher::interpolate_between(const Vec3f&, const Vec3f&)’:
../lib/fab/src/tree/triangulate/mesher.cpp:632:61: warning: missing initializer for member ‘InterpolateCommand::cached’ [-Wmissing-field-initializers]
         .cmd=InterpolateCommand::INTERPOLATE, .v0=v0, .v1=v1};
                                                             ^
../lib/fab/src/tree/triangulate/mesher.cpp: In member function ‘void Mesher::triangulate_region(const Region&)’:
../lib/fab/src/tree/triangulate/mesher.cpp:742:58: warning: missing initializer for member ‘InterpolateCommand::v0’ [-Wmissing-field-initializers]
                     .cmd=InterpolateCommand::END_OF_VOXEL});
                                                          ^
../lib/fab/src/tree/triangulate/mesher.cpp:742:58: warning: missing initializer for member ‘InterpolateCommand::v1’ [-Wmissing-field-initializers]
../lib/fab/src/tree/triangulate/mesher.cpp:742:58: warning: missing initializer for member ‘InterpolateCommand::cached’ [-Wmissing-field-initializers]
ninja: build stopped: subcommand failed.
`

mkeeter, 2018-12-15:
The line that matters is 
```
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(signalmodule.o): undefined reference to symbol 'pthread_sigmask@@GLIBC_2.2.5'
```
Can you re-run with a verbose build (I think that's `ninja -v`) and see if it's passing the appropriate flags to link `pthreads`?

gravit22, 2018-12-16:
Oh sorry. Here is the full output
`gravit@Gravit-PC:~/antimony/build$ ninja -v
[1/78] : && /usr/bin/cmake -E remove lib/fab/libSbFab.a && /usr/bin/ar qc lib/fab/libSbFab.a  lib/fab/CMakeFiles/SbFab.dir/src/fab.cpp.o lib/fab/CMakeFiles/SbFab.dir/src/formats/png.c.o lib/fab/CMakeFiles/SbFab.dir/src/formats/stl.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/eval.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_f.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_g.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_i.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_r.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/node/node_c.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/node/opcodes.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/node/printers.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/node/printers_ss.cpp.o lib/fab/CMakeFiles/SbFab.dir/src/tree/node/results.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/parser.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/render.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/tree.c.o lib/fab/CMakeFiles/SbFab.dir/src/tree/v2parser.cpp.o lib/fab/CMakeFiles/SbFab.dir/src/tree/triangulate/mesher.cpp.o lib/fab/CMakeFiles/SbFab.dir/src/tree/triangulate/triangle.cpp.o lib/fab/CMakeFiles/SbFab.dir/src/tree/triangulate.cpp.o lib/fab/CMakeFiles/SbFab.dir/src/types/bounds.cpp.o lib/fab/CMakeFiles/SbFab.dir/src/types/shape.cpp.o lib/fab/CMakeFiles/SbFab.dir/src/types/transform.cpp.o lib/fab/CMakeFiles/SbFab.dir/src/util/region.c.o lib/fab/CMakeFiles/SbFab.dir/src/util/ustack.c.o lib/fab/CMakeFiles/SbFab.dir/v2syntax.lemon.cpp.o lib/fab/CMakeFiles/SbFab.dir/v2syntax.yy.cpp.o && /usr/bin/ranlib lib/fab/libSbFab.a && :
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
[2/78] : && /usr/bin/c++  -Wall -Wextra -g -Werror=switch -O3 -DRELEASE  -rdynamic lib/graph/CMakeFiles/SbGraphTest.dir/tests/datum.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/graph.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/link.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/main.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/node.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/script.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/subgraph.cpp.o  -o lib/graph/SbGraphTest  lib/graph/libSbGraph.a -lboost_python3 /usr/local/lib/libpython3.7m.a && :
FAILED: lib/graph/SbGraphTest 
: && /usr/bin/c++  -Wall -Wextra -g -Werror=switch -O3 -DRELEASE  -rdynamic lib/graph/CMakeFiles/SbGraphTest.dir/tests/datum.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/graph.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/link.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/main.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/node.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/script.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/subgraph.cpp.o  -o lib/graph/SbGraphTest  lib/graph/libSbGraph.a -lboost_python3 /usr/local/lib/libpython3.7m.a && :
/usr/bin/ld: /usr/local/lib/libpython3.7m.a(signalmodule.o): undefined reference to symbol 'pthread_sigmask@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
[3/78] /usr/bin/c++   -isystem ../lib/fab/../../vendor -isystem /usr/local/include/python3.7m -I../lib/fab/inc -Wall -Wextra -g -Werror=switch -O3 -DRELEASE   -std=gnu++11 -MD -MT lib/fab/CMakeFiles/SbFabTest.dir/tests/shape.cpp.o -MF lib/fab/CMakeFiles/SbFabTest.dir/tests/shape.cpp.o.d -o lib/fab/CMakeFiles/SbFabTest.dir/tests/shape.cpp.o -c ../lib/fab/tests/shape.cpp
[4/78] cd /home/gravit/antimony/build/app && /usr/bin/cmake -E cmake_autogen /home/gravit/antimony/build/app/CMakeFiles/antimony_autogen.dir RELEASE
/usr/bin/cmake: /usr/local/lib/libcurl.so.4: no version information available (required by /usr/bin/cmake)
ninja: build stopped: subcommand failed.
`

johhnry, 2020-01-17:
Hi there,

I have the same issue on Pop!_OS 19.10 (highly derived from Ubuntu) : 
<details><summary>Ninja output</summary>
<p>

```
$ ninja 
[16/131] Generating v2syntax.lemon.hpp, v2syntax.lemon.cpp
Parser statistics:
  terminal symbols...................    60
  non-terminal symbols...............     7
  total symbols......................    67
  rules..............................    58
  states.............................   159
  conflicts..........................     0
  action table entries...............  1936
  lookahead table entries............  1964
  total table size (bytes)...........  6594
[22/131] Building C object lib/fab/CMakeFiles/SbFab.dir/src/formats/png.c.o
../lib/fab/src/formats/png.c: In function ‘save_png16L’:
../lib/fab/src/formats/png.c:60:29: warning: ‘%g’ directive output may be truncated writing between 1 and 13 bytes into a region of size 10 [-Wformat-truncation=]
   60 |         snprintf(zmin, 10, "%g", bounds[2]);
      |                             ^~
../lib/fab/src/formats/png.c:60:28: note: assuming directive output of 12 bytes
   60 |         snprintf(zmin, 10, "%g", bounds[2]);
      |                            ^~~~
In file included from /usr/include/stdio.h:867,
                 from /usr/include/pngconf.h:46,
                 from /usr/include/png.h:339,
                 from ../lib/fab/src/formats/png.c:7:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:67:10: note: ‘__builtin___snprintf_chk’ output between 2 and 14 bytes into a destination of size 10
   67 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   68 |        __bos (__s), __fmt, __va_arg_pack ());
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../lib/fab/src/formats/png.c:66:29: warning: ‘%g’ directive output may be truncated writing between 1 and 13 bytes into a region of size 10 [-Wformat-truncation=]
   66 |         snprintf(zmax, 10, "%g", bounds[5]);
      |                             ^~
../lib/fab/src/formats/png.c:66:28: note: assuming directive output of 12 bytes
   66 |         snprintf(zmax, 10, "%g", bounds[5]);
      |                            ^~~~
In file included from /usr/include/stdio.h:867,
                 from /usr/include/pngconf.h:46,
                 from /usr/include/png.h:339,
                 from ../lib/fab/src/formats/png.c:7:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:67:10: note: ‘__builtin___snprintf_chk’ output between 2 and 14 bytes into a destination of size 10
   67 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   68 |        __bos (__s), __fmt, __va_arg_pack ());
      |        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[49/131] Building CXX object lib/fab/CMakeFiles/SbFab.dir/v2syntax.lemon.cpp.o
FAILED: lib/fab/CMakeFiles/SbFab.dir/v2syntax.lemon.cpp.o 
/usr/bin/c++   -Ilib/fab -I../lib/fab/inc -isystem /usr/include/python3.7m -isystem ../lib/fab/vendor -Wall -Wextra -g -Werror=switch -O3 -DRELEASE   -std=gnu++11 -MD -MT lib/fab/CMakeFiles/SbFab.dir/v2syntax.lemon.cpp.o -MF lib/fab/CMakeFiles/SbFab.dir/v2syntax.lemon.cpp.o.d -o lib/fab/CMakeFiles/SbFab.dir/v2syntax.lemon.cpp.o -c lib/fab/v2syntax.lemon.cpp
/home/josephhenry/Téléchargements/antimony/lib/fab/src/tree/v2syntax.y:7: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
    7 |  #pragma clang diagnostic ignored "-Wunused-parameter"
      | 
/home/josephhenry/Téléchargements/antimony/lib/fab/src/tree/v2syntax.y:8: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
    8 |  #pragma clang diagnostic ignored "-Wunused-variable"
      | 
/home/josephhenry/Téléchargements/antimony/lib/fab/src/tree/v2syntax.c: In function ‘void yy_destructor(yyParser*, unsigned char, YYMINORTYPE*)’:
/home/josephhenry/Téléchargements/antimony/lib/fab/src/tree/v2syntax.c:123:31: warning: unused variable ‘environment’ [-Wunused-variable]
/home/josephhenry/Téléchargements/antimony/lib/fab/src/tree/v2syntax.c:1002:3: note: in expansion of macro ‘v2ParseARG_FETCH’
/home/josephhenry/Téléchargements/antimony/lib/fab/src/tree/v2syntax.c:1000:16: warning: unused parameter ‘yypminor’ [-Wunused-parameter]
/home/josephhenry/Téléchargements/antimony/lib/fab/src/tree/v2syntax.y: In function ‘void yy_syntax_error(yyParser*, int, const char*)’:
/home/josephhenry/Téléchargements/antimony/lib/fab/src/tree/v2syntax.y:43:32: error: invalid conversion from ‘yyParser*’ to ‘unsigned char’ [-fpermissive]
   43 |   int a = yy_find_shift_action(yypParser, (YYCODETYPE)i);
      |                                ^~~~~~~~~
      |                                |
      |                                yyParser*
[... 11 lines cut ...]
    9 |  #pragma clang diagnostic ignored "-Wunused-function"
      | 
/home/josephhenry/Téléchargements/antimony/build/lib/fab/v2syntax.yy.cpp:1497:17: warning: ‘void yyunput(int, char*, yyscan_t)’ defined but not used [-Wunused-function]
 1497 |     static void yyunput (int c, char * yy_bp , yyscan_t yyscanner)
      |                 ^~~~~~~
[54/131] Building CXX object lib/fab/CMakeFiles/SbFab.dir/src/tree/triangulate/mesher.cpp.o
../lib/fab/src/tree/triangulate/mesher.cpp: In member function ‘std::__cxx11::list<Eigen::Matrix<double, 3, 1> > Mesher::get_normals(const std::__cxx11::list<Eigen::Matrix<double, 3, 1> >&)’:
[... cut ...]
../lib/fab/src/tree/triangulate/mesher.cpp:742:58: warning: missing initializer for member ‘InterpolateCommand::v0’ [-Wmissing-field-initializers]
  742 |                     .cmd=InterpolateCommand::END_OF_VOXEL});
      |                                                          ^
../lib/fab/src/tree/triangulate/mesher.cpp:742:58: warning: missing initializer for member ‘InterpolateCommand::v1’ [-Wmissing-field-initializers]
../lib/fab/src/tree/triangulate/mesher.cpp:742:58: warning: missing initializer for member ‘InterpolateCommand::cached’ [-Wmissing-field-initializers]
ninja: build stopped: subcommand failed.
```

</p>
</details>