mkeeter/antimony #204: Cannot build antimony on Mint Linux

B1uemax, 2019-01-16:
I'm following the build instruction up until cmake -GNinja ..
This is the output:
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python3; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:28 (find_package)
CMake Error at app/CMakeLists.txt:12 (find_package):
  By not providing "FindQt5OpenGL.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Qt5OpenGL", but CMake did not find one.
  Could not find a package configuration file provided by "Qt5OpenGL" with
  any of the following names:
    Qt5OpenGLConfig.cmake
    qt5opengl-config.cmake
  Add the installation prefix of "Qt5OpenGL" to CMAKE_PREFIX_PATH or set
  "Qt5OpenGL_DIR" to a directory containing one of the above files.  If
  "Qt5OpenGL" provides a separate development package or SDK, be sure it has
  been installed.
-- Configuring incomplete, errors occurred!
See also "/home/bluemax/antimony/build/CMakeFiles/CMakeOutput.log".

mkeeter, 2019-01-16:
Try to specify `CMAKE_PREFIX_PATH`, e.g. from the Mac example:

```
cmake -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.10.1 -GNinja ..
```
(adjust for your local Qt install path).

B1uemax, 2019-01-17:
I tried cmake -DCMAKE_PREFIX_PATH=/home/bluemax/Qt5.12.0/5.12.0/ -GNinja ..
CMake Error: The source directory "/home" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
*** then I performed a search for cmakelist.txt and included some specific folders, but received the same error

mkeeter, 2019-01-17:
What directory did you run this in?  You need to be in the Antimony build directory when you invoke the command, so that it finds the top-level `CMakeLists.txt` in the `antimony` repo folder.

B1uemax, 2019-01-18:
bluemax@bluebox:~/antimony/build$ cmake -DCMAKE_PREFIX_PATH=/usr/share/qt5 -GNinja ..
**CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python3; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:28 (find_package)
CMake Error at app/CMakeLists.txt:12 (find_package):
  By not providing "FindQt5OpenGL.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Qt5OpenGL", but CMake did not find one.
  Could not find a package configuration file provided by "Qt5OpenGL" with
  any of the following names:
    Qt5OpenGLConfig.cmake
    qt5opengl-config.cmake
 prefix of "Qt5OpenGL" to CMAKE_PREFIX_PATH or set
  "Qt5OpenGL_DIR" to a directory containing one of the above files.  If
  "Qt5OpenGL" provides a separate development package or SDK, be sure it has
  been installed.
-- Configuring incomplete, errors occurred!
See also "/home/bluemax/antimony/build/CMakeFiles/CMakeOutput.log".

mkeeter, 2019-01-19:
I don't have a Mint system to debug this myself, so you'll need to be more pro-active in hunting this down.

Here are the [docs for Qt + CMake](http://doc.qt.io/qt-5/cmake-manual.html).

> In order for find_package to be successful, Qt 5 must be found below the CMAKE_PREFIX_PATH, or the Qt5<Module>_DIR must be set in the CMake cache to the location of the Qt5WidgetsConfig.cmake file. The easiest way to use CMake is to set the CMAKE_PREFIX_PATH environment variable to the install prefix of Qt 5.

Is `/usr/share/qt5` the "install prefix" of Qt 5?

Another theory:
It's not failing to find Qt5Core or Qt5Widgets ([here](https://github.com/mkeeter/antimony/blob/develop/app/CMakeLists.txt#L10-L11)), which makes me suspicious – does your distro have a separate package for Qt5OpenGL?

thetomorrowman, 2019-07-04:
I'm having issues also

dh_auto_configure: cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var returned exit code 1
debian/rules:8: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2

Any thoughts?

mkeeter, 2019-07-04:
Can you post the complete log?  Are you trying to build the app or make a package?

The [build instructions](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md#linux) don't mention `dh_auto_configure` at all, so I'm not sure where that's coming from.

thetomorrowman, 2019-07-05:
Hi , i tried to follow the build but it wouldn't work so i went down the Debian Package route as follows
this seems to repeat a few times before it spits it's dummy out

Run Build Command:"/usr/bin/make" "cmTC_6175f/fast"
make[1]: Entering directory '/home/graeme/antimony/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_6175f.dir/build.make CMakeFiles/cmTC_6175f.dir/build
make[2]: Entering directory '/home/graeme/antimony/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_6175f.dir/feature_tests.cxx.o
/usr/bin/x86_64-linux-gnu-g++     -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2     -std=c++98 -o CMakeFiles/cmTC_6175f.dir/feature_tests.cxx.o -c /home/graeme/antimony/obj-x86_64-linux-gnu/CMakeFiles/feature_tests.cxx
Linking CXX executable cmTC_6175f
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_6175f.dir/link.txt --verbose=1
/usr/bin/x86_64-linux-gnu-g++   -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2     -Wl,-Bsymbolic-functions -Wl,-z,relro  CMakeFiles/cmTC_6175f.dir/feature_tests.cxx.o  -o cmTC_6175f -rdynamic 
make[2]: Leaving directory '/home/graeme/antimony/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
make[1]: Leaving directory '/home/graeme/antimony/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'

    Feature record: CXX_FEATURE:0cxx_aggregate_default_initializers
    Feature record: CXX_FEATURE:0cxx_alias_templates
    Feature record: CXX_FEATURE:0cxx_alignas
    Feature record: CXX_FEATURE:0cxx_alignof
    Feature record: CXX_FEATURE:0cxx_attributes
    Feature record: CXX_FEATURE:0cxx_attribute_deprecated
    Feature record: CXX_FEATURE:0cxx_auto_type
    Feature record: CXX_FEATURE:0cxx_binary_literals
    Feature record: CXX_FEATURE:0cxx_constexpr
    Feature record: CXX_FEATURE:0cxx_contextual_conversions
    Feature record: CXX_FEATURE:0cxx_decltype
    Feature record: CXX_FEATURE:0cxx_decltype_auto
    Feature record: CXX_FEATURE:0cxx_decltype_incomplete_return_types
    Feature record: CXX_FEATURE:0cxx_default_function_template_args
    Feature record: CXX_FEATURE:0cxx_defaulted_functions
    Feature record: CXX_FEATURE:0cxx_defaulted_move_initializers
    Feature record: CXX_FEATURE:0cxx_delegating_constructors
    Feature record: CXX_FEATURE:0cxx_deleted_functions
    Feature record: CXX_FEATURE:0cxx_digit_separators
    Feature record: CXX_FEATURE:0cxx_enum_forward_declarations
    Feature record: CXX_FEATURE:0cxx_explicit_conversions
    Feature record: CXX_FEATURE:0cxx_extended_friend_declarations
    Feature record: CXX_FEATURE:0cxx_extern_templates
    Feature record: CXX_FEATURE:0cxx_final
    Feature record: CXX_FEATURE:0cxx_func_identifier
    Feature record: CXX_FEATURE:0cxx_generalized_initializers
    Feature record: CXX_FEATURE:0cxx_generic_lambdas
    Feature record: CXX_FEATURE:0cxx_inheriting_constructors
    Feature record: CXX_FEATURE:0cxx_inline_namespaces
    Feature record: CXX_FEATURE:0cxx_lambdas
    Feature record: CXX_FEATURE:0cxx_lambda_init_captures
    Feature record: CXX_FEATURE:0cxx_local_type_template_args
    Feature record: CXX_FEATURE:0cxx_long_long_type
    Feature record: CXX_FEATURE:0cxx_noexcept
    Feature record: CXX_FEATURE:0cxx_nonstatic_member_init
    Feature record: CXX_FEATURE:0cxx_nullptr
    Feature record: CXX_FEATURE:0cxx_override
    Feature record: CXX_FEATURE:0cxx_range_for
    Feature record: CXX_FEATURE:0cxx_raw_string_literals
    Feature record: CXX_FEATURE:0cxx_reference_qualified_functions
    Feature record: CXX_FEATURE:0cxx_relaxed_constexpr
    Feature record: CXX_FEATURE:0cxx_return_type_deduction
    Feature record: CXX_FEATURE:0cxx_right_angle_brackets
    Feature record: CXX_FEATURE:0cxx_rvalue_references
    Feature record: CXX_FEATURE:0cxx_sizeof_member
    Feature record: CXX_FEATURE:0cxx_static_assert
    Feature record: CXX_FEATURE:0cxx_strong_enums
    Feature record: CXX_FEATURE:1cxx_template_template_parameters
    Feature record: CXX_FEATURE:0cxx_thread_local
    Feature record: CXX_FEATURE:0cxx_trailing_return_types
    Feature record: CXX_FEATURE:0cxx_unicode_literals
    Feature record: CXX_FEATURE:0cxx_uniform_initialization
    Feature record: CXX_FEATURE:0cxx_unrestricted_unions
    Feature record: CXX_FEATURE:0cxx_user_literals
    Feature record: CXX_FEATURE:0cxx_variable_templates
    Feature record: CXX_FEATURE:0cxx_variadic_macros
    Feature record: CXX_FEATURE:0cxx_variadic_templates
	cd /home/graeme/antimony
dh_auto_configure: cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var returned exit code 1
debian/rules:8: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2

mkeeter, 2019-07-05:
I'm sorry, but I don't maintain the Debian package – those are user-contributed, and may have bit-rotted.  If you'd like to follow the build instructions in the README and post a complete log, I can take a look.

thetomorrowman, 2019-07-06:
it's probably obvious to those in the know...........
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable version "3.5.2", minimum required is "3.3") 
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.2.54") 
-- Found FLEX: /usr/bin/flex (found version "2.6.0") 
CMake Error at app/CMakeLists.txt:12 (find_package):
  By not providing "FindQt5OpenGL.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Qt5OpenGL", but CMake did not find one.

  Could not find a package configuration file provided by "Qt5OpenGL" with
  any of the following names:

    Qt5OpenGLConfig.cmake
    qt5opengl-config.cmake

  Add the installation prefix of "Qt5OpenGL" to CMAKE_PREFIX_PATH or set
  "Qt5OpenGL_DIR" to a directory containing one of the above files.  If
  "Qt5OpenGL" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/graeme/antimony/build/CMakeFiles/CMakeOutput.log".

mkeeter, 2019-07-06:
Yup, it looks like this is the exact same problem as above.  Please refer to [this comment](https://github.com/mkeeter/antimony/issues/204#issuecomment-455783333) for next debugging steps.

mbd-shift, 2020-09-17:
looks like you might want to install qt5opengl5-dev instead.
Worked for me on mx linux 19