mkeeter/antimony #208: Failed to build under Ubuntu 18.04

Renha, 2019-11-21:
Tried building Antimony under Ubuntu 18.04, cmake gives me error with or without Ninja:
```
CMake Error at CMakeLists.txt:11 (find_package):
  By not providing "FindPython.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Python", but
  CMake did not find one.

  Could not find a package configuration file provided by "Python" (requested
  version 3.3) with any of the following names:

    PythonConfig.cmake
    python-config.cmake

  Add the installation prefix of "Python" to CMAKE_PREFIX_PATH or set
  "Python_DIR" to a directory containing one of the above files.  If "Python"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```

Renha, 2019-11-21:
Versions prior to commit 222ae890da81a1eacf5c528598e79a91c4bab58a work.

mkeeter, 2019-11-21:
You may need a newer version of `cmake` – it looks like the new Python finding functionality is in 3.12 and later.

spider65, 2019-11-26:
I have the same problem. 
I installed the latest version of cmake (3.13) but the problem remained.

spider65, 2019-11-26:
I no longer have the cmake problem, but now i have this problem

-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python: /usr/local/bin/python3.7 (found suitable version "3.7.4", minimum required is "3.3") found components:  Interpreter Development 
CMake Error at /home/raniero/.local/lib/python3.6/site-packages/cmake/data/share/cmake-3.13/Modules/FindBoost.cmake:2100 (message):
  Unable to find the requested Boost libraries.

  Boost version: 1.65.1

  Boost include path: /usr/include

  Could not find the following Boost libraries:

          boost_python37

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.
Call Stack (most recent call first):
  CMakeLists.txt:17 (find_package)

CMake Error at CMakeLists.txt:20 (message):
  Could not find boost::python3

-- Configuring incomplete, errors occurred!
See also "/home/raniero/antimony/build/CMakeFiles/CMakeOutput.log".

mkeeter, 2019-11-26:
Have you installed Boost's Python 3 bindings?  Last time I checked, they were part of `libboost-all-dev`, but that may not be true anymore – if that's not the case, please open a PR against [`BUILDING.md`](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md).

Fellfalla, 2020-03-12:
Try to check out the **[master](https://github.com/mkeeter/antimony/tree/master) branch** and build that one.

1. Go to your build directory
2. `git checkout -t origin/master`
3. `cmake -GNinja ..`
4. `ninja`
5. `./app/antimony`

spider65, 2020-03-12:
Hi Matt,
I solved the problem.
I manager to installa Antimony

Best
Raniero

mkeeter, 2020-03-13:
Great, thanks for the update!  Anything you'd suggest adding to the README?

closed 2020-03-13 by mkeeter