mkeeter/antimony #209: [Q] About project

trsh, 2019-12-06:
What isosurface Extraction method are you using?

dcanelhas, 2019-12-06:
I'm guessing it's marching tetrahedrons. But have a look at https://github.com/mkeeter/antimony/blob/develop/lib/fab/src/tree/triangulate/mesher.cpp if you're interested in the details.

mkeeter, 2019-12-07:
This is marching tetrahedrons, plus the algorithm from "Feature Sensitive Surface Extraction from Volume Data" to drop an extra vertex on detected edges and vertices.

My more recent project, [libfive](https://libfive.com), is using Manifold Dual Contouring, which is more advanced + robust, but still not perfect.

trsh, 2019-12-07:
OK, Thank you for the info.

closed 2019-12-07 by trsh

trsh, 2019-12-07:
@mkeeter question, Why you don't use Dual Marching Cubes? The papers promise manifold surfaces and simplified mesh :)

mkeeter, 2019-12-07:
I tried it, but it doesn't work as well as the paper implies.  I've also tried "Isosurfaces Over Simplicial Partitions of Multiresolution Grids", which runs into similar issues.  The root cause is that a straight line between two features on your model's surface *isn't* guaranteed to intersect that feature at all points along the line, so you end up with "crinkly" edges.

trsh, 2019-12-07:
MKay! Good to know such info

trsh, 2019-12-07:
@mkeeter you class is very popular along the internet https://github.com/mkeeter/kokopelli/blob/master/libfab/asdf/cms.c , as the only readable reference for the method CMS. Can you comment on the problems with that - why you dropped for your newer projects?

mkeeter, 2019-12-08:
CMS is a weird, complicated algorithm – I implemented about half of it, and am still not sure if I got it totally right.  The paper doesn't give enough information to fully reproduce it, e.g. a section which ends with:
>We use a dynamic programming algorithm to connect and triangulate these two components to form the surface.

trsh, 2019-12-08:
@mkeeter true. And the provided LibCMS is marked "will be available soon" since 2005 :D