mkeeter/antimony #210: libboost_python36.so.1.67.0: cannot open shared object file

Pupo1, 2020-01-16:
Hello, 

on Debian 10 I get

`antimony: error while loading shared libraries: libboost_python36.so.1.67.0: cannot open shared object file: No such file or directory
`

Linux 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11) x86_64 GNU/Linux

regards, Pupo

mkeeter, 2020-01-17:
What bindings to boost::python3 have you installed?

Last time I checked, they were part of `libboost-all-dev`, but that may not be true anymore – if that's not the case, please open a PR against [`BUILDING.md`](https://github.com/mkeeter/antimony/blob/develop/BUILDING.md).

Pupo1, 2020-01-17:
Sorry, I have no idea how I check this. What command should I use to check the python3 binding?

mkeeter, 2020-01-18:
I can't debug your specific Linux system, but try looking around in `/usr/lib`, `/usr/local/lib`, etc, for something that looks like `libboost_python.*`.

Pupo1, 2020-01-19:
>    [1]BUILDING.md.
> References
>    Visible links
> /home/pulmo/.elinks//elinks.conf:3: unknown command

Thank you for the hint. What do you mean with PR, a pull request?
I installed libboost-all-dev, and find the following files looking similar to what you said:
./lib/x86_64-linux-gnu/libboost_python3-py37.so
./lib/x86_64-linux-gnu/libboost_mpi_python37.so.1.67.0
./lib/x86_64-linux-gnu/libboost_mpi_python3-py37.a
./lib/x86_64-linux-gnu/libboost_python37.a
./lib/x86_64-linux-gnu/libboost_mpi_python3-py37.so
./lib/x86_64-linux-gnu/libboost_python37.so.1.67.0
./lib/x86_64-linux-gnu/libboost_python3-py37.a

I don't know how to write a pull request, because I never did it... :-(

Best regards, Bernd

orthzar, 2020-01-19:
Bernd,

I had the exact same issue on Debian 11 (Bullseye).  However, Antimony is now working for me after a recent update.

From the Debian [patches list](https://sources.debian.org/patches/antimony/), 0.9.3-1.1 resolves the issue.  Debian 11 and Sid have that patch, but Debian 10 (Buster) does not.

Your options are:

- Upgrade to Debian 11;
- Add Debian 11 to your repository sources list and only upgrade Antimony; or
- Politely ask [Tiago](https://qa.debian.org/developer.php?login=tiago%40debian.org) to get the 0.9.3-1.1 patch added to Debian 10.

Regards,
orthzar

P.S.  If Tiago sees this: thanks!  I can now proceed with a project that I've been putting off.

mkeeter, 2020-01-20:
Great, I'm going to close this issue as it's been resolved in newer versions of Debian.

closed 2020-01-20 by mkeeter

Pupo1, 2020-01-20:
I still wanted to say thank you. to orthzar. I will just upgrade to Debian 11.

ceremcem, 2021-01-14:
Anyone has a ready-to-apply solution on Debian Buster? Can we download current package and apply that patch manually, and then `apt install antimony-patched.deb`?

ceremcem, 2021-01-15:
FYI: Upgrading to Debian 10 was not an option to me. I followed the following path instead: 

1. Create an LXC container: 

       $ sudo lxc-create -n antimony -t debian -B btrfs -- -r bullseye --packages xbase-clients nano sudo
       $ sudo lxc-attach -n antimony
       antimony:~# adduser myuser

2. Install Antimony via package manager:
  
       antimony:~# apt-get install antimony

3. Run Antimony over ssh: 

       ssh -X myuser@ip-of-lxc-container antimony

nerdCopter, 2021-04-14:
if only `Antimony` came in .AppImage :thinking:  ...
same boat, Debian 10 Stable + buster-backports, still an issue.