mkeeter/antimony #211: Build failure with Cmake 3.16

rnhmjoj, 2020-02-06:
I maintain the [nixpkgs](https://github.com/NixOS/nixpkgs) package for antimony, which broke [recently](https://hydra.nixos.org/build/109447088).
After bisecting it appears the problem has been introduced with the upgrade to cmake 3.16.2.
The [build log](https://hastebin.com/raw/zopetimuve) pretty much says cmake found python but the compiler still can't find `Python.h`.

Bomi0, 2020-02-28:
Encountered a similar problem that I could not find Python.h when I typed ninja to build the program on Ubuntu

TheBeachLab, 2020-03-25:
Same in arch linux

mkeeter, 2020-03-28:
Do folks want to see if this is fixed by PR #212 ?

rnhmjoj, 2020-03-28:
@mkeeter Thank you, I'll be sure to check it as soon as I have some spare time.

TheBeachLab, 2020-03-28:
In arch #212 fixed it, thx @havardAasen 

I also had to create this empty folder: `/include` otherwise it complains. But could be just my system

```
CMake Error in lib/fab/CMakeLists.txt:
  Imported target "Boost::python" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

rnhmjoj, 2020-03-28:
I also confirm that was it. I didn't have to make a include folder, though. Maybe it's being created Nix.

closed 2020-03-28 by rnhmjoj