mkeeter/antimony #214: ninja: build stopped: subcommand failed. (Ubuntu)

Bomi0, 2020-03-28:
Problem while building the program on Ubuntu;p

[2/18] Linking CXX executable lib/fab/SbFabTest
FAILED: lib/fab/SbFabTest 
: && /usr/bin/c++  -Wall -Wextra -g -Werror=switch -O3 -DRELEASE  -rdynamic lib/fab/CMakeFiles/SbFabTest.dir/tests/main.cpp.o lib/fab/CMakeFiles/SbFabTest.dir/tests/parser.cpp.o lib/fab/CMakeFiles/SbFabTest.dir/tests/shape.cpp.o  -o lib/fab/SbFabTest  lib/fab/libSbFab.a  -lboost_python3  /usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a  -lpng  -lz && :
/usr/bin/ld: /usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(thread.o): undefined reference to symbol 'pthread_condattr_setclock@@GLIBC_2.3.3'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
[3/18] Linking CXX executable lib/graph/SbGraphTest
FAILED: lib/graph/SbGraphTest 
: && /usr/bin/c++  -Wall -Wextra -g -Werror=switch -O3 -DRELEASE  -rdynamic lib/graph/CMakeFiles/SbGraphTest.dir/tests/datum.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/graph.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/link.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/main.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/node.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/script.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/subgraph.cpp.o  -o lib/graph/SbGraphTest  lib/graph/libSbGraph.a  -lboost_python3  /usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a && :
/usr/bin/ld: /usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(thread.o): undefined reference to symbol 'pthread_condattr_setclock@@GLIBC_2.3.3'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
[13/18] Building CXX object app/CMakeF.../antimony.dir/export/export_mesh.cpp.o
../app/export/export_mesh.cpp: In member function ‘virtual void ExportMeshWorker::async()’:
../app/export/export_mesh.cpp:79:5: warning: missing initializer for member ‘Region_::voxels’ [-Wmissing-field-initializers]
     };
     ^
../app/export/export_mesh.cpp:79:5: warning: missing initializer for member ‘Region_::X’ [-Wmissing-field-initializers]
../app/export/export_mesh.cpp:79:5: warning: missing initializer for member ‘Region_::Y’ [-Wmissing-field-initializers]
../app/export/export_mesh.cpp:79:5: warning: missing initializer for member ‘Region_::Z’ [-Wmissing-field-initializers]
../app/export/export_mesh.cpp:79:5: warning: missing initializer for member ‘Region_::L’ [-Wmissing-field-initializers]
[14/18] Building CXX object app/CMakeF...mony.dir/export/export_heightmap.cpp.o
../app/export/export_heightmap.cpp: In member function ‘virtual void ExportHeightmapWorker::async()’:
../app/export/export_heightmap.cpp:83:5: warning: missing initializer for member ‘Region_::voxels’ [-Wmissing-field-initializers]
     };
     ^
../app/export/export_heightmap.cpp:83:5: warning: missing initializer for member ‘Region_::X’ [-Wmissing-field-initializers]
../app/export/export_heightmap.cpp:83:5: warning: missing initializer for member ‘Region_::Y’ [-Wmissing-field-initializers]
../app/export/export_heightmap.cpp:83:5: warning: missing initializer for member ‘Region_::Z’ [-Wmissing-field-initializers]
../app/export/export_heightmap.cpp:83:5: warning: missing initializer for member ‘Region_::L’ [-Wmissing-field-initializers]
[15/18] Building CXX object app/CMakeF...ntimony.dir/export/export_worker.cpp.o
ninja: build stopped: subcommand failed.

mkeeter, 2020-03-29:
Can you try adding `-lpthread` to the compiler flags?  That's what I found with [30 seconds of Googling](https://github.com/facebook/Surround360/issues/3).

mkeeter, 2020-03-29:
Is this something that you could patch in a PR, so that it's fixed for other folks as well?

Bomi0, 2020-03-30:
After adding 
SET(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11 -pthread") 
#include <util.h>
#pragma clang diagnostic push
#pragma clang diagnostic ignored "-Wunused-parameter"
#pragma clang diagnostic ignored "-Wunknown-pragmas"
#pragma clang diagnostic ignored "-Wunused-variable"
#pragma clang diagnostic ignored "-Wunused-function"
#pragma clang diagnostic ignored "-Wdeprecated-register"
#pragma clang diagnostic ignored "-Wmissing-field-initializers"
#pragma clang diagnostic ignored "-Wunused-but-set-variable"
#pragma clang diagnostic ignored "-Wmissing-field-initializers"
#pragma GCC diagnostic ignored "-Wmaybe-uninitialized"
#pragma clang diagnostic pop 

I am having the result like below

[3/70] Linking CXX executable lib/graph/SbGraphTest
FAILED: lib/graph/SbGraphTest 
: && /usr/bin/c++  -Wall -Wextra -g -Werror=switch -std=c++11 -pthread -O3 -DRELEASE  -rdynamic lib/graph/CMakeFiles/SbGraphTest.dir/tests/datum.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/graph.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/link.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/main.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/node.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/script.cpp.o lib/graph/CMakeFiles/SbGraphTest.dir/tests/subgraph.cpp.o  -o lib/graph/SbGraphTest  lib/graph/libSbGraph.a  -lboost_python3  /usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a && :
/usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(dynload_shlib.o): In function `_PyImport_FindSharedFuncptr':
/opt/Python-3.8.1/./Python/dynload_shlib.c:86: undefined reference to `dlsym'
/opt/Python-3.8.1/./Python/dynload_shlib.c:99: undefined reference to `dlopen'
/opt/Python-3.8.1/./Python/dynload_shlib.c:130: undefined reference to `dlsym'
/opt/Python-3.8.1/./Python/dynload_shlib.c:99: undefined reference to `dlopen'
/opt/Python-3.8.1/./Python/dynload_shlib.c:105: undefined reference to `dlerror'
/usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(posixmodule.o): In function `os_openpty_impl':
/opt/Python-3.8.1/./Modules/posixmodule.c:6586: undefined reference to `openpty'
/usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(posixmodule.o): In function `os_forkpty_impl':
/opt/Python-3.8.1/./Modules/posixmodule.c:6684: undefined reference to `forkpty'
collect2: error: ld returned 1 exit status
[4/70] Linking CXX executable lib/fab/SbFabTest
FAILED: lib/fab/SbFabTest 
: && /usr/bin/c++  -Wall -Wextra -g -Werror=switch -std=c++11 -pthread -O3 -DRELEASE  -rdynamic lib/fab/CMakeFiles/SbFabTest.dir/tests/main.cpp.o lib/fab/CMakeFiles/SbFabTest.dir/tests/parser.cpp.o lib/fab/CMakeFiles/SbFabTest.dir/tests/shape.cpp.o  -o lib/fab/SbFabTest  lib/fab/libSbFab.a  -lboost_python3  /usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a  -lpng  -lz && :
/usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(dynload_shlib.o): In function `_PyImport_FindSharedFuncptr':
/opt/Python-3.8.1/./Python/dynload_shlib.c:86: undefined reference to `dlsym'
/opt/Python-3.8.1/./Python/dynload_shlib.c:99: undefined reference to `dlopen'
/opt/Python-3.8.1/./Python/dynload_shlib.c:130: undefined reference to `dlsym'
/opt/Python-3.8.1/./Python/dynload_shlib.c:99: undefined reference to `dlopen'
/opt/Python-3.8.1/./Python/dynload_shlib.c:105: undefined reference to `dlerror'
/usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(posixmodule.o): In function `os_openpty_impl':
/opt/Python-3.8.1/./Modules/posixmodule.c:6586: undefined reference to `openpty'
/usr/local/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(posixmodule.o): In function `os_forkpty_impl':
/opt/Python-3.8.1/./Modules/posixmodule.c:6684: undefined reference to `forkpty'
collect2: error: ld returned 1 exit status
[16/70] Building CXX object app/CMakeFiles/antimony.dir/graph/hooks/ui.cpp.o
ninja: build stopped: subcommand failed.