mkeeter/antimony #216: Problem building on elementary OS

johnygum, 2020-04-29:
Hello!
I'm on elementary OS 5.1.3 Hera (built on Ubuntu 18.04.3 LTS), and I followed the build instructions, but I get this error:
```
$ cmake -GNinja ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:11 (find_package):
  By not providing "FindPython.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Python", but
  CMake did not find one.

  Could not find a package configuration file provided by "Python" (requested
  version 3.3) with any of the following names:

    PythonConfig.cmake
    python-config.cmake

  Add the installation prefix of "Python" to CMAKE_PREFIX_PATH or set
  "Python_DIR" to a directory containing one of the above files.  If "Python"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
```
Thanks for the help!

mkeeter, 2020-05-01:
Did you install all of the dependencies, including Python?

johnygum, 2020-05-01:
Thanks for the fast response!

Yes, I have copied the whole line from the instructions:
`# Install dependencies
sudo apt install git build-essential libpng-dev python3-dev libboost-all-dev  libgl1-mesa-dev lemon flex qt5-default ninja-build cmake
`
and
 `(On some distros, you may need to install 'libqt5opengl5' as well).`

Also python3 was already installed on the OS and I think it necessary for the system to work (I wanted to do a clean python install once and it broke the whole os).

johnygum, 2020-05-02:
I tried building it on another machine, that runs almost unmodified elementary OS and I have the same issue there, so I guess there is some "incompatibility" with the OS.