mkeeter/antimony #217: Add custom nodes to 'Add' and right-click menus

Faberix, 2020-06-11:
It would be great to have an option to save (custom) nodes that you often need and then be able to access them using the Menu. This would be a lot more convenient than having to create a new script node (or whatever other base node you are using) and then copy the code you want or write new code each time you need it.
Adding a node to the saved nodes could be done using a GUI like in the sketch below:
[image]
It could be accessed using a new entry in the right click menu when right clicked on a node and/or using a new entry (e.g. "Edit available nodes") in either the "Edit" or the "Add" menu.

Implementing the dynamic addition/removal of menu items should not be difficult; it can be done using QMenu.addAction, QMenu.removeAction and QMenu.insertAction (of course, the nodes and the order would have to be saved in the program data and the menus would have to be adjusted accordingly on each program startup).

Saving the additional nodes in the program data should also not be a problem, as the nodes should actually be fully defined by their Python code snippet.

One node that would be useful to have accessible would be the following node (that evaluates an arbitrary mathematical expression in Python code):
` import fab
import math

varTuple = ('a', 'b')

title('formula')
for v in varTuple:
	input(v, float)
input('formula', str)

out = eval(formula)
output('out', out) `

mkeeter, 2020-06-11:
It's already possible to install (global) custom nodes: when populating the node menu [here](https://github.com/mkeeter/antimony/blob/a3d821b0c2df449d33079a3a032914591f5152c4/app/app/app.cpp#L69-L73), we search for files in the `nodes` subfolder of [`AppDataLocation`](https://doc.qt.io/qt-5/qstandardpaths.html), which is platform-specific.

Does that accomplish what you're looking for, or do you specifically want to store extra nodes on a per-file basis?  It would also be useful to have a UI for saving a new global node, which would be welcome in a PR.

Faberix, 2020-06-11:
> Does that accomplish what you're looking for, or do you specifically want to store extra nodes on a per-file basis?
Yes, it does.
As for the UI, I will not have time for that any time soon (I already have too many other projects running). In case anyone else wants to implement it: I noticed that the nodes are stored in /usr/share/antimony (at least in Ubuntu), which is read-only (except for root, of course). So, if a node should be stored from GUI, QStandardPaths::writableLocation should be used for all custom nodes.

I will leave this issue open, as the GUI would still be useful (especially because the option of manually creating the files will probably not be discovered by most users). Feel free to close it, though.