mkeeter/antimony #230: Unable to build on Windows 10

antariandel, 2021-02-12:
Hi! I'd really love to try out Antimony, however I'm unable to build it on Windows by following the instructions on the wiki.
I initially got an error with lemon not finding the `lempar.c` template, which I solved by passing it to lemon explicitly in `lib/fab/CMakeFiles/SbFab.dir/build.make` on line 81 (note the *-T/usr/share/lemon/lempar.c* part):

```
cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/fab && /C/MSYS2_64/usr/bin/lemon.exe -q -c -s -T/usr/share/lemon/lempar.c C:/MSYS2_64/home/Zsolt/antimony/lib/fab/src/tree/v2syntax.y
```

Originally it was trying to find the template on an incorrect path, I think it was `/usr/share/lemon//C/MSYS2_64/usr/share/lemon/lempar.c` (as if the generator came up with an incorrect path).

Now I run in the following error:

```
[  1%] Generating v2syntax.yy.hpp, v2syntax.yy.cpp
[  1%] Generating v2syntax.lemon.hpp, v2syntax.lemon.cpp
Parser statistics:
  terminal symbols...................    60
  non-terminal symbols...............     7
  total symbols......................    67
  rules..............................    58
  states.............................   159
  conflicts..........................     0
  action table entries...............  1936
  lookahead table entries............  1964
  total table size (bytes)...........  6594
[  2%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/graph.cpp.obj
[  3%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/datum.cpp.obj
[  4%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/node.cpp.obj
[  5%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/script_node.cpp.obj
[  5%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/graph_node.cpp.obj
[  6%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/proxy.cpp.obj
make[2]: *** [lib/graph/CMakeFiles/SbGraph.dir/build.make:82: lib/graph/CMakeFiles/SbGraph.dir/src/datum.cpp.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [lib/graph/CMakeFiles/SbGraph.dir/build.make:95: lib/graph/CMakeFiles/SbGraph.dir/src/graph.cpp.obj] Error 1
make[2]: *** [lib/graph/CMakeFiles/SbGraph.dir/build.make:108: lib/graph/CMakeFiles/SbGraph.dir/src/node.cpp.obj] Error 1
make[2]: *** [lib/graph/CMakeFiles/SbGraph.dir/build.make:121: lib/graph/CMakeFiles/SbGraph.dir/src/script_node.cpp.obj] Error 1
make[2]: *** [lib/graph/CMakeFiles/SbGraph.dir/build.make:147: lib/graph/CMakeFiles/SbGraph.dir/src/proxy.cpp.obj] Error 1
make[2]: *** [lib/graph/CMakeFiles/SbGraph.dir/build.make:134: lib/graph/CMakeFiles/SbGraph.dir/src/graph_node.cpp.obj] Error 1
[  7%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/script.cpp.obj
make[2]: *** [lib/graph/CMakeFiles/SbGraph.dir/build.make:160: lib/graph/CMakeFiles/SbGraph.dir/src/script.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:187: lib/graph/CMakeFiles/SbGraph.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Scanning dependencies of target SbFab
[  8%] Building CXX object lib/fab/CMakeFiles/SbFab.dir/src/fab.cpp.obj
[  9%] Building C object lib/fab/CMakeFiles/SbFab.dir/src/formats/png.c.obj
[  9%] Building C object lib/fab/CMakeFiles/SbFab.dir/src/formats/stl.c.obj
[ 10%] Building C object lib/fab/CMakeFiles/SbFab.dir/src/tree/eval.c.obj
[ 11%] Building C object lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_f.c.obj
[ 12%] Building C object lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_g.c.obj
[ 12%] Building C object lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_i.c.obj
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:98: lib/fab/CMakeFiles/SbFab.dir/src/fab.cpp.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 13%] Building C object lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_r.c.obj
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:150: lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_f.c.obj] Error 1
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:189: lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_r.c.obj] Error 1
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:137: lib/fab/CMakeFiles/SbFab.dir/src/tree/eval.c.obj] Error 1
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:176: lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_i.c.obj] Error 1
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:163: lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_g.c.obj] Error 1
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:111: lib/fab/CMakeFiles/SbFab.dir/src/formats/png.c.obj] Error 1
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:124: lib/fab/CMakeFiles/SbFab.dir/src/formats/stl.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:241: lib/fab/CMakeFiles/SbFab.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```

You're help is very much appreciated, @mkeeter !

Btw. I congratulate you for creating this and `libfive` as well. I understand that that project surpassed this one for you, but I think the approach with the nodes in Antimony is really ingenious, and that's what is making it really distinctive for me. I'd really like to play with it. I'm also a little bit sad, that this project didn't take off more and wasn't continued.

mkeeter, 2021-02-12:
Thanks for the kind words!

Can you run `make` in verbose mode, probably `VERBOSE=1 make`?

Right now, the only error shown is `Error 1`, which isn't enough to debug.

antariandel, 2021-02-12:
Sure thing, here's what I got:

```
Zsolt@CASSIOPEIA MSYS ~/antimony/build
$ VERBOSE=1 make -j8
/C/MSYS2_64/mingw64/bin/cmake.exe -S/C/MSYS2_64/home/Zsolt/antimony -B/C/MSYS2_64/home/Zsolt/antimony/build --check-build-system CMakeFiles/Makefile.cmake 0
/C/MSYS2_64/mingw64/bin/cmake.exe -E cmake_progress_start /C/MSYS2_64/home/Zsolt/antimony/build/CMakeFiles /C/MSYS2_64/home/Zsolt/antimony/build//CMakeFiles/progress.marks
make  -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/Zsolt/antimony/build'
make  -f lib/graph/CMakeFiles/SbGraph.dir/build.make lib/graph/CMakeFiles/SbGraph.dir/depend
make  -f lib/fab/CMakeFiles/SbFab.dir/build.make lib/fab/CMakeFiles/SbFab.dir/depend
make[2]: Entering directory '/home/Zsolt/antimony/build'
/C/MSYS2_64/mingw64/bin/cmake.exe -E cmake_depends "MSYS Makefiles" /C/MSYS2_64/home/Zsolt/antimony /C/MSYS2_64/home/Zsolt/antimony/lib/graph /C/MSYS2_64/home/Zsolt/antimony/build /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph/CMakeFiles/SbGraph.dir/DependInfo.cmake --color=
make[2]: Entering directory '/home/Zsolt/antimony/build'
/C/MSYS2_64/mingw64/bin/cmake.exe -E cmake_depends "MSYS Makefiles" /C/MSYS2_64/home/Zsolt/antimony /C/MSYS2_64/home/Zsolt/antimony/lib/fab /C/MSYS2_64/home/Zsolt/antimony/build /C/MSYS2_64/home/Zsolt/antimony/build/lib/fab /C/MSYS2_64/home/Zsolt/antimony/build/lib/fab/CMakeFiles/SbFab.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/Zsolt/antimony/build'
make  -f lib/graph/CMakeFiles/SbGraph.dir/build.make lib/graph/CMakeFiles/SbGraph.dir/build
make[2]: Leaving directory '/home/Zsolt/antimony/build'
make  -f lib/fab/CMakeFiles/SbFab.dir/build.make lib/fab/CMakeFiles/SbFab.dir/build
make[2]: Entering directory '/home/Zsolt/antimony/build'
make[2]: Entering directory '/home/Zsolt/antimony/build'
[  1%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/datum.cpp.obj
cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph && /C/MSYS2_64/mingw64/bin/g++.exe -DBOOST_PYTHON_STATIC_LIB -I/C/MSYS2_64/home/Zsolt/antimony/lib/graph/inc -isystem /C/MSYS2_64/mingw64/include/python3.8 -Wall -Wextra -g -Werror=switch -O3 -DRELEASE -std=gnu++11 -o CMakeFiles/SbGraph.dir/src/datum.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/graph/src/datum.cpp
[  2%] Building CXX object lib/fab/CMakeFiles/SbFab.dir/src/fab.cpp.obj
[  3%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/graph.cpp.obj
[  4%] Building C object lib/fab/CMakeFiles/SbFab.dir/src/formats/png.c.obj
cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph && /C/MSYS2_64/mingw64/bin/g++.exe -DBOOST_PYTHON_STATIC_LIB -I/C/MSYS2_64/home/Zsolt/antimony/lib/graph/inc -isystem /C/MSYS2_64/mingw64/include/python3.8 -Wall -Wextra -g -Werror=switch -O3 -DRELEASE -std=gnu++11 -o CMakeFiles/SbGraph.dir/src/graph.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/graph/src/graph.cpp
[  5%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/node.cpp.obj
cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph && /C/MSYS2_64/mingw64/bin/g++.exe -DBOOST_PYTHON_STATIC_LIB -I/C/MSYS2_64/home/Zsolt/antimony/lib/graph/inc -isystem /C/MSYS2_64/mingw64/include/python3.8 -Wall -Wextra -g -Werror=switch -O3 -DRELEASE -std=gnu++11 -o CMakeFiles/SbGraph.dir/src/node.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/graph/src/node.cpp
cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/fab && /C/MSYS2_64/mingw64/bin/g++.exe -DBOOST_PYTHON_STATIC_LIB -I/C/MSYS2_64/home/Zsolt/antimony/build/lib/fab -I/C/MSYS2_64/home/Zsolt/antimony/lib/fab/inc -isystem /C/MSYS2_64/mingw64/include/python3.8 -isystem /C/MSYS2_64/home/Zsolt/antimony/lib/fab/vendor -Wall -Wextra -g -Werror=switch -O3 -DRELEASE -std=gnu++11 -o CMakeFiles/SbFab.dir/src/fab.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/fab/src/fab.cpp
[  5%] Building C object lib/fab/CMakeFiles/SbFab.dir/src/formats/stl.c.obj
cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/fab && /MinGW/bin/gcc.exe -DBOOST_PYTHON_STATIC_LIB -I/C/MSYS2_64/home/Zsolt/antimony/build/lib/fab -I/C/MSYS2_64/home/Zsolt/antimony/lib/fab/inc -I/C/MSYS2_64/mingw64/include/python3.8 -I/C/MSYS2_64/mingw64/include -I/C/MSYS2_64/home/Zsolt/antimony/lib/fab/vendor  -o CMakeFiles/SbFab.dir/src/formats/png.c.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/fab/src/formats/png.c
cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/fab && /MinGW/bin/gcc.exe -DBOOST_PYTHON_STATIC_LIB -I/C/MSYS2_64/home/Zsolt/antimony/build/lib/fab -I/C/MSYS2_64/home/Zsolt/antimony/lib/fab/inc -I/C/MSYS2_64/mingw64/include/python3.8 -I/C/MSYS2_64/mingw64/include -I/C/MSYS2_64/home/Zsolt/antimony/lib/fab/vendor  -o CMakeFiles/SbFab.dir/src/formats/stl.c.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/fab/src/formats/stl.c
[  6%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/script_node.cpp.obj
cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph && /C/MSYS2_64/mingw64/bin/g++.exe -DBOOST_PYTHON_STATIC_LIB -I/C/MSYS2_64/home/Zsolt/antimony/lib/graph/inc -isystem /C/MSYS2_64/mingw64/include/python3.8 -Wall -Wextra -g -Werror=switch -O3 -DRELEASE -std=gnu++11 -o CMakeFiles/SbGraph.dir/src/script_node.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/graph/src/script_node.cpp
[  6%] Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/graph_node.cpp.obj
cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph && /C/MSYS2_64/mingw64/bin/g++.exe -DBOOST_PYTHON_STATIC_LIB -I/C/MSYS2_64/home/Zsolt/antimony/lib/graph/inc -isystem /C/MSYS2_64/mingw64/include/python3.8 -Wall -Wextra -g -Werror=switch -O [... cut ...]
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:111: lib/fab/CMakeFiles/SbFab.dir/src/formats/png.c.obj] Error 1
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:137: lib/fab/CMakeFiles/SbFab.dir/src/tree/eval.c.obj] Error 1
make[2]: *** [lib/fab/CMakeFiles/SbFab.dir/build.make:124: lib/fab/CMakeFiles/SbFab.dir/src/formats/stl.c.obj] Error 1
make[2]: Leaving directory '/home/Zsolt/antimony/build'
make[1]: *** [CMakeFiles/Makefile2:241: lib/fab/CMakeFiles/SbFab.dir/all] Error 2
make[1]: Leaving directory '/home/Zsolt/antimony/build'
make: *** [Makefile:149: all] Error 2
```

mkeeter, 2021-02-12:
That still doesn't show any actual errors, besides "Error 1".

What happens if you run a single one of the commands, e.g.
```
cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph
/C/MSYS2_64/mingw64/bin/g++.exe -DBOOST_PYTHON_STATIC_LIB -I/C/MSYS2_64/home/Zsolt/antimony/lib/graph/inc -isystem /C/MSYS2_64/mingw64/include/python3.8 -Wall -Wextra -g -Werror=switch -O3 -DRELEASE -std=gnu++11 -o CMakeFiles/SbGraph.dir/src/graph_node.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/graph/src/graph_node.cpp
```

antariandel, 2021-02-13:
That command completed silently, without errors. Can I do something to make it more verbose?

mkeeter, 2021-02-13:
That's very strange.  Can you check the actual line of the `build.make` which is reporting an error, to see what it's trying to do?
(e.g. line 95 for the error `make[2]: *** [lib/graph/CMakeFiles/SbGraph.dir/build.make:95: lib/graph/CMakeFiles/SbGraph.dir/src/graph.cpp.obj] Error 1`, plus some surrounding area for context)

antariandel, 2021-02-13:
I have copied all. Here they are:

### Errors from `build/lib/graph/CMakeFiles/SbGraph.dir/build.make`:

**Line 121**
```
118 lib/graph/CMakeFiles/SbGraph.dir/src/script_node.cpp.obj: lib/graph/CMakeFiles/SbGraph.dir/flags.make
119 lib/graph/CMakeFiles/SbGraph.dir/src/script_node.cpp.obj: ../lib/graph/src/script_node.cpp
120         @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --progress-dir=/C/MSYS2_64/home/Zsolt/antimony/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_4) "Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/script_node.cpp.obj"
121         cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph && /C/MSYS2_64/mingw64/bin/g++.exe $(CXX_DEFINES) $(CXX_INCLUDES) $(CXX_FLAGS) -o CMakeFiles/SbGraph.dir/src/script_node.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/graph/src/script_node.cpp
122
```

**Line 82**
```
 79 lib/graph/CMakeFiles/SbGraph.dir/src/datum.cpp.obj: lib/graph/CMakeFiles/SbGraph.dir/flags.make
 80 lib/graph/CMakeFiles/SbGraph.dir/src/datum.cpp.obj: ../lib/graph/src/datum.cpp
 81         @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --progress-dir=/C/MSYS2_64/home/Zsolt/antimony/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_1) "Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/datum.cpp.obj"
 82         cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph && /C/MSYS2_64/mingw64/bin/g++.exe $(CXX_DEFINES) $(CXX_INCLUDES) $(CXX_FLAGS) -o CMakeFiles/SbGraph.dir/src/datum.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/graph/src/datum.cpp
 83
```

**Line 108**
```
105 lib/graph/CMakeFiles/SbGraph.dir/src/node.cpp.obj: lib/graph/CMakeFiles/SbGraph.dir/flags.make
106 lib/graph/CMakeFiles/SbGraph.dir/src/node.cpp.obj: ../lib/graph/src/node.cpp
107         @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --progress-dir=/C/MSYS2_64/home/Zsolt/antimony/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_3) "Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/node.cpp.obj"
108         cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph && /C/MSYS2_64/mingw64/bin/g++.exe $(CXX_DEFINES) $(CXX_INCLUDES) $(CXX_FLAGS) -o CMakeFiles/SbGraph.dir/src/node.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/graph/src/node.cpp
109
```

**Line 95**
```
 92 lib/graph/CMakeFiles/SbGraph.dir/src/graph.cpp.obj: lib/graph/CMakeFiles/SbGraph.dir/flags.make
 93 lib/graph/CMakeFiles/SbGraph.dir/src/graph.cpp.obj: ../lib/graph/src/graph.cpp
 94         @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --progress-dir=/C/MSYS2_64/home/Zsolt/antimony/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_2) "Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/graph.cpp.obj"
 95         cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph && /C/MSYS2_64/mingw64/bin/g++.exe $(CXX_DEFINES) $(CXX_INCLUDES) $(CXX_FLAGS) -o CMakeFiles/SbGraph.dir/src/graph.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/graph/src/graph.cpp
 96
```

**Line 134**
```
131 lib/graph/CMakeFiles/SbGraph.dir/src/graph_node.cpp.obj: lib/graph/CMakeFiles/SbGraph.dir/flags.make
132 lib/graph/CMakeFiles/SbGraph.dir/src/graph_node.cpp.obj: ../lib/graph/src/graph_node.cpp
133         @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --progress-dir=/C/MSYS2_64/home/Zsolt/antimony/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_5) "Building CXX object lib/graph/CMakeFiles/SbGraph.dir/src/graph_node.cpp.obj"
134         cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/graph && /C/MSYS2_64/mingw64/bin/g++.exe $(CXX_DEFINES) $(CXX_INCLUDES) $(CXX_FLAGS) -o CMakeFiles/SbGraph.dir/src/graph_node.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/graph/src/graph_node.cpp
135
```

There's one error in `build/CMakeFiles/Makefile2`, on **line 187**, but I suppose that's just the other errors propagating up:
```
184 # All Build rule for target.
185 lib/graph/CMakeFiles/SbGraph.dir/all:
186         $(MAKE) $(MAKESILENT) -f lib/graph/CMakeFiles/SbGraph.dir/build.make lib/graph/CMakeFiles/SbGraph.dir/depend
187         $(MAKE) $(MAKESILENT) -f lib/graph/CMakeFiles/SbGraph.dir/build.make lib/graph/CMakeFiles/SbGraph.dir/build
188         @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --progress-dir=/C/MSYS2_64/home/Zsolt/antimony/build/CMakeFiles --progress-num=26,27,28,29,30,31,32,33,34,35,36 "Built target SbGraph"
189 .PHONY : lib/graph/CMakeFiles/SbGraph.dir/all
190
```
&nbsp;

### Errors from `build/lib/fab/CMakeFiles/SbFab.dir/build.make`:

**Line 98**
```
 95 lib/fab/CMakeFiles/SbFab.dir/src/fab.cpp.obj: lib/fab/CMakeFiles/SbFab.dir/flags.make
 96 lib/fab/CMakeFiles/SbFab.dir/src/fab.cpp.obj: ../lib/fab/src/fab.cpp
 97         @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --progress-dir=/C/MSYS2_64/home/Zsolt/antimony/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_3) "Building CXX object lib/fab/CMakeFiles/SbFab.dir/src/fab.cpp.obj"
 98         cd /C/MSYS2_64/home/Zsolt/antimony/build/lib/fab && /C/MSYS2_64/mingw64/bin/g++.exe $(CXX_DEFINES) $(CXX_INCLUDES) $(CXX_FLAGS) -o CMakeFiles/SbFab.dir/src/fab.cpp.obj -c /C/MSYS2_64/home/Zsolt/antimony/lib/fab/src/fab.cpp
 99
[... 32 lines cut ...]
148 lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_f.c.obj: ../lib/fab/src/tree/math/math_f.c
149         @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --progress-dir=/C/MSYS2_64/home/Zsolt/antimony/build/CMakeFiles --progress-num=$(CMAKE_PROGRESS_7) "Building C object lib/fab/CMakeFiles/SbFab.dir/src/tree/math/math_f.c.obj"
[... cut ...]
240         $(MAKE) $(MAKESILENT) -f lib/fab/CMakeFiles/SbFab.dir/build.make lib/fab/CMakeFiles/SbFab.dir/depend
241         $(MAKE) $(MAKESILENT) -f lib/fab/CMakeFiles/SbFab.dir/build.make lib/fab/CMakeFiles/SbFab.dir/build
242         @$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --progress-dir=/C/MSYS2_64/home/Zsolt/antimony/build/CMakeFiles --progress-num=1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22 "Built target SbFab"
243 .PHONY : lib/fab/CMakeFiles/SbFab.dir/all
244
```