mkeeter/antimony #232: Renaming step of Windows build instructions

omgitsraven, 2021-03-14:
I'm running the experimental Windows build steps, but several things have updated since they were written; I've managed to adapt most of the steps for the new filenames, but there's one step that I'm not sure how to handle.

After compiling boost, I have to "Rename the generated library to something that cmake will find", with this command:
`mv libboost_python-mgw63-mt-1_63.a libboost_python3-mt.a`

but there are four `.a` files in my `stage/lib` directory:
```
libboost_python38-mgw10-mt-d-x32-1_75.a
libboost_python38-mgw10-mt-d-x64-1_75.a
libboost_python38-mgw10-mt-x32-1_75.a
libboost_python38-mgw10-mt-x64-1_75.a
```
and I don't know which one of these corresponds to `libboost_python-mgw63-mt-1_63.a`.

mkeeter, 2021-03-15:
I'm curious whether we still need this workaround, or whether the `mingw-w64-x86_64-boost` recipe works out of the box at this point – it would be nice to remove the workaround if possible.

If that's not possible, it's probably `libboost_python38-mgw10-mt-x64-1_75.a` (64-bit, and I'm guessing the `-d-` marks a debug build)

omgitsraven, 2021-03-15:
I noticed it wasn't in your instructions over on libfive (I'll try them this evening!), did it work there?

mkeeter, 2021-03-15:
`libfive` doesn't need Boost::Python, so I don't know 😆

omgitsraven, 2021-03-15:
oh my god, well in that case you can go ahead and close this issue if you'd like, heh...  I may as well leave it open in case anyone else has the same problem in the future, but, my experimenting with it is hereby complete <_<