mkeeter/antimony #233: Build explodes with errors on OpenBSD CURRENT in `v2syntax.c`.

anoduck, 2022-06-06:
System: OpenBSD Snapshot Current
CMake Version: 3.23.1
Ninja Version: 1.11.0
Flex Version: 2.5.39
Lemon Version: ??? (It has more than one version?)

----

It took some finagling to acquire solely the lemon and lempar source code, but appears to have been successful... I think. Admittedly, compiling antimony on OpenBSD just may not be possible.

Upon running ninja to build antimony, the console erupted with errors and stopped on `lib/fab/CMakeFiles/SbFabTest.dir/tests/main.cpp.0`. Here is the output:

```shell
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:699:3: error: unknown type name 'v2ParseCTX_SDECL'
  v2ParseCTX_SDECL                /* A place to hold %extra_context */
  ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:705:15: error: expected ';' at end of declaration list
  yyStackEntry yystack[YYSTACKDEPTH];  /* The parser's stack */
              ^
              ;
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:706:3: error: must use 'struct' tag to refer to type 'yyStackEntry' in this scope
  yyStackEntry *yystackEnd;            /* Last entry in the stack */
  ^
  struct 
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:705:3: note: struct 'yyStackEntry' is hidden by a non-type declaration of 'yyStackEntry' here
  yyStackEntry yystack[YYSTACKDEPTH];  /* The parser's stack */
  ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:878:37: error: expected ')'
void v2ParseInit(void *yypRawParser v2ParseCTX_PDECL){
                                    ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:878:17: note: to match this '('
void v2ParseInit(void *yypRawParser v2ParseCTX_PDECL){
                ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:880:3: error: unknown type name 'v2ParseCTX_STORE'
  v2ParseCTX_STORE
  ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:894:12: error: expected ';' at end of declaration
  yypParser->yyerrcnt = -1;
           ^
           ;
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:896:33: error: no member named 'yystack' in 'yyParser'
  yypParser->yytos = yypParser->yystack;
                     ~~~~~~~~~  ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:897:14: error: no member named 'yystack' in 'yyParser'
  yypParser->yystack[0].stateno = 0;
  ~~~~~~~~~  ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:898:14: error: no member named 'yystack' in 'yyParser'
  yypParser->yystack[0].major = 0;
  ~~~~~~~~~  ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:900:39: error: no member named 'yystack' in 'yyParser'
  yypParser->yystackEnd = &yypParser->yystack[YYSTACKDEPTH-1];
                           ~~~~~~~~~  ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:917:57: error: expected ')'
void *v2ParseAlloc(void *(*mallocProc)(YYMALLOCARGTYPE) v2ParseCTX_PDECL){
                                                        ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:917:19: note: to match this '('
void *v2ParseAlloc(void *(*mallocProc)(YYMALLOCARGTYPE) v2ParseCTX_PDECL){
                  ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:921:5: error: unknown type name 'v2ParseCTX_STORE'
    v2ParseCTX_STORE
    ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:922:17: error: unknown type name 'yypParser'
    v2ParseInit(yypParser v2ParseCTX_PARAM);
                ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:941:19: error: expected ';' at end of declaration
  v2ParseARG_FETCH
                  ^
                  ;
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:969:37: error: no member named 'yystack' in 'yyParser'
  assert( pParser->yytos > pParser->yystack );
                           ~~~~~~~  ^
/usr/include/assert.h:56:23: note: expanded from macro 'assert'
#  define       assert(e)       ((e) ? (void)0 : __assert2(__FILE__, __LINE__, __func__, #e))
                                  ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:986:34: error: no member named 'yystack' in 'yyParser'
  while( pParser->yytos>pParser->yystack ) yy_pop_parser_stack(pParser);
                        ~~~~~~~  ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:1079:15: error: use of undeclared identifier 'YYNTOKEN'
    assert( i+YYNTOKEN<=(int)YY_NLOOKAHEAD );
              ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:1081:26: error: use of undeclared identifier 'YYNTOKEN'
    assert( iLookAhead < YYNTOKEN );
                         ^
/home/USER/antimony/lib/fab/src/tree/v2syntax.c:1159:20: error: expected ';' at end of declaration
   v2ParseARG_FETCH
                   ^
                   ;
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
[29/108] Building CXX object lib/fab/CMakeFiles/SbFabTest.dir/tests/main.cpp.o
```  
If there is anything, let me know.

fros1y, 2022-06-07:
Hi there,

Lemon is distributed as part of SQLite and is just given the same version number as the source SQLite release. Can you confirm the SQLite release you built lemon from?

Best,

anoduck, 2022-06-08:
> Hi there,
> 
> Lemon is distributed as part of SQLite and is just given the same version number as the source SQLite release. Can you confirm the SQLite release you built lemon from?
> 
> Best,

Oh...nice... Grabbing src "sqlite-src-3380500.zip" and compiled with the basic `cc -o lemon lemon.c`. 

Regardless of acquiring it from the sqlite src or downloading it directly from sqlite's lemon page, the result appears to be the same. Output of `./lemon -x` is `Lemon version 1.0`.

Now, I have more sensical output, and it still fails to build. Now outputs error as `Unknown OS!`

```bash
[1/66] Building CXX object app/CMakeFiles/antimony.dir/app/main.cpp.o
FAILED: app/CMakeFiles/antimony.dir/app/main.cpp.o 
/usr/bin/c++ -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -I/home/USER/antimony/build/app -I/home/USER/antimony/app -I/home/USER/antimony/build/app/antimony_autogen/include -I/home/USER/antimony/lib/graph/inc -I/home/USER/antimony/lib/fab/inc -isystem /usr/local/include/python3.9 -isystem /usr/local/include -isystem /usr/local/include/X11/qt5 -isystem /usr/local/include/X11/qt5/QtWidgets -isystem /usr/local/include/X11/qt5/QtGui -isystem /usr/X11R6/include -isystem /usr/local/include/X11/qt5/QtCore -isystem /usr/local/lib/qt5/./mkspecs/openbsd-clang -isystem /usr/local/include/X11/qt5/QtOpenGL -isystem /usr/local/include/X11/qt5/QtNetwork -isystem /usr/local/include/X11/qt5/QtConcurrent -Wall -Wextra -g -Werror=switch -O3 -DRELEASE   -D'GITREV="'ee525bbd'"' -D'GITTAG=""' -D'GITBRANCH="develop "' -fPIC -std=gnu++11 -MD -MT app/CMakeFiles/antimony.dir/app/main.cpp.o -MF app/CMakeFiles/antimony.dir/app/main.cpp.o.d -o app/CMakeFiles/antimony.dir/app/main.cpp.o -c /home/USER/antimony/app/app/main.cpp
/home/USER/antimony/app/app/main.cpp:65:2: error: "Unknown OS!"
#error "Unknown OS!"
 ^
1 error generated.
[3/66] Building CXX object app/CMakeFiles/antimony.dir/app/app.cpp.o
FAILED: app/CMakeFiles/antimony.dir/app/app.cpp.o 
/usr/bin/c++ -DQT_CONCURRENT_LIB -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WIDGETS_LIB -I/home/USER/antimony/build/app -I/home/USER/antimony/app -I/home/USER/antimony/build/app/antimony_autogen/include -I/home/USER/antimony/lib/graph/inc -I/home/USER/antimony/lib/fab/inc -isystem /usr/local/include/python3.9 -isystem /usr/local/include -isystem /usr/local/include/X11/qt5 -isystem /usr/local/include/X11/qt5/QtWidgets -isystem /usr/local/include/X11/qt5/QtGui -isystem /usr/X11R6/include -isystem /usr/local/include/X11/qt5/QtCore -isystem /usr/local/lib/qt5/./mkspecs/openbsd-clang -isystem /usr/local/include/X11/qt5/QtOpenGL -isystem /usr/local/include/X11/qt5/QtNetwork -isystem /usr/local/include/X11/qt5/QtConcurrent -Wall -Wextra -g -Werror=switch -O3 -DRELEASE   -D'GITREV="'ee525bbd'"' -D'GITTAG=""' -D'GITBRANCH="develop "' -fPIC -std=gnu++11 -MD -MT app/CMakeFiles/antimony.dir/app/app.cpp.o -MF app/CMakeFiles/antimony.dir/app/app.cpp.o.d -o app/CMakeFiles/antimony.dir/app/app.cpp.o -c /home/USER/antimony/app/app/app.cpp
/home/USER/antimony/app/app/app.cpp:66:2: error: "Unknown OS!"
#error "Unknown OS!"
 ^
/home/USER/antimony/app/app/app.cpp:87:27: warning: 'toList' is deprecated: Use values() instead. [-Wdeprecated-declarations]
    return existing_paths.toList();
                          ^
/usr/local/include/X11/qt5/QtCore/qset.h:264:5: note: 'toList' has been explicitly marked deprecated here
    QT_DEPRECATED_X("Use values() instead.")
    ^
/usr/local/include/X11/qt5/QtCore/qglobal.h:294:33: note: expanded from macro 'QT_DEPRECATED_X'
#  define QT_DEPRECATED_X(text) Q_DECL_DEPRECATED_X(text)
                                ^
/usr/local/include/X11/qt5/QtCore/qcompilerdetection.h:675:55: note: expanded from macro 'Q_DECL_DEPRECATED_X'
#    define Q_DECL_DEPRECATED_X(text) __attribute__ ((__deprecated__(text)))
                                                      ^
1 warning and 1 error generated.
[4/66] Building CXX object app/CMakeFiles/antimony.dir/canvas/inspector/export.cpp.o
ninja: build stopped: subcommand failed.
```

So, OpenBSD might need to be added to source code as a valid OS option.

fros1y, 2022-06-08:
Yep, different issue.  I think you're the first person to try to get it running on OpenBSD.

mkeeter, 2022-06-08:
You're welcome to try building with [this line](https://github.com/mkeeter/antimony/blob/develop/app/app/main.cpp#L50) changed to include `Q_OS_OPENBSD`, but I expect that other things will not work!

anoduck, 2022-06-08:
Oh, highly probable. Running OpenBSD is part blessing and part curse. 

Let me give a whirl, and see what comes of it. If worse comes to worst, I have a Linux workstation I VNC into, and will just continue to use antimony on it.

anoduck, 2022-06-08:
> You're welcome to try building with [this line](https://github.com/mkeeter/antimony/blob/develop/app/app/main.cpp#L50) changed to include `Q_OS_OPENBSD`, but I expect that other things will not work!

And boom goes the dynamite!  We have now successfully built Antimony on OpenBSD!
Which was needed, since there are few CAD options for OpenBSD.

More than [line 50](https://github.com/mkeeter/antimony/blob/develop/app/app/main.cpp#L50) was needed, as `Q_OS_OPENBSD` was needed in [line 58 of app.cpp](https://github.com/mkeeter/antimony/blob/develop/app/app/app.cpp#L58) and in [line 127 of app.cpp](https://github.com/mkeeter/antimony/blob/develop/app/app/app.cpp#L127). Since OpenBSD is partly influenced by Linux, the parameters are exactly the same, and can be literally copied and pasted. Providing, they are under the `Q_OS_OPENBSD` heading/thinga-ma-bobber.

mkeeter, 2022-06-13:
Sure, go for it!

closed 2023-12-30 by anoduck