mkeeter/antimony #236: configure succeeds, and then the build fails: /bin/sh: LEMON_EXECUTABLE-NOTFOUND: not found

yurivict, 2023-01-16:
```
[  0% 3/111] cd /usr/ports/cad/antimony/work/.build/lib/fab && LEMON_EXECUTABLE-NOTFOUND -q -c -s /usr/ports/cad/antimony/work/antimony-0.9.3b-37-g8b805c67/lib/fab/src/tree/v2syntax.y && mv /usr/ports/cad/antimony/work/antimony-0.9.3b-37-g8b805c67/lib/fab/src/tree/v2syntax.c /usr/ports/cad/antimony/work/.build/lib/fab/v2syntax.lemon.cpp && mv /usr/ports/cad/antimony/work/antimony-0.9.3b-37-g8b805c67/lib/fab/src/tree/v2syntax.h /usr/ports/cad/antimony/work/.build/lib/fab/v2syntax.lemon.hpp
FAILED: lib/fab/v2syntax.lemon.hpp lib/fab/v2syntax.lemon.cpp /usr/ports/cad/antimony/work/.build/lib/fab/v2syntax.lemon.hpp /usr/ports/cad/antimony/work/.build/lib/fab/v2syntax.lemon.cpp 
cd /usr/ports/cad/antimony/work/.build/lib/fab && LEMON_EXECUTABLE-NOTFOUND -q -c -s /usr/ports/cad/antimony/work/antimony-0.9.3b-37-g8b805c67/lib/fab/src/tree/v2syntax.y && mv /usr/ports/cad/antimony/work/antimony-0.9.3b-37-g8b805c67/lib/fab/src/tree/v2syntax.c /usr/ports/cad/antimony/work/.build/lib/fab/v2syntax.lemon.cpp && mv /usr/ports/cad/antimony/work/antimony-0.9.3b-37-g8b805c67/lib/fab/src/tree/v2syntax.h /usr/ports/cad/antimony/work/.build/lib/fab/v2syntax.lemon.hpp
/bin/sh: LEMON_EXECUTABLE-NOTFOUND: not found
```

mkeeter, 2023-01-17:
Did you follow the guide in `BUILDING.md`, including installing `lemon`?

If you run `lemon` from the command line, does it work?

Also, I don't know what operating system you're running, which makes this harder to debug.

yurivict, 2023-01-17:
The point is that configure doesn't check that the lemon executable is installed.

mkeeter, 2023-01-17:
If you add `REQUIRED` [here](https://github.com/mkeeter/antimony/blob/ee525bbdad34ae94879fd055821f92bcef74e83f/lib/fab/CMakeLists.txt#L10), does it then fail at config time?  If so, I'd be happy to take a PR.

mkeeter, 2023-01-17:
Great, please open a PR with that fix.

yurivict, 2023-01-17:
Sorry, I am not dependent on the fix and have no resources for PRs.

commit f6a56dd7 (2023-01-18): Mark lemon as required (fixes #236)
closed 2023-01-18 by mkeeter